silvanmelchior/RPi_Cam_Web_Interface #213: Starting issue

Jonesinho, 2015-10-27:
Raspberry Pi Rev 2 - Raspian

scheduleLog.txt:
{2015/10/27 14:30:00} RaspiMJPEG Version 5.2.11
{2015/10/27 14:30:01} Error: Could not create camera
{2015/10/27 14:30:01} Executing macro /var/www/rpi-cam/macros/error_hard.sh "Could not create camera"  
[2015/10/27 14:30:02] RaspiCam support started
[2015/10/27 14:30:02] Capture Pipe already exists /var/www/rpi-cam/FIFO1
[2015/10/27 14:30:02] Scheduler loop is started
[2015/10/27 14:30:03] New period detected 0
[2015/10/27 14:30:03] Scheduled management tasks. Next at 1445956203
{2015/10/27 14:30:06} RaspiMJPEG Version 5.2.11
{2015/10/27 14:30:06} Error: Could not create camera
{2015/10/27 14:30:06} Executing macro /var/www/rpi-cam/macros/error_hard.sh "Could not create camera"

roberttidey, 2015-10-27:
Could not create camera means that raspimjpeg is unable to make a MMAL connection to the camera hardware.

This can be caused by something else using the camera or by a problem with the camera itself, or by power supply problems.

Check the ribbon cable connection to the camera at both ends. It is quite easy for this to get slightly skewed or pull out a bit.

Make sure you are using a good PS and a short USB cable as the extra current drawn by the camera can cause extra drop in that.

Also test camera using raspistill or raspivid with the RpiCam stopped completely.

Jonesinho, 2015-10-27:
Im using a USB camera, is this a problem?

roberttidey, 2015-10-27:
RPiCam is specifically for the Raspberry Pi camera. It does not support USB cameras.

The use of a Pi camera allows the interface to open up multiple channels to the camera for video, preview, still and motion detection and the raspimjpeg exploits this to achieve its functionality.

There are other software solutions on this forum suitable for USB cameras but they can't necessarily do all the things that RPiCam does at full HD resolution and frame rate.

Jonesinho, 2015-10-27:
Hi roberttidey,
thanks for this information.
Can you recommend a software for the raspberry pi, which has motion detection and a live stream web interface with audio?

roberttidey, 2015-10-27:
I use Pi camera exclusively, so I don't have any experience of other camera software.

You could look at https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=83369 which supports USB cameras and Pi cameras, but because it treats the Pi camera like a USB it has some limitations in resolution and frame rate. I don't think it does audio either.