silvanmelchior/RPi_Cam_Web_Interface #214: motion.php not showing configurations

a-l-e-x-d-s-9, 2015-10-30:
Tested on 2 Raspberry Pi 2, RPi Cam Control v6.0.12, with updated Raspbian Jessie:
     Linux raspberrypi 4.1.12-v7+ #824 SMP PREEMPT Wed Oct 28 16:46:35 GMT 2015 armv7l GNU/Linux

When turning on motion detection, motion.php only shows buttons, "Show All" \ "Show Less" do nothing:
[image]
[image]

I can still edit motion detection configuration in: /etc/motion/motion.conf.

roberttidey, 2015-10-31:
This is because of Jessie. motion.conf changed around quite a bit for Jessie and the names of some of the parameters changed compared to a wheezy install.. This is being worked on.

hekigan, 2018-12-20:
Hi, it looks like after 2 years, this issue is coming back.

I have the latest install of Raspbian `raspberrypi 4.14.79-v7+`, the latest version also for RPi_Cam_Web_Interface and the page only shows the buttons (like the screenshot above).
Anyone else experienced the issue and is there any fix for that?

a-l-e-x-d-s-9, 2018-12-20:
I'm guessing it's related to upgrade to Debian 9, "Stretch" release... 
I use the internal method of motion detection ("Camera Settings > Motion detect mode: Internal"), it's working fine with Debian 9 release.

hekigan, 2018-12-20:
That is correct. And the problem is that I want to use the "gap" property and it's only available in the external mode it seems.
Do you know if there is some quick fix that I could do?

a-l-e-x-d-s-9, 2018-12-20:
Nope, haven't used the external method for long time.

hekigan, 2018-12-20:
That's unfortunate, I hope someone from the dev side will be able to have an explanation then.
Thanks for your reply.

hekigan, 2018-12-20:
Ok, I managed to fix it by changing the permissions on the folders/files

roberttidey, 2018-12-20:
It should work fine on stretch. I have several cameras on this.

Can you say what change you made to the permissions on folders / files? A normal install should set ownership and permissions OK. If you have changed or linked to a different location (e.g. ona USB drive), then it is necessary that the destination is given permissions for user www-data to read and write there.

The still frames to stop setting on internal detection is effectively the same as the gap setting on external detection. So for example a setting of 150 frames means that 6 seconds of little activity is needed to stop the video.

hekigan, 2018-12-22:
@roberttidey Well, to be honest I don't remember how I installed the whole thing, but I am certain that I did not touch the permission of the files. But after changing it to `www-data`, the properties were showing at last.
Maybe it depends on the way it was installed -with or without `sudo`-, I am not sure.

Thank you for the extra info by the way