silvanmelchior/RPi_Cam_Web_Interface #217: Camera extremely slow and 'Could not open PIPE' on Jessie

macecchi, 2015-10-31:
After following the instructions on the wiki page to install the app on **Raspbian Jessie** (using **nginx**), I get the web interface working but the camera feed is extremely slow (the image refreshes about every 5s).

Every time the image refreshes,** the camera led goes on and then off again for a few seconds**, as if it was being switched off every time. When it comes back on, the white-balancing starts to work as if the camera has just started.

`scheduleLog.txt` repeats the same log messages and errors every 5s:

```
{2015/10/30 23:59:02} RaspiMJPEG Version 5.2.11
{2015/10/30 23:59:02} recommended video buffers 1
{2015/10/30 23:59:02} h264 buffers set to recommended 1
{2015/10/30 23:59:03} MJPEG streaming, ready to receive commands
{2015/10/30 23:59:03} Error: Could not open PIPE
{2015/10/30 23:59:08} RaspiMJPEG Version 5.2.11
{2015/10/30 23:59:08} recommended video buffers 1
{2015/10/30 23:59:08} h264 buffers set to recommended 1
{2015/10/30 23:59:08} MJPEG streaming, ready to receive commands
{2015/10/30 23:59:08} Error: Could not open PIPE
{2015/10/30 23:59:13} RaspiMJPEG Version 5.2.11
{2015/10/30 23:59:13} recommended video buffers 1
{2015/10/30 23:59:13} h264 buffers set to recommended 1
{2015/10/30 23:59:13} MJPEG streaming, ready to receive commands
{2015/10/30 23:59:13} Error: Could not open PIPE
```

Also, none of the buttons (record video/image/stop camera/etc) seem to be working. 

After a while I sometimes get a **`504 Gateway Time-out`** trying to reach the server, and I have to restart the Pi. 

Any clues how to proceed? I'm running a clean, 100% updated install of Raspbian Jessie.

roberttidey, 2015-10-31:
Jessie is not fully supported yet. Unfortunately a lot of things changed in the config files for Apache and motion. We haven't even looked at nginx. The only one tested so far is an install under Apache and using the default /var/www/html folder

The problem you have with 5 second refresh is because it is detecting a problem and going through a restart cycle which is taking 5 seconds. If it can't open the pipe then it is probably a permissions problem caused by the installer not being fully Jessie aware yet. All files and the FIFO pipes in the installed folder should be owned by www-data. If they aren't then you will need to chown them.

macecchi, 2015-10-31:
Thanks for the quick reply. I realise Jessie is not fully supported yet. 

**About nginx**, the only workaround needed was adding the line `fastcgi_param SCRIPT_FILENAME $document_root/$fastcgi_script_name;` to /etc/nginx/sites-available/rpicam, and fixing root to /var/www/html (the file was still pointing to /var/www even after selecting html during installation).

**About the camera problem**, I've double checked and everything under /var/www/html is owned by www-data. Is there another way to debug where the problem is coming from? I'm interested in contributing :)

roberttidey, 2015-10-31:
Thanks. I am working on some new scripts here as well (referred to in the forum). I also use that nginx info.

I'll add further comments about camera problem later

roberttidey, 2015-11-01:
I've updated the rpicam.1 config file with that change, but not sure if I hav eput it in the right place.

On the root issue. For Apache we leave the root at /var/www and specify a directory at a subfolder as required. So one browses to //localhost/subfolder. This method allows multiple mini web sites in different subfolders sharing the same port. So for nginx does browsing to localhost/html work even with root left at the default?

macecchi, 2015-11-02:
I've removed everything and reinstalled after a `git pull`. **Browsing to /html works!** (using nginx)

I did get a few errors while running the installer though, right after the **"Install path is set /var/www/html/" ** screen:

```
./RPi_Cam_Web_Interface_Installer.sh: line 666: []: command not found
cp: cannot create regular file ‘/etc/php5/conf.d/20-apc.ini’: No such file or directory
chmod: cannot access ‘/etc/php5/conf.d/20-apc.ini’: No such file or directory
cp: cannot stat ‘/etc/raspimjpeg’: No such file or directory
```

roberttidey, 2015-12-01:
php5 has changed around its cache and where its cache settings go and that is what causes these errors. It is not particularly important and needs a bit more research to sort out how to handle.

macecchi, 2015-12-29:
I'm closing this issue as this seems to be fully fixed now :)

closed 2015-12-29 by macecchi