silvanmelchior/RPi_Cam_Web_Interface #218: Watchdog terminates motion detection, internal mode

a-l-e-x-d-s-9, 2015-11-08:
Using Jessie, v6.0.12, RaspiMJPEG Version 5.2.11.
Internal motion detection stop working after "Watchdog detected problem".

Configurations:
[image]

Log:

``` log
{2015/11/07 20:18:21} RaspiMJPEG Version 5.2.11
{2015/11/07 20:18:21} recommended video buffers 1
{2015/11/07 20:18:21} h264 buffers set to recommended 1
{2015/11/07 20:18:21} Set up internal detect width=121 height=68
{2015/11/07 20:18:21} MJPEG streaming, ready to receive commands
{2015/11/07 20:18:21} send smd 9
[2015/11/07 20:18:21] Reload parameters command requested
[2015/11/07 20:18:21] Scheduler loop is started
[2015/11/07 20:18:22] Scheduled management tasks. Next at 1446921502
{2015/11/07 20:34:37} RaspiMJPEG Version 5.2.11
{2015/11/07 20:34:37} recommended video buffers 1
{2015/11/07 20:34:37} h264 buffers set to recommended 1
{2015/11/07 20:34:37} Set up internal detect width=121 height=68
{2015/11/07 20:34:37} MJPEG streaming, ready to receive commands
{2015/11/07 20:34:37} send smd 9
[2015/11/07 20:34:39] RaspiCam support started
[2015/11/07 20:34:39] Capture Pipe already exists /var/www/html/FIFO1
[2015/11/07 20:34:39] Scheduler loop is started
[2015/11/07 20:34:39] Reload parameters command requested
[2015/11/07 20:34:39] Scheduler loop is started
[2015/11/07 20:34:40] New period detected 0
[2015/11/07 20:34:40] Scheduled management tasks. Next at 1446922480
{2015/11/07 20:34:40} Internal motion detection started
{2015/11/07 20:34:40} recommended video buffers 1
{2015/11/07 20:34:40} h264 buffers set to recommended 1
{2015/11/07 20:34:40} send smd 1
[2015/11/07 20:34:40] Start capture requested from Pipe
[2015/11/07 20:34:40] Send ca 1
{2015/11/07 20:34:40} Set up internal detect width=121 height=68
{2015/11/07 20:34:40} Capturing started
{2015/11/07 20:35:19} send smd 1
[2015/11/07 20:35:19] Start capture requested from Pipe
[2015/11/07 20:35:19] Send ca 1
{2015/11/07 20:35:19} Already capturing. Ignore 
{2015/11/07 20:35:24} send smd 0
[2015/11/07 20:35:24] Stop capture requested
[2015/11/07 20:35:24] Send ca 0
{2015/11/07 20:35:25} Capturing stopped
{2015/11/07 20:35:25} Add /var/www/html/media/vi_0816_20151107_203440.mp4 to Box Queue at pos 0
{2015/11/07 20:35:25} Start boxing /var/www/html/media/vi_0816_20151107_203440.mp4.h264 to /var/www/html/media/vi_0816_20151107_203440.mp4 Queue pos 0
{2015/11/07 20:35:26} send smd 1
[2015/11/07 20:35:26] Start capture requested from Pipe
[2015/11/07 20:35:26] Send ca 1
{2015/11/07 20:35:27} Capturing started
{2015/11/07 20:35:32} send smd 0
[2015/11/07 20:35:32] Stop capture requested
[2015/11/07 20:35:32] Send ca 0
{2015/11/07 20:35:32} Capturing stopped
{2015/11/07 20:35:32} Add /var/www/html/media/vi_0817_20151107_203527.mp4 to Box Queue at pos 1
{2015/11/07 20:35:46} Finished boxing /var/www/html/media/vi_0816_20151107_203440.mp4 from Box Queue at pos 0
{2015/11/07 20:35:46} Start boxing /var/www/html/media/vi_0817_20151107_203527.mp4.h264 to /var/www/html/media/vi_0817_20151107_203527.mp4 Queue pos 1
{2015/11/07 20:35:47} Finished boxing /var/www/html/media/vi_0817_20151107_203527.mp4 from Box Queue at pos 1
{2015/11/07 20:36:33} send smd 1
[2015/11/07 20:36:33] Start capture requested from Pipe
[2015/11/07 20:36:33] Send ca 1
{2015/11/07 20:36:33} Capturing started
{2015/11/07 20:36:37} send smd 0
[2015/11/07 20:36:37] Stop capture requested
[2015/11/07 20:36:37] Send ca 0
{2015/11/07 20:36:37} Capturing stopped
{2015/11/07 20:36:37} Add /var/www/html/media/vi_0818_20151107_203633.mp4 to Box Queue at pos 2
[... 874 lines cut ...]
[2015/11/08 06:13:09] Scheduled management tasks. Next at 1446957189
[2015/11/08 06:13:09] Purged 51 Files
{2015/11/08 06:18:06} Watchdog detected problem. Stopping{2015/11/08 06:18:08} SIGINT/SIGTERM received, stopping
{2015/11/08 06:18:13} RaspiMJPEG Version 5.2.11
{2015/11/08 06:18:13} recommended video buffers 1
{2015/11/08 06:18:13} h264 buffers set to recommended 1
{2015/11/08 06:18:13} Set up internal detect width=121 height=68
{2015/11/08 06:18:13} MJPEG streaming, ready to receive commands
{2015/11/08 06:18:13} send smd 9
[2015/11/08 06:18:13] Reload parameters command requested
[2015/11/08 06:18:13] Scheduler loop is started
[2015/11/08 06:18:14] Scheduled management tasks. Next at 1446957494
[2015/11/08 06:18:14] Purged 20 Files
[2015/11/08 06:38:15] Scheduled management tasks. Next at 1446958695
[2015/11/08 06:58:16] Scheduled management tasks. Next at 1446959896
[2015/11/08 07:18:17] Scheduled management tasks. Next at 1446961097
[2015/11/08 07:38:18] Scheduled management tasks. Next at 1446962298
[2015/11/08 07:58:19] Scheduled management tasks. Next at 1446963499
[2015/11/08 08:18:20] Scheduled management tasks. Next at 1446964700
[2015/11/08 08:38:21] Scheduled management tasks. Next at 1446965901
[2015/11/08 08:58:22] Scheduled management tasks. Next at 1446967102
[2015/11/08 09:18:23] Scheduled management tasks. Next at 1446968303
[2015/11/08 09:38:24] Scheduled management tasks. Next at 1446969504
[2015/11/08 09:58:25] Scheduled management tasks. Next at 1446970705
```

SIYB64, 2015-11-10:
I'm getting a similar problem, internal motion detect seems to turn off after a "Scheduled management tasks. Next at xxx" event.

Log file and motion setting attached.

Simon
[image]

roberttidey, 2015-11-10:
In the second log everything is going along Ok until 09:50 then it looks like the raspimjpeg restarted (version number logged). I can't see anything leading up to this that would have caused that. 

The normal start up default is motion detect off and then it is only set to on either by a schedule command, by pushing the button or some external command. None of the happen until 14:01:33 when I guess some schedule operation was done. It is possible to set the default start up to be motion detect on by changing motion_detection false to motion_detection true in /etc/raspimjpeg or by adding that to uconfig in the www folder.

SIYB64, 2015-11-10:
Hi,

the schedule is:-
[image: Inline images 1]

so I'd expect it to be running all the time.

I'll take a look at the /etc/raspimjpeg file and change the default to true
- just to see if this works.

Thanks

Simon

a-l-e-x-d-s-9, 2015-11-30:
@roberttidey 
Can you please explain what may cause watchdog error? And if there is some workaround/recovery procedure?

roberttidey, 2015-12-01:
The watchdog is a built in second copy of raspimjpeg which is running and monitoring the main operational copy of raspimjpeg. The main raspimjpeg does some checks whilst it is running like making sure that previews are still being generated and kills itself if it detects an error. This could be caused , for example, by  a failure in the camera driver connection. The watchdog raspimjpeg sees this and automatically starts up a fresh copy of raspimjpeg which should then operate normally.