silvanmelchior/RPi_Cam_Web_Interface #219: Motion detection noise level and threshold values

a-l-e-x-d-s-9, 2015-11-10:
Currently, it's pretty hard to determine desired parameters for noise level and threshold, for motion detection internal/external, due to changes in lighting conditions, auto exposure-mode/white-balance. I think that the main problem with current situation, is that the levels of these parameters unavailable for user to see while they are changing and it's hard to guess them. Resulting difficulties to set/debug/understand behaviour and quirks in motion detection mode.
Is it possible to add noise level and threshold fields, with live values as they change alongside video preview, while in motion detection mode?

roberttidey, 2015-11-10:
This is possible for internal motion as this code is part of raspimjpeg which is displaying the previews. It is not really possible for external motion as that is in its own self contained process with no real access to what is going on prior to a trigger.

For internal there are two annotation parameters for this. %c give the current change level and %f counts the number of changed frames. So by including those in the annotation template then you get a real time update of the measurements causing motion start and  stop signals. With the video preview buffer you get to see these values leading up to the start.

a-l-e-x-d-s-9, 2015-11-10:
@roberttidey 
Great, thank you very much!

closed 2015-11-10 by a-l-e-x-d-s-9