silvanmelchior/RPi_Cam_Web_Interface #222: Main buttons don't change their state after press

a-l-e-x-d-s-9, 2015-11-25:
RPi Cam Control v6.0.17 with Jessie.
For example "motion detection start" when pressed, not changing to "motion detection stop", and not disabling "record video start" and "timelapse start" buttons.
They always look like this:
[image]
When pressed, they work, but it's hard to tell in which exact mode, without enabled/disabled and text change indications.
Tested on Chrome and Firefox.

a-l-e-x-d-s-9, 2015-11-25:
Reverted to commit 7818460b3afcc7d92e76af04a9720093726a97c8 (Date:   Tue Nov 24 10:31:30 2015 +0000)
Buttons are perfectly fine there.

roberttidey, 2015-11-26:
I think this is because the link to status_mjpeg.txt did not get set up properly for some reason. Prior to 6.0.17 this file was a real file in the web folder, but it now gets updated every 5 seconds. So to avoid rewriting flash memory the actual status_mjpeg.txt file is now in /dev/shm/mjpeg and a link made in the web folder.

/etc/raspimjpeg should now have status_file  /dev/shm/mjpeg/status_mjpeg.txt

The installer should have created a link in the web folder
lrwxrwxrwx 1 www-data www-data 31 Nov 25 09:51 status_mjpeg.txt -> /run/shm/mjpeg/status_mjpeg.txt

You can manually create the link with
sudo ln -sf /run/shm/mjpeg/status_mjpeg.txt /var/www/sub/status_mjpeg.txt

substitute sub as required

a-l-e-x-d-s-9, 2015-11-29:
I installed 6.0.19, status_mjpeg.txt located directly in the in /var/www/ and not in /var/www/html/.
When I created a soft link in /var/www/html/, buttons started to work as expected.
So there is some confusion with the directory of the link.

``` shell
ls -la /var/www/
total 12
drwxr-xr-x  3 root     root     4096 Nov 28 23:22 .
drwxr-xr-x 12 root     root     4096 Oct 29 21:17 ..
drwxr-xr-x  7 www-data www-data 4096 Nov 25 23:04 html
lrwxrwxrwx  1 root     root       31 Nov 28 23:22 status_mjpeg.txt -> /run/shm/mjpeg/status_mjpeg.txt
```

roberttidey, 2015-11-29:
Which install method did you use? I am concentrating on install.sh now.

a-l-e-x-d-s-9, 2015-11-29:
I wasn't aware of install.sh. I'll try it later...

a-l-e-x-d-s-9, 2015-11-30:
I tried install.sh, it messed up everything for me, got error "403 Forbidden nginx/1.6.2" from the server.
Then I thought just to reinstall, with remove.sh & install.sh, but it removed the whole /var/www/html folder, including the media folder with all the videos, it's kinda a bummer... :( 
I don't think it's wise to automatically remove media folder without any warning...

In install.sh I've chosen to install apache, but I have nginx error, not sure what's the deal with that...
Apparently the "cancel" in install.sh installing the software just as "ok", so now I have all the files both in /var/www/ and in /var/www/html/ folders.
My /etc/nginx/sites-enabled/rpicam has the line "fastcgi_param SCRIPT_FILENAME..." as it should. But I still stuck with error "403 Forbidden".
Have no idea what I'm doing wrong... :(

roberttidey, 2015-12-01:
There is an issue if one tries nginx and then apache. One would need to do a remove as otherwise you end up with two web servers running fighting to serve the pages which nginx wins. However, the nginx remove procedure seems to remove the niginx package but the core nginx remains still running. The only way to fix this is to manually remove nginx or start from a clean image.

The install script does try to move the media folder if one changes the install location. It might be an idea to move the media folder to the users home directory if doing a total remove.

a-l-e-x-d-s-9, 2015-12-02:
Thanks, I uninstalled nginx, and reinstall everything as before, it's working fine now.
Regarding the media folder, remove.sh can just ask the user if he want to remove it or leave the media folder and remove the rest. Automatic copy of dozens of gigabytes is probably not a good idea...

roberttidey, 2015-12-02:
I'll think about remove options while I work out nginx uninstall issues. If the media is moved not copied it shouldn't involve any physical data movement just updates to the filing system to say where the media folder root is.

JoergPf, 2016-01-22:
Same problem here:
I just installed Jessie 4.1 and RPi Cam Control 6.0.25.
The buttons don't change.

When I do 
sudo ln -sf /run/shm/mjpeg/status_mjpeg.txt /var/www/html/rcam/status_mjpeg.txt

it works, but now the just recorded videos disappear. There are only the posterframes left in the media folder.

Any idea?

roberttidey, 2016-01-22:
I assume you are using the install.sh method. You are using install 2 subfolders deep; the link command might have a problem with that, I'll check. The install.txt is a log of the installer actions which is helpful to check out what happened.

I don't understand the comment about recorded videos disappearing and "posterframes" left. Do you mean thumbnail files get left and mp4 files disappear just by adding the link? I am not sure how that could happen as the thumbnails themselves are used to find the video files. Looking at the log under Scheduler might help understand what is going on.

JoergPf, 2016-02-01:
I did a complete reinstall with raspi jessy and used install.sh.
It's working now, that the video button changes form 'start recording' to 'stop recording'. But there are 2 issues:

When I installed apache, /var/www/html was my root directory for loclahost. When I use install.sh to install RPi_Cam_web_interface and I choose the default path 'html', then it changes my root directory of apache localhost to /var/www.
So when I want to start  RPi_Cam_web_interface, I have to call localhost/html in my browser.

The other problem ist still like reported above: When I record a video, I can see, that there are a *.jpg file and a *.h264 file created in the media folder.
But when I stop the recording, the .h264 file disappears and I can find it nowhere. Only the .jpg file stays in the folder. 

Thanks for any help and for the script!

roberttidey, 2016-02-01:
The first part changing the root directory is deliberate. There was a lot of discussion about this on the forums a while back. Basically there are two schemes. The first is where the root directory points directly at the install of the camera web files and the main url is just the hostname. The other is where the root directory is always set to /var/www and the url to the camera then includes the path to the camera web files. The first allows multiple independent web sites using the default 80 port by including the path in the url. The second can only really separate different web sites by using different ports but does allow the url to be just the hostname. On balance most people preferred the former scheme and if it was just the camera software then it can be installed directly in /var/www by leaving the subfolder blank and then the url is just the hostname.

There was also a change in the default config of Apache between Wheezy and Jessie. Wheezy had the default root as /var/www and Jessie as /var/www/html. install.sh unifies it. If you really want to have everything under html but no path in the url then you have to change the Apache root after install.

On the second issue when you start a recording then the th,jpg is created straight away and a file is created with an extension mp4.h264. The raw h264 video data is recorded into this file. When the video is stopped then one of two things happens depending on the Boxing mode control under camera settings. If this is set off then the file is just renamed to .h264 and the video is left in a raw state. This is for users who wish to convert the files to some video container format separately. h264 files can't be viewed directly in the browser. If Boxing mode is set to background then the software starts a background process to convert the file to a real video format in mp4 format. When that completes the new file is labelled just .mp4 and the .mp4.h264 file is deleted.

It sounds like for some reason the conversion process (MP4Box) is failing to produce the file but I haven't seen a problem with this.

First you should check the log (under scheduler). It should have lines like

{2016/02/01 23:07:51} Capturing started
{2016/02/01 23:08:04} Capturing stopped
{2016/02/01 23:08:04} Add /var/www/media/vi_0000_20160201_230751.mp4 to Box Queue at pos 0
{2016/02/01 23:08:05} Start boxing /var/www/media/vi_0000_20160201_230751.mp4.h264 to /var/www/media/vi_0000_20160201_230751.mp4 Queue pos 0
{2016/02/01 23:08:06} Finished boxing /var/www/media/vi_0000_20160201_230751.mp4 from Box Queue at pos 0
{2016/02/01 23:08:06} Removed item from Box Queue

If that is happening but you don't get the mp4 then the MP4Box is failing for some reason. Maybe some dependency is missing although the installer does include stuff for this.

What you can do to test is to set Boxing Mode to Off. This should give you .mp4.h264 files in the media folder. Then run MP4Box manually in a command window using a cmd line like

MP4Box -fps 25 -add sourcefile.h264 destfile.mp4

replace sourcefile with real .h264 filename.

The command window should show any errors that are occurring with running the command.

JoergPf, 2016-02-02:
ok, here is the problem: MP4Box: Command not found.
I did the installation according to this tutorial:http://elinux.org/RPi-Cam-Web-Interface

Did I forget something? What do I have to install?

roberttidey, 2016-02-02:
MP4Box is part of gpac. Wheezy had this already but was not included in Jessie. The install.sh does install it for both anyway.

Try installing it with sudo apt-get install gpac

JoergPf, 2016-02-02:
I still wonder, why install.sh didn't install gpac when I installed RPi_Cam_web_interface, but now it's working. Great! Thanks

roberttidey, 2016-02-02:
The install.txt lists the activity of the install.sh script and should show the apt-get of all modules being installed including gpac. 

There is no specific error handling there though so if there was a temporary problem with the install of a module then it would likely go unnoticed.