silvanmelchior/RPi_Cam_Web_Interface #224: Additional buttons for controlling other servos

AFUDIrk, 2015-12-07:
Is it possible to bring additional buttons for controlling other servos on the web interface?
I want to use it to control a zoom lens, so zoom in and zoom out, as well as the focus of the lens.

roberttidey, 2015-12-07:
It should be possible. That would need adding the buttons into the pan section of the web interface and then adding extra actions into the pipan.php back end process to control the extra servos. The extra buttons would need to be optional as not many would need this.

Have you got zoom and focus lens in front of the standard pi camera module?

AFUDIrk, 2015-12-07:
​That's right, I also thought that you can enable it only when needed.
One must also only pipan_off the file set to ON when you want to have the
control.
I want to implement the example with a 15x zoom lens from an old mini-DV
camcorders.
Bought on e-bay for about 8 Euro.​