silvanmelchior/RPi_Cam_Web_Interface #229: How to get status of motion detector 'md' from terminal

herious1989, 2015-12-29:
I want to get notification when motion detect is turned on or off. How can I retrieve this info from terminal?

roberttidey, 2015-12-29:
The status file in the web install folder (status_mjpeg.txt) holds the state of the system and will have one of the following 4 values if motion detection is on (tl_md_video, md_video, tl_md_ready, md_ready). Any other value means motion detection is off.

If you want the software to run something else (like a macro) when motion is turned on or off then that would need a change. Let me know if you wanted that. Also would be good to know the intended purpose in case there are other solutions.

herious1989, 2015-12-29:
@roberttidey To be able to run a macro when motion is on/off is useful though. I want run a script to notify me or execute some scripts when motion is switched to on or off [by mistake or intentionally by other people]

roberttidey, 2016-01-10:
Latest version now includes a do_cmd macro which gets called (if present) when the md start/stop is executed. It gets passed the cmd line as a parameter so would be either md 1 (start) or md 0 (stop)

ScheissSchiesser, 2017-10-23:
I've created a macro `macros/do_cmd`:
```
#!/usr/bin/env bash
/var/www/html/macros/motionstate.sh "mdtest"
```
motionstate.sh is a script that I have already tested (and I can run using the scheduler).
Testing `do_cmd` on the command line works fine:
```
_pi@rpi:/var/www/html $_ sudo -u www-data macros/do_cmd
mdtest
{"event":"motionState","trackers":{"motionState":"TesT"}}
{"msg":"Notification Sent Successfully","error":false,"status":200}
```
But this script does not seem to run when I change the `md` state on or off via FIFO.

After typing this out, I seem to have solved the problem: `do_cmd` must be named `do_cmd.sh`, it seems. Does this mean the shebang line doesn't matter? How are these pre-defined scripts actually run?

roberttidey, 2017-10-23:
macros have a logical name like do_cmd or end_vid. These are mapped onto physical script filenames in the /etc/raspimjpeg config file as in
do_cmd &do_cmd.sh
If the physical script filename is prepended with an & then that indicates that it should be run asynchronously; the & is effectively moved to the end after parameters have been added.. Otherwise it is run synchronously.

Scripts are run by the raspimjpeg process by executing them as a command process. They must be in the macros folder (as defined in the config file), and they must be owned by and have execution privileges of user www-data. This helps protect against security issues.

The latest versions of the web interface allow the mapping of scripts to physical filenames to be edited from the browser under the system menu.

ScheissSchiesser, 2017-10-23:
Let me see if I have this right: `do_cmd` is the name of the "macro" command. When raspimjpeg calls the macro `do_cmd`, it runs an actual file named `do_cmd.sh`, or whatever is defined in the `raspimjpeg` config file under the `do_cmd` macro mapping.

Thus, I could define: 
```
do_cmd &any_script_name.sh
```
in the `raspimjpeg` config file, and as long as `any_script_name.sh` is in the `.../macros/` directory, `raspimjpeg` will run `any_script_name.sh` when `{some command}` is run. Current behavior is that `{some command}` is only the `md` command. Is that right?

roberttidey, 2017-10-23:
Yes. That is correct. Also the mapping part of the raspimjpeg config file can be changed using the um FIFO command. This is what the web interface uses to change the mapping. See wiki for details.

Currently the do_cmd macro is called when either the ru or md command is processed. The cmd that is processed is passed as a parameter to the macro