silvanmelchior/RPi_Cam_Web_Interface #230: Problems with FIFO on manual installation of raspimjpeg

steinjo, 2016-01-03:
Hi,

first of all, thanks for providing this fantastic tool raspimjpeg!

I want top use only the raspimjpeg as a service on my raspi B mod 2, w/o all the other php-software.  I went through the installation script and extracted all what seemed necessary to me for that purpose.  The problem is that after that manual installation raspimjpeg does not seam to read from FIFO-pipe.

These are the manual installation steps:
git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
sudo cp -r bin/raspimjpeg /opt/vc/bin/
sudo chmod 755 /opt/vc/bin/raspimjpeg
sudo ln -s /opt/vc/bin/raspimjpeg /usr/bin/raspimjpeg
cat etc/raspimjpeg/raspimjpeg.1 > etc/raspimjpeg/raspimjpeg
sudo cp -r etc/raspimjpeg/raspimjpeg /etc/
sudo chmod 644 /etc/raspimjpeg
sudo ln -s /etc/raspimjpeg /var/www/raspimjpeg
cat etc/motion/motion.conf.1 > etc/motion/motion.conf
sudo mkdir /etc/motion
sudo cp -r etc/motion/motion.conf /etc/motion/
sudo chmod 664 /etc/motion/motion.conf
sudo mkdir /var/www/media
sudo mkdir /var/www/macros
sudo mknod /var/www/FIFO p
sudo chmod 666 /var/www/FIFO
sudo mknod /var/www/FIFO1 p
sudo chmod 666 /var/www/FIFO1
sudo mkdir -p /dev/shm/mjpeg
sudo chmod 777 /dev/shm/mjpeg
sudo ln -sf /run/shm/mjpeg/cam.jpg /var/www/cam.jpg
sudo ln -sf /run/shm/mjpeg/status_mjpeg.txt /var/www/status_mjpeg.txt

When I start raspimjpeg from command line (sudo raspimjpeg > /dev/null 2>&1 &) it runs nicely as a process (checked with top) and the schduleLog.txt contains:

{2016/01/03 13:58:58} RaspiMJPEG Version 5.2.14
{2016/01/03 13:58:58} recommended video buffers 1
{2016/01/03 13:58:58} h264 buffers set to recommended 1
{2016/01/03 13:58:58} MJPEG streaming, ready to receive commands
{2016/01/03 13:58:58} send smd 9

The image in /dev/shm/mjpeg/cam.jpg is regularly updated, so everything seems fine.

PROBLEM:  when I write into FIFO via command line (sudo echo 'im 1' > FIFO), the terminal does not return to command line prompt.  Only way to get back to the prompt is ctrl + c.  I also tried other commands like ru 0, ru 1 - same problem.

However, if I open another terminal and enter cat FIFO in command line, it correctly reads out the 'im 1' when I enter it by above command in the first terminal.  Therefore I guess, that the FIFO-pipe is working, but raspimjpeg is not picking up what is written into FIFO.  In /etc/raspimjpeg, the FIFO-pipe is mentioned as 'control_file /var/www/FIFO', which is where the FIFO-pipe resides.

I would very much appreciate your help or hint to the right direction to search for the origin of this problem.

Thanks in advance!

steinjo

roberttidey, 2016-01-03:
Two thoughts. 
Do you have the scheduler background process running as well?
The send smd 9 item in the log is raspimjpeg trying to write to FIFO1 which the scheduler reads. The 9 command is used to tell the scheduler to reset itself. If the scheduler is not running then FIFO1 may be blocking the part of raspimjpeg that is accepting commands although the image generation would still be working. One can avoid this if not using the scheduler at all by configuring /etc/raspimjpeg. The motion_pipe value is normally /var/www/FIFO1. Change that so there is no text after motion_pipe. You will still see the log message but raspimjpeg won't attempt to write to FIFO1. I should change that so the log only happens for a real write.

Second thought is that there may be some permissions issue due to the manual installation.. Normally raspimjpeg runs as user www-data (default web user) and all other permissions revolve around this.

steinjo, 2016-01-03:
Thanks a lot, the first thought was spot-on!  I can now successfully send commands to FIFO.

However, does that mean now, that with no value after motion_pipe, I can not use the motion-detection-module, because this change completely disables all output to FIFO1?  That would be a pity. I do not want to use the scheduler, but would love to deploy the motion-detection.

Thx again!

steinjo, 2016-01-03:
Uuups, I just looked at the scheduleLog.txt.  It contains numerous (one every 5 seconds) entries like this:
{2016/01/03 21:45:46} RaspiMJPEG Version 5.2.14
{2016/01/03 21:45:46} Error: Could not enable camera

But that can not be true, because the camera is streaming and I can take pictures (im 1) ...

steinjo, 2016-01-03:
Sorry, please ignore my last entry on the 'Error: Could not enable camera', I had a second faulty process for raspimjpeg running.  After killingall and starting it again, there are no more error-messages.

roberttidey, 2016-01-03:
Good to here its working now. The point about motion detection relying on scheduler is because the scheduler set up currently is what determines what to do when motion start and stop triggers are generated. This allows for different sequences of actions to be performed including video recording, still image captures, macros plus these can be dependent on the time of day when triggers occur.

It may be possible to have a config option to allow just fixed video recording without a scheduler. Another method would be to have a simple FIFO1 reader that reflected the motion start and stop back into FIFO pipe commands; the equivalent of a simple scheduler process running in whatever environment you choose (e.g. a python module).

steinjo, 2016-01-05:
Thanks again!

I digged into this a bit, but not by looking into the code for scheduler.php etc. in detail.  However, I think, I get your point as follows:

When I leave the entry in /etc/rapsimjpeg to the original "motion_pipe /var/www/FIFO1" and start raspimjpeg, then I have the problem, described in my first post.  In that state, FIFO1 contains "9".  If I read that out from FIFO1 via command line (cat FIFO1), I can then enter commands in FIFO (tested only with "im 1") which are also executed perfectly.  

Since you suggest as a solution to implement a small scripted process (e.g. in python) which continuously feeds anything showing up in FIFO1 into FIFO (if I understood correctly), there are two questions, popping up:

1. Is such a continuously running process necessary?  Isn't it sufficient, if I read out the "9" in FIFO1 only once after having started raspimjpeg?  That might not be sufficient if raspimjpeg sends commands FIFO1 at a later point, leading to the same dead-lock like on first start-up?  If that is the case, which instances lead to these FIFO1-commands?  I am thinking about just adding an exec_shell('cat FIFO1') after these instances that lead to a dead-lock, instead of having a script continuously running.
2. In that context:  would a simple "read-out" of FIFO1 (like exec_shell('cat FIFO1')) be sufficient at all or does the content of FIFO1 also have to be written into FIFO?

Thanks for your ongoing support!

roberttidey, 2016-01-06:
There are 2 types of events which trigger raspimjpeg to write to FIFO1.

The first is at start up when it sends a 9 which is the signal to scheduler process to reset itself. This is to ensure that things like time period change handling in the scheduler sends back appropriate FIFO commands to raspimjpeg.

The second is when internal motion detection is used. raspimjpeg is then analysing vector change data from the camera and producing motion start and motion stop triggers. A motion start trigger sends a 1 to FIFO1 and a motion stop sends a 0 to FIFO1. That is all they do; they do not initiate any recording action directly in raspimjpeg. It is up to the external FIFO1 process (normally scheduler) to interpret these triggers and then send in normal FIFO commands to raspimjpeg to perform whatever action is required. That can be things like ignore triggers, start stop video, take a still image, do time lapse recording etc.

So with the current raspimjpeg one must either turn off all FIFO1 output (disabling the possibility of using internal motion detection) or have a FIFO1 monitoring process taking in motion start/stop triggers and turning them into FIFO commands to do what is intended. For example a 1 incoming on FIFO1 could result in ca 1 being sent to FIFO and a 0 sends ca 0.

What I suggested above was that it could be possible to have a config option in raspimjpeg to avoid the external process and perform just video start / stop but that would mean that a lot of the logic flexibility of the external process (like scheduler) would be lost.

steinjo, 2016-01-17:
Sorry, I was off a few days due to business.

Thanks a bunch for your great explanations.  I can certainly take it from here.

Cheers!