silvanmelchior/RPi_Cam_Web_Interface #231: Sense Hat integration in RPI CAM

degufo, 2016-01-09:
HI Guys!

I would like to add the Raspberry Pi Sense Hat module to the annotation text line.

Is this possible?

Any help would be very appreciated!

Thanks

Script:

import picamera
import datetime as dt
from subprocess import call
import threading
from sense_hat import SenseHat
import time
from stick import SenseStick

CamIdStr = 'Cam_LB01'
SecondsFile = 1800       # must be > 60
VideoQuality = 20
TextSize = 25
CamRotate = 180

sense = SenseHat()
stick = SenseStick()

mag_x = 0.0
mag_y = 0.0
mag_z = 0.0
acc_x = 0.0
acc_y = 0.0
acc_z = 0.0
hum = 0.0
temp = 0.0
press = 0.0

run = True

def SenseThread(threadname):
  global mag_x
  global mag_y
  global mag_z
  global acc_x
  global acc_y
  global acc_z
  global hum
  global temp
  global press
  while(run):
    mag_raw = sense.get_compass_raw()
    mag_x = 0.99*mag_x + 0.01*mag_raw['x']
    mag_y = 0.99*mag_y + 0.01*mag_raw['y']
    mag_z = 0.99*mag_z + 0.01*mag_raw['z']
    acc_raw = sense.get_accelerometer_raw()
    acc_x = 0.99*acc_x + 0.01*acc_raw['x']
    acc_y = 0.99*acc_y + 0.01*acc_raw['y']
    acc_z = 0.99*acc_z + 0.01*acc_raw['z']
    hum = 0.99*hum + 0.01*sense.get_humidity()
    temp = 0.99*temp + 0.01*sense.get_temperature()
    press = 0.99*press + 0.01*sense.get_pressure()
    time.sleep(0.01)

# Create new threads
thread1 = threading.Thread(target = SenseThread, args = ("SenseThread", ))
thread1.start()

	
class myThread (threading.Thread):
  def __init__(self, threadID, name, counter):
      threading.Thread.__init__(self)
      self.threadID = threadID
      self.name = name
      self.counter = counter
  def run(self):
    call('MP4Box -fps 30 -add '+self.name+'.h264 '+self.name+'.mp4', shell=True)
    call('rm '+self.name+'.h264', shell=True)

with picamera.PiCamera() as camera:
  camera.resolution = (1920, 1080)

  camera.framerate = 2
  camera.annotate_background = picamera.Color('black')
  camera.annotate_text_size = TextSize
  camera.annotate_frame_num = False
  camera.exposure_mode = 'night'
  camera.rotation = CamRotate

  thread_id = 1

  while run:
    fname = '/media/usb0/' + dt.datetime.now().strftime('%Y%m%d_%H%M')
    fext = '.h264'
    camera.start_recording(fname+fext, format='h264', quality=VideoQuality)

    n = 0

    while n < SecondsFile:
      Ax = acc_x
      Ay = acc_y
      Az = acc_z
      Mx = 1000.0 * mag_x
      My = 1000.0 * mag_y
      Mz = 1000.0 * mag_z
       
      astr = CamIdStr                                       \
        + dt.datetime.now().strftime(' %Y-%m-%d %H:%M:%S')  \
        + '  SHTR %d us' % camera.exposure_speed            \
        + '  AGain %f' % float(camera.analog_gain)          \
        + '  DGain %f\n' % float(camera.digital_gain)       \
        + '  Mx %6.0f nT' % Mx                              \
        + '  My %6.0f nT' % My                              \
        + '  Mz %6.0f nT' % Mz                              \
        + '  Ax %6.3f g' % Ax                               \
        + '  Ay %6.3f g' % Ay                               \
        + '  Az %6.3f g\n' % Az                             \
		+ '  T %6.3f C' % temp                              \
		+ '  rH %5.2f %%' % hum                             \
        + '  P %6.2f hPa' % press		
      print(astr)
      camera.annotate_text = astr
      #camera.wait_recording(1)

      n += 1

      if stick.wait(timeout=1):
        sk = stick.read()
        if sk.key == stick.KEY_ENTER:
          n = SecondsFile
          run = False

    camera.stop_recording()

    if run:
      # convert to mp4 using a seperate thread
      thread = myThread(thread_id, fname, thread_id)
      thread.start()
      thread_id = thread_id + 1

  camera.close()

  print("***** STOP *****")

roberttidey, 2016-01-10:
I am confused as to what is being asked for here.

It looks like you are trying to add a bunch of sensor data to the annotation text. However, the code shows picamera being used not RPi_Cam_Web_Interface.

Is that just an example and you are trying to do the same thing in RPi_Cam_Web_Interface?

degufo, 2016-01-10:
Yes. You are totally right. I would like to add the sensor data from the Raspberry Sense Hat to the RPI CAM Control by the annotation text feature under wheezy. 

I am not big on programing and so i thought i could get some help here.

The code above is a py prog that was written by an friend of mine where he makes short three seconds videos or pics with the sensor data shown also with the annotate text feature.

But i would like to use the features ov RPI CAM Control because it is a really really good program. I also need the motion demon and the website feature and the youtube kind library, so the way to go is only by RPI CAM control.

So if anybody can help me here, it would be really really cool.

best,

roberttidey, 2016-01-10:
I have added a facility to add arbitrary user annotation strings. This is done by writing the required text to a user_annotate.txt file in the RAM storage location /dev/shm/mjpeg.txt. A %z variable in th eannotation string reads this on every annotation generated so dynamic data will show up immediately.
The text in the file should be just the pure text to be substituted. (no line ending)

degufo, 2016-01-10:
Hi could you send me here the sample code? I also don´t have the mjeg.txt  - there is just a folder called "mjeg" and .tmpfs .

I also must add that i switched back to Wheezy - jessie was just not stable and didn´t worked really.

roberttidey, 2016-01-11:
I corrected a mistake in the comment for folder where you write.

To see a simple demo add %z on end of annotation string in the web interface then put following in annotate.py and run it.

filename = "/dev/shm/mjpeg/user_annotate.txt"
while True:
  ann = raw_input("Enter annotation (empty to end) : ")
  if (ann == "") : break
  annotate = open(filename, 'w')
  annotate.write(ann)
  annotate.close()
print "Annotate finished" 

Any text you type in is then put into annotation. You can now construct the string how you want before you write it.

I am using Jessie for all my cameras. Let me know if you try again and what the problem actually was.

degufo, 2016-01-15:
thanks you soooo much!

jebelcodes, 2016-01-17:
Hi,

I've been trying to make use of the %z annotation feature but I can't get anything to display, including with your test code above (cat /dev/shm/mjpeg/user_annotate.txt displays the output of the script successfully but it doesn't appear as an annotation on the image)

Any thoughts on why the interface isn't picking up the contents of the file?

Regards

wingers999, 2016-01-19:
Hi, same problem as jebelcodes above - have created file and put some test text in it - and added %z to annotation but nothing appears?

Any thoughts?

degufo, 2016-01-19:
How can i include in the annotation.text a script which generates the dynamic data?

I mean the dynamic data has to come frome somewhere, right? an a txt file is not an executable file.

wingers999, 2016-01-19:
you use another process to get the dymanic data and write it to that text file - i.e. I have a temperature sensor and a script runs every 20 mins, gets the temperature and the writes it to the user_annotate.txt file

degufo, 2016-01-19:
oh ok... so i write the dynamic data in the annotation txt file. the data is generated in the python script which automatically boots in the rc.local.

NICE!!!

But... this means that i can only have one single data string like temp or humidity.
But i have several data strings. The Raspberry Sense Hat has Accelerometer, Gyro, Magnetometer, Humidity, Temperature and Barometric Pressure.  Together these are about 12 values i want to integrate by the %z annotation in real time, which means about 20 each new values per second.

Plus there is the annotation limit in rpi cam control.
hm...

wingers999, 2016-01-19:
Well that's how I understand it anyway - and works for me

You can add more than one result to the txt file at a time - for example my sensor will do air pressure as well, so I can read results from sensors and then send both bits of info to the txt file for it to show

So suspect you can do the same with yours, if you amend your script for the sense hat to read all values and then send them to txt file in one long string i.e. "Temp = x Humidity = y etc"

degufo, 2016-01-19:
thats also how i understand it.

But whats about the refresh rate?

i need a high refresh rate, approx 20 times a second..

wingers999, 2016-01-19:
okay not sure about that then as I think annotation by default updates every 1 second -and for me I only read sensors every 20 minutes anyway so don't need it that often - think it may slow down system if doing it that often - but sure someone else will have better answer

Just need to actually get it working so it shows the %z annotation - hopefully Robert will reply sometime soon

degufo, 2016-01-19:
me too. i really need some sample code for the %z annotation where i can put the string. because my py script will make a new annotation.txt files at least each second.
Higher rates would be very appreciated.

Or do have you a some sample code for the annotation.txt file?

wingers999, 2016-01-19:
For mine I have simply added below lines to bottom of my temp.py script - read_temp contains the current temperature taken from sensor - so it just writes it to the user_annotate.txt file in the format of 

Temp = 25.57C

f = open("/dev/shm/mjpeg/user_annotate.txt", "w")
s = ("Temp = ") + str(read_temp()) + ("C")
f.write(s)
f.close()

roberttidey, 2016-01-22:
The annotation that appears on the screen is created for every frame. Normally this is the video fps so the preview is then rewritten at full video rate (e.g. 25 times per second) and the user text file will also be updated at that rate. There is a divider factor which can be used to slow down the preview rate e.g. divider 2 gives a preview at 12.5 fps. This is intended to lower the bandwidth needed for the preview.

wingers999, 2016-01-22:
Hi Robert, - could you look into why %z annotation doesn't appear - as per message above a few days ago from me and jebelcodes - have everything else I need working perfectly, just need to get it showing the text from user_annotate.txt in my annotation - done everything you say above, but it just doesn't show anything when I add %z to annotation string - cheers

roberttidey, 2016-01-23:
That's my fault. I originally had it as %z but changed it to %a and didn't update the docs.

Just change the %z to %a

wingers999, 2016-01-23:
Thanks Robert - all working perfectly now!!!

jebelcodes, 2016-01-25:
Can confirm %a works perfectly for me too, many thanks!