silvanmelchior/RPi_Cam_Web_Interface #232: Error in RaspiMJPEG popup all the time

vavalomi, 2016-01-10:
Hi,

It used to work before, but hadn't used this particular board for a while. Recently I got back to it, run all the updates (but don't think I have installed any other packages that would interfere). So now, even after many tries of reboots, uninstalls and installs, I can't seem to make it work. 
On a vanilla Jessie, camera only worked, as soon as I installed pi-pan, the error started popping up again. It seems the conflict is between servobluster and raspimjpeg modules but so far I wasn't able to pinpoint the exact reason. Any help or suggestion would be welcome!

roberttidey, 2016-01-10:
I need some more details on what is not working and what error pops up.

Are you using pipan or servoblaster?

vavalomi, 2016-01-10:
I'm sorry, should've used quotes, the popup said "Error in RaspiMKPEG"...
I'm using pipan, and last night was able to figure out what was wrong - the included servod file is the problem, after reading https://github.com/richardghirst/PiBits/issues/11 and servoblaster README, I downloaded and compiled the user space version and now everything works fine.

closed 2016-01-10 by vavalomi