silvanmelchior/RPi_Cam_Web_Interface #234: Installation Issue

brucegibbins, 2016-01-25:
Hello, I am very interested in your project and attempted to install. New to GIT and RPi etc. Followed instructions up to cloning from GIT where I am prompted for a UserName and Password. After creating a GIT account I entered these details but now get an error of "The requested URL returned error : 403 while accessing https://github.com/silvanmelchoir/RPi_Cam_Web_Interface.git/info/refs HTTP request failed. Was hoping you could shed some light on this as I can't see any other references in the Issues list. Thanks

roberttidey, 2016-01-25:
You shouldn't need a github account to clone the repository for installation. I haven't heard of anybody with that problem before. What OS are you using Wheezy, Jessie, Jessie Lite?

Note that you must enter the commands exactly as is. They are case sensitive.

Also you can test by putting
https://github.com/silvanmelchior/RPi_Cam_Web_Interface
in a browser which should take you to browser access to the repository.

brucegibbins, 2016-01-25:
Thanks Robert.

 

New to RPi and bought the basic starter kit. I know it is Raspbian and I see Wheezy mentioned during the APT-UPDATEs

 

I am not at device at the moment but 95% confident commands were keyed exactly including the case. It did not seem to have an issue locating the repository but it did prompt for a userid and password and seemed to validate my recently created github credentials. I know because I didn’t have a login prior to this and when blank credentials did not work I assumed I needed a valid git account to download your solution. 
 
I think that the url I entered can be checked by looking at the URL included in the error that was returned and I included in the original post. More specifically it seems to be going to the url …
 
https://github.com/silvanmelchoir/RPi_Cam_Web_Interface.git/info/refs
 
When I attempt to use a browser (out of curiosity) to go to the same URL I am presented with the following error which of course could be unrelated. Please keep in mind that I am doing this from my home pc out of curiosity as I saw your reply come in. 
 
Please upgrade your git client.

GitHub.com no longer supports git over dumb-http: https://github.com/blog/809-git-dumb-http-transport-to-be-turned-off-in-90-days

 

In the RPi itself I have run apt-update and apt-upgrade and all seemed to end successfully. So I am assuming that the git client should be the most recent on the Pi.

 

Thanks again for your time

 

https://github.com/silvanmelchior/RPi_Cam_Web_Interface

roberttidey, 2016-01-25:
Strange. I just tried browsing to the base URL https://github.com/silvanmelchoir/RPi_Cam_Web_Interface.git

from a platform that has no Git login or git client installed and I get through to the repository OK

If I add the /info/refs onto the URL then I do get the error you describe but that is not the URL one should be using.

So the first thing to find out is why info/refs is getting added.

brucegibbins, 2016-01-25:
Yes I agree and that was my experience as well as I performed the same test. As I am new to the specifics of GIT I thought that it was somehow following a dependancy tree and downloading related resource files or something similar. I have not adjusted any local GIT setting but maybe that is the issue

I will continue to research and let you know if I have any success. Has to be my environment by the sounds of it.

Thanks again for your input

Bruce

roberttidey, 2016-01-25:
I would suggest putting the standard OS on the card again from scratch and maybe use Jessie. When you tried to browse from a standard PC did you use just the basic URL or did you add the /info/refs on? I can't think of any reason why that environment would add it.

BNNorman, 2016-09-12:
The one point missed is the spelling of melchior - I made the same mistake and spelt it melchoir - what a numpty I was. Anyway, hope that helps others.