silvanmelchior/RPi_Cam_Web_Interface #235: updatePreview(cycle) always enables motion jpeg

tom-2015, 2016-01-31:
I noticed in both stream modes the motion jpeg is running. This caused the preview image to refresh very slowly on a slow 3G modem connection.

I changed:
`function updatePreview(cycle)
{
	if (mjpegmode && cycle !== undefined && cycle == true)
	{
		mjpeg_img.src = "/updating.jpg";
		setTimeout("mjpeg_img.src = \"cam_pic_new.php?time=\" + new Date().getTime()  + \"&pDelay=\" + preview_delay;", 1000);
		return;
	}
        if (previous_halted != halted)
        {
            if(!halted)
            {
                mjpeg_img.src = "cam_pic_new.php?time=" + new Date().getTime() + "&pDelay=" + preview_delay;			
            }
            else
            {
                mjpeg_img.src = "/unavailable.jpg";
            }
        previous_halted = halted;
    }
}`

to:

`function updatePreview(cycle)
{
	if (mjpegmode && cycle !== undefined && cycle == true)
	{
		mjpeg_img.src = "/updating.jpg";
		setTimeout("mjpeg_img.src = \"cam_pic_new.php?time=\" + new Date().getTime()  + \"&pDelay=\" + preview_delay;", 1000);
		return;
	}
    **if (mjpeg_mode){**
        if (previous_halted != halted)
        {
            if(!halted)
            {
                mjpeg_img.src = "cam_pic_new.php?time=" + new Date().getTime() + "&pDelay=" + preview_delay;			
            }
            else
            {
                mjpeg_img.src = "/unavailable.jpg";
            }
        }
        previous_halted = halted;
    **}**

Now I get +/- 1 frames/sec instead of 1 frames/min.

roberttidey, 2016-01-31:
Not sure I understand that as the img.src which gets the data is either linked to the jpeg images or the mjpeg stream.

The update_preview though is only really used in mjpeg stream mode to allow the mjpeg stream server end to be adjusted. So I have put a mjpegmode condition around the whole function now.

In your code above you have mjpeg_mode rather than mjpegmode  which would be an undefined var.

If you have any problems with updated version then let me know and which browser / device / mlpeg mode you are using. Some browsers don't work too well on Apple Safari with mjpeg streaming which is why the default jpeg fetching is left in. It doesn't actually use much more bandwidth.

Recent checks on jpeg quality factor show raspberry has a very non-linear Q compression. This means the default Q has been lowered to 10 (equivalent to normal 75). If you haven't changed this already then I would lower preview quality to 10 to lower bandwidth usage three times compared to 25.

tom-2015, 2016-01-31:
Sorry you are correct it must be mjpegmode NOT mjpeg_mode.
For some reason when I pressed F12 there was 1 non ending request to cam_pic_new.php and every 300~500ms a request to cam_pic.php and/or status_mjpeg.php

roberttidey, 2016-01-31:
If you use mjpegmode then there is one request that basically just streams mjpeg continuously so the connection is left open. If you use the default mode then it will request separate images one by one. The rate will be determined by the divider ratio but if the connection is slow then that will also slow it down.

Previously there could have been a request to the new.php even in default mode but it shouldn't have been streaming as the img.src was set to request the jpegs. With the recent change it won't make a call to new.php in default mode.

tom-2015, 2016-02-01:
Yes it probably only makes one call to the new.php but if you set the src to an mjpeg stream in Firefox, it will continue to stream it also after you changed the src back to a normal jpeg picture. This is probably what happend.