silvanmelchior/RPi_Cam_Web_Interface #236: Watchdog detected a problem - need more information (debug)

bossu2005, 2016-02-04:
Everything is running fine then every now and then and quite unpredictably a Watchdog detected problem... 

{2016/02/04 07:05:50} Watchdog detected problem. Stopping{2016/02/04 07:05:52} SIGINT/SIGTERM received, stopping

Unfortunately the message does not provide any information as to the nature of the problem.

Is there a way to have it produce some debugging output? (command line option to raspimjpeg perhaps?)

This eventually restarts the ?scheduler? loop and blows out any running timelapse.

It seems this can happen a few times a day or just once a day.  No pattern observed, 
not close to or tied to day period changes or management scheduler intervals either.  Completely
random it seems...

I have multiple RPI's with the same configuration and all of them are having this happen 
at some point daily.   I've checked free SD disk space and there is no issue there.

In addition just after the failure. there are then about 1 and 1/2 minutes of these
entries in the scheduleLog.txt.   

[2016/02/04 07:06:04] Autocapture request.
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet

Here are the log entries just before the start of the problem....

{2016/02/04 07:04:09} Capturing image
{2016/02/04 07:04:40} Capturing image
{2016/02/04 07:05:10} Capturing image
[2016/02/04 07:05:33] Maximum Capture reached. Sending off command
[2016/02/04 07:05:33] Send ca 0
{2016/02/04 07:05:33} Capturing stopped
{2016/02/04 07:05:33} Add /var/www/html/media/vi_10119_20160204_070336.mp4 to Box Queue at pos 18
{2016/02/04 07:05:33} Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/media/vi_10119_20160204_070336.mp4.h264"
{2016/02/04 07:05:34} Start boxing /var/www/html/media/vi_10119_20160204_070336.mp4.h264 to /var/www/html/media/vi_10119_20160204_070336.mp4 Queue pos 18
[2016/02/04 07:05:37] Autocapture request.
[2016/02/04 07:05:37] Start triggered by autocapture
[2016/02/04 07:05:37] Send ca 1
{2016/02/04 07:05:37} Executing macro /var/www/html/macros/start_vid.sh "/var/www/html/media/vi_10120_20160204_070537.mp4" &
{2016/02/04 07:05:37} Capturing started
{2016/02/04 07:05:40} Capturing image

The Watchdog problem

{2016/02/04 07:05:50} Watchdog detected problem. Stopping{2016/02/04 07:05:52} SIGINT/SIGTERM received, stopping
{2016/02/04 07:06:02} RaspiMJPEG Version 5.4.1
{2016/02/04 07:06:03} recommended video buffers 1
{2016/02/04 07:06:03} h264 buffers set to recommended 1
{2016/02/04 07:06:03} Set up internal detect width=81 height=45
{2016/02/04 07:06:03} MJPEG streaming, ready to receive commands
{2016/02/04 07:06:03} send smd 9
{2016/02/04 07:06:03} Starting command loop
[2016/02/04 07:06:03] Motion pipe read =[9]
[2016/02/04 07:06:03] Reload parameters command requested
[2016/02/04 07:06:03] Scheduler loop is started
{2016/02/04 07:06:03} send smd 1
[2016/02/04 07:06:03] Motion pipe read =[1]
[2016/02/04 07:06:03] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Autocapture request.
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet
[2016/02/04 07:06:04] Start capture request ignored, day period not initialised yet

roberttidey, 2016-02-05:
The watchdog is intended to restart the main raspimjpeg process if it stops working correctly. It does this by having a second copy of the raspimjpeg whose main job is to see if the main raspimjpeg is running and if not to start another one. The main one is checking itself to see if the preview is being generated OK and will self terminate if it detects a problem, and at this point another one is started.

The watchdog set up is under camera settings. It will self terminate if more than the error count is encountered in the interval.

So there are 2 possibilities here. Either you are getting genuine errors and the watchdog is kicking in, or the watchdog itself is seeing errors when there aren't actually any.

To distinguish between these then set the watchdog time to 0 which disables it. You won't get watchdog errors but you might get genuine camera errors if that is the cause.

The day period initialised is a separate scheduler issue I think triggered by getting a capture request before it has really got going. I'll look into that separately.

bossu2005, 2016-02-05:
Yes, I do have autocapture on - basically on a constant basis saving a new recorded segment every 
2 minutes.... 

The period commands which are partially shown in the screenshot2 are as follows
FOR NIGHT md 0;px 1280 0720 01 01 2592 1944;em night;
FOR DAWN  md 0;px 1280 0720 01 01 2592 1944;em night;
FOR DAY     md 0;px 1280 0720 25 25 2592 1944;em auto;md 1;
FOR DUSK   md 0;px 1280 0720 01 01 2592 1944;em night;

In essence record standard 25 fps during the day with motion detection on.
and all other periods 1 fps (so it can pick up some things in the dark) with no motion detection.

Motion detection would seem useless with these settings,  but I have the modified the motion detection code in the scheduler to flag the current recording file as having motion detected and
the download page displaying how many motion events occurred during that recorded segment and  - otherwise the motion detection does sometimes trigger its own
capture during that 5 second gap between the end of each recording (max 115 secs) to the next
scheduled autocapture (120 seconds)

The watchdog errors do not seem to have any correlation with any motion events as 2 of the RPIs 
practically never detect any motion and they too are experiencing the same amount of random watchdog failures.

I only occasionally start a timelapse recording and this is what brought the issue to my attention
since they stopped without me manually stopping them and since a day period change had not 
occurred I then examined the log spotting the watchdog error.

I will try the watchdog time set to 0 and see what happens...

Thank You.

[image]
[image]

roberttidey, 2016-02-05:
It's a bit unusual to use AutoCapture an motion detect together which is why scheduler might get confused.. AutoCapture was really designed to do a regular repeated recording as an alternative to motion detect.

The slow frame rate during dark recordings may be contributing to the watchdog firing as it is dependent on the checking the latest preview image time. It would probably want the watchdog time increased to reduce the risk.

bossu2005, 2016-02-05:
Understood, yes the idea here was that motion detection not being perfect, I wanted a continuous recording but not so long that it would be too big to be downloadable, but still be able to quickly "pick out the good stuff" with the motion detection.

Ok - will see what happens with Watchdog disabled - the watchdog errors have occurred during day period so it may be something more basic as in camera hardware errors or even power fluctuations.

Nothing else is running on the RPIs - no X desktop - boots to command console.

bossu2005, 2016-02-05:
Followup Update:  

Just after 9:30 AM, I updated the watchdog_interval to 0 on each RPI and completely rebooted each
RPI 

After waiting 5 minutes after each rebooted , I then started a timelapse (30 second interval) on each RPI.   

I just checked each RPI and the watchdog errors no longer appear, but in the case of each RPI , the timelapse stopped working - one within 20 minutes another just less than 1 hour , another just over 1 hour and the other over 3 hours.

Also, I did a pstree on each and each shows 1 master raspimjpeg running and 1 child raspimjpeg still running.   Video recording has continued without interruption on each RPI. 

RPI 1

{2016/02/05 09:48:35} Capturing image
{2016/02/05 09:49:06} Capturing image
{2016/02/05 09:49:36} Capturing image
{2016/02/05 09:50:06} Capturing image
[2016/02/05 09:50:08] Maximum Capture reached. Sending off command
[2016/02/05 09:50:09] Send ca 0
{2016/02/05 09:50:09} Capturing stopped
{2016/02/05 09:50:09} Add /var/www/html/media/vi_4490_20160205_094811.mp4 to Box Queu
e at pos 9
{2016/02/05 09:50:09} Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/
media/vi_4490_20160205_094811.mp4.h264"  
{2016/02/05 09:50:09} Start boxing /var/www/html/media/vi_4490_20160205_094811.mp4.h2
64 to /var/www/html/media/vi_4490_20160205_094811.mp4 Queue pos 9
[2016/02/05 09:50:12] Autocapture request.
[2016/02/05 09:50:12] Start triggered by autocapture
[2016/02/05 09:50:12] Send ca 1
{2016/02/05 09:50:12} Executing macro /var/www/html/macros/start_vid.sh "/var/www/htm
l/media/vi_4491_20160205_095012.mp4" &
{2016/02/05 09:50:12} Capturing started
{2016/02/05 09:50:21} Finished boxing /var/www/html/media/vi_4490_20160205_094811.mp4
 from Box Queue at pos 9
{2016/02/05 09:50:21} Executing macro /var/www/html/macros/end_box.sh "/var/www/html/
media/vi_4490_20160205_094811.mp4" &
{2016/02/05 09:50:21} Removed item from Box Queue
[2016/02/05 09:52:09] Maximum Capture reached. Sending off command
[2016/02/05 09:52:09] Send ca 0
{2016/02/05 09:52:09} Capturing stopped
{2016/02/05 09:52:09} Add /var/www/html/media/vi_4491_20160205_095012.mp4 to Box Queu
e at pos 10
{2016/02/05 09:52:09} Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/
media/vi_4491_20160205_095012.mp4.h264"  
{2016/02/05 09:52:09} Start boxing /var/www/html/media/vi_4491_20160205_095012.mp4.h2
64 to /var/www/html/media/vi_4491_20160205_095012.mp4 Queue pos 10
[2016/02/05 09:52:13] Autocapture request.

RPI 2

{2016/02/05 12:43:59} Capturing image
{2016/02/05 12:44:29} Capturing image
{2016/02/05 12:44:59} Capturing image
[2016/02/05 12:45:25] Maximum Capture reached. Sending off command
[2016/02/05 12:45:25] Send ca 0
{2016/02/05 12:45:25} Capturing stopped
{2016/02/05 12:45:25} Add /var/www/html/media/vi_4409_20160205_124328.mp4 to Box Queue at pos 12
{2016/02/05 12:45:25} Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/media/vi_4409_20160205_124328.mp4.h264"  
{2016/02/05 12:45:25} Start boxing /var/www/html/media/vi_4409_20160205_124328.mp4.h264 to /var/www/html/media/vi_4409_20160205_124328.mp4 Queue pos 12
[2016/02/05 12:45:27] Autocapture request.
{2016/02/05 12:45:29} Capturing image
[2016/02/05 12:45:32] Start triggered by autocapture
[2016/02/05 12:45:32] Send ca 1
{2016/02/05 12:45:32} Executing macro /var/www/html/macros/start_vid.sh "/var/www/html/media/vi_4410_20160205_124532.mp4" &
{2016/02/05 12:45:32} Capturing started
{2016/02/05 12:45:33} Finished boxing /var/www/html/media/vi_4409_20160205_124328.mp4 from Box Queue at pos 12
{2016/02/05 12:45:33} Executing macro /var/www/html/macros/end_box.sh "/var/www/html/media/vi_4409_20160205_124328.mp4" &
{2016/02/05 12:45:33} Removed item from Box Queue
[2016/02/05 12:47:28] Autocapture request.
[2016/02/05 12:47:28] Start triggered by autocapture
[... 33 lines cut ...]
{2016/02/05 10:14:49} Capturing image
{2016/02/05 10:15:19} Capturing image
{2016/02/05 10:15:50} Capturing image
[2016/02/05 10:16:03] Maximum Capture reached. Sending off command
[2016/02/05 10:16:03] Send ca 0
{2016/02/05 10:16:03} Capturing stopped
{2016/02/05 10:16:03} Add /var/www/html/media/vi_11216_20160205_101205.mp4 to Box 
Queue at pos 25
{2016/02/05 10:16:03} Executing macro /var/www/html/macros/end_vid.sh "/var/www/ht
ml/media/vi_11216_20160205_101205.mp4.h264"  
{2016/02/05 10:16:03} Start boxing /var/www/html/media/vi_11216_20160205_101205.mp
4.h264 to /var/www/html/media/vi_11216_20160205_101205.mp4 Queue pos 25
[2016/02/05 10:16:07] Autocapture request.
[2016/02/05 10:16:08] Start triggered by autocapture
[2016/02/05 10:16:08] Send ca 1
{2016/02/05 10:16:08} Executing macro /var/www/html/macros/start_vid.sh "/var/www/html/media/vi_11217_20160205_101608.mp4" &
{2016/02/05 10:16:08} Capturing started
{2016/02/05 10:16:20} Capturing image
{2016/02/05 10:16:32} Finished boxing /var/www/html/media/vi_11216_20160205_101205.mp4 from Box Queue at pos 25
{2016/02/05 10:16:39} Executing macro /var/www/html/macros/end_box.sh "/var/www/html/media/vi_11216_20160205_101205.mp4" &
{2016/02/05 10:16:39} Removed item from Box Queue
[2016/02/05 10:18:05] Maximum Capture reached. Sending off command
[2016/02/05 10:18:05] Send ca 0
{2016/02/05 10:18:05} Capturing stopped
{2016/02/05 10:18:05} Add /var/www/html/media/vi_11217_20160205_101608.mp4 to Box Queue

roberttidey, 2016-02-06:
I am checking into why the time lapse would stop like that.

If you get it into that state again then it would be interesting to know
1) What do the buttons look like? In particular what does the time lapse button say?
2) If it says stop can you stop it and try to take a still image capture. Does it take a picture
3) cat the status_mjpeg.txt in the web folder

bossu2005, 2016-02-06:
You may have something there...
1) all of the buttons are dimmed!  even though video is still recording and I've verified that the recorded files do play and look good and motion detection is showing in the logs as running...
2) Nothing can be stopped since all buttons dimmed
3) status_mjpeg.txt has only "image" as content despite video recording and motion detection still working ...

Normally when video recording and motion detect is on it contains "md_video" and when timelapse is running (and working) too with them it contains "tl_md_video"

perhaps something is getting out of whack with the timelapse autocapture happening between the autocapture stops (max out) and autocapture starts??? - you may be able to replicate it with the settings i have 115 for max_capture and 120 for autocapture_Interval

roberttidey, 2016-02-06:
I think I might know what is happening here. When an image is taken either by timelapse or manually then a flag is set to indicate that an image capture has started and the status is set to image. This flag gets reset by a callback function from the underlying camera MMAL layer when all the data for the image is available and the status then returns to its prior state.

If this callback never happens then this state will never get reset. I don't know why the callback never happens but it could be because you have several activities going on in parallel with your set up. I haven't heard of this problem before.

What I propose to do is to put a timeout on this state so that it does exit from this condition. That particular image will probably be damaged or missing but at least operation will continue as normal.

I'll check further into why the callback could go missing but that might be outside of my control.

roberttidey, 2016-02-06:
Latest version now has a 5 second timeout on image capture operations. If this happens it should log a message
 {2016/02/06 11:19:27} Image capture timed out /var/www/media/im_0013_20160206_111923.jpg

It will also attempt to run the soft error macro if defined.

bossu2005, 2016-02-06:
Sounds reasonable.   I have downloaded the new raspimjpeg and installed on each RPI, re-enabled the watchdog and rebooted, and started a timelapse.   Will see what happens.     Thank You!

bossu2005, 2016-02-06:
With video recording with motion detection and timelapse running it took about 2 hours then
an image timeout message appeared.

{2016/02/06 08:49:35} Image capture timed out hJG    (see log below) 

No Watchdog error... I double checked that watchdog was actually enabled.. watchdog_interval 30

raspimjpeg reinitializes and the scheduler loop restarting.
Afterwards status shows "md_video" ... no timelapse running.

{2016/02/06 08:47:26} Capturing image
{2016/02/06 08:47:56} Capturing image
{2016/02/06 08:48:26} Capturing image
{2016/02/06 08:48:56} Capturing image
[2016/02/06 08:48:58] Maximum Capture reached. Sending off command
[2016/02/06 08:48:58] Send ca 0
{2016/02/06 08:48:58} Capturing stopped
{2016/02/06 08:48:58} Add /var/www/html/media/vi_14300_20160206_084701.mp4 to Box Que
ue at pos 19
{2016/02/06 08:48:58} Executing macro /var/www/html/macros/end_vid.sh "/var/www/html/
media/vi_14300_20160206_084701.mp4.h264"  
{2016/02/06 08:48:59} Start boxing /var/www/html/media/vi_14300_20160206_084701.mp4.h
264 to /var/www/html/media/vi_14300_20160206_084701.mp4 Queue pos 19
[2016/02/06 08:49:01] Autocapture request.
[2016/02/06 08:49:01] Start triggered by autocapture
[2016/02/06 08:49:01] Send ca 1
{2016/02/06 08:49:01} Executing macro /var/www/html/macros/start_vid.sh "/var/www/htm
l/media/vi_14301_20160206_084901.mp4" &
{2016/02/06 08:49:01} Capturing started
{2016/02/06 08:49:17} Finished boxing /var/www/html/media/vi_14300_20160206_084701.mp
4 from Box Queue at pos 19
{2016/02/06 08:49:17} Executing macro /var/www/html/macros/end_box.sh "/var/www/html/
media/vi_14300_20160206_084701.mp4" &
{2016/02/06 08:49:17} Removed item from Box Queue
{2016/02/06 08:49:26} Capturing image
{2016/02/06 08:49:35} Image capture timed out hJG
{2016/02/06 08:49:40} RaspiMJPEG Version 5.4.2
{2016/02/06 08:49:40} recommended video buffers 1
{2016/02/06 08:49:40} h264 buffers set to recommended 1
{2016/02/06 08:49:40} Set up internal detect width=81 height=45
{2016/02/06 08:49:40} MJPEG streaming, ready to receive commands
{2016/02/06 08:49:40} send smd 9
{2016/02/06 08:49:40} Starting command loop
[2016/02/06 08:49:40] Motion pipe read =[9]
[2016/02/06 08:49:40] Reload parameters command requested
[2016/02/06 08:49:40] Scheduler loop is started
[2016/02/06 08:49:41] Scheduled management tasks. Next at 1454770181
[2016/02/06 08:49:41] PurgeHours: video 0 image 31 lapse 30
[2016/02/06 08:49:41]  free space purge total 30642996 current: 1216184 target: 3145728
[2016/02/06 08:49:41] Finished. Current now: 3252988.3867188
[2016/02/06 08:49:41] Purged 25 Files
[2016/02/06 08:49:41] Autocapture request.
[2016/02/06 08:49:41] Start capture request ignored, day period not initialised yet
[2016/02/06 08:49:41] Start capture request ignored, day period not initialised yet
{2016/02/06 08:49:41} send smd 1
[2016/02/06 08:49:41] Motion pipe read =[1]
[2016/02/06 08:49:41] Start capture request ignored, day period not initialised yet
[2016/02/06 08:49:41] Start capture request ignored, day period not initialised yet
.......many of these for a minute or 2 
.......then
[2016/02/06 08:50:58] Maximum Capture reached. Sending off command
[2016/02/06 08:50:58] Send ca 0
{2016/02/06 08:50:58} Already stopped. Ignore 
[2016/02/06 08:51:00] New period detected 3
[2016/02/06 08:51:00] Send md 0
{2016/02/06 08:51:00} Internal motion detection stopped
{2016/02/06 08:51:00} recommended video buffers 1
{2016/02/06 08:51:00} h264 buffers set to recommended 1
{2016/02/06 08:51:00} Set up internal detect width=81 height=45
[2016/02/06 08:51:02] Send px 1280 0720 25 25 2592 1944
{2016/02/06 08:51:02} Change: video_width = 1280
{2016/02/06 08:51:02} Change: video_height = 0720
{2016/02/06 08:51:02} Change: video_fps = 25
{2016/02/06 08:51:02} Change: MP4Box_fps = 25
{2016/02/06 08:51:02} Change: image_width = 2592
{2016/02/06 08:51:02} Change: image_height = 1944
{2016/02/06 08:51:02} recommended video buffers 1
{2016/02/06 08:51:02} h264 buffers set to recommended 1
{2016/02/06 08:51:02} Set up internal detect width=81 height=45
[2016/02/06 08:51:04] Send em auto
{2016/02/06 08:51:04} Change: exposure_mode = auto
[2016/02/06 08:51:06] Send md 1
{2016/02/06 08:51:06} Internal motion detection started
{2016/02/06 08:51:06} recommended video buffers 1
{2016/02/06 08:51:06} h264 buffers set to recommended 1
{2016/02/06 08:51:06} Set up internal detect width=81 height=45
{2016/02/06 08:51:07} send smd 1
[2016/02/06 08:51:08] Motion pipe read =[1]
[2016/02/06 08:51:08] Start capture requested from Pipe

konstantinj, 2016-02-15:
Hi,

I think I have the same or at least a similar issue. Sometimes the camera stop working when the watchdog kills the process for whatever reason. This is an example log:

> [2016/02/15 12:20:38] Send md 1
> {2016/02/15 12:22:21} Finished boxing /var/www/media/vi_0399_20160215_121640.mp4 from Box Queue at pos 5
> {2016/02/15 12:22:21} Removed item from Box Queue
> {2016/02/15 12:22:26} send smd 1
> [2016/02/15 12:22:26] Start capture requested from Pipe
> [2016/02/15 12:22:26] Send ca 1
> {2016/02/15 12:22:26} Missing macro definition
> {2016/02/15 12:22:26} Capturing started
> {2016/02/15 12:26:19} Watchdog detected problem. Stopping{2016/02/15 12:26:19} SIGINT/SIGTERM received, stopping
> [2016/02/15 12:27:28] Maximum Capture reached. Sending off command
> [2016/02/15 12:27:28] Send ca 0
> {2016/02/15 13:17:22} RaspiMJPEG Version 5.4.2
> {2016/02/15 13:17:22} recommended video buffers 1
> {2016/02/15 13:17:22} h264 buffers set to recommended 1
> {2016/02/15 13:17:22} Set up internal detect width=82 height=61
> {2016/02/15 13:17:22} MJPEG streaming, ready to receive commands
> {2016/02/15 13:17:22} send smd 9
> {2016/02/15 13:17:23} DEBUG 3
> [2016/02/15 13:18:54] RaspiCam support started
> [2016/02/15 13:18:54] Capture Pipe already exists /var/www/FIFO1
> [2016/02/15 13:18:54] Scheduler loop is started
> {2016/02/15 13:18:54} Starting command loop
> [2016/02/15 13:18:54] Reload parameters command requested
> [2016/02/15 13:18:54] Scheduler loop is started
> {2016/02/15 13:18:54} send smd 1
> [2016/02/15 13:18:54] Start capture request ignored, day period not initialised yet
> [2016/02/15 13:18:55] New period detected 0
> [2016/02/15 13:18:55] Scheduled management tasks. Next at 1455542335

The camera was gone for almost an hour. I'm also using multiple pies and all of them have this problem once a day.

I've now disabled the watchdog to see what happens.

Btw: What is "smd"?

roberttidey, 2016-02-15:
smd is just logging of commands sent from raspimjpeg to the scheduler;smd 1 when internal motion detected a start, smd 0 when internal motion detected a stop, smd 9 is used to tell scheduler to reload its settings and get into synce when raspimjpeg starts up.

Try turning watchdog off for a while (watchdog time = 0) and see what happens.

konstantinj, 2016-03-05:
I've tried different settings now but it's always the same. Without the watchdog it just crashes without notice and I have to restart manually. Still working on getting sound together with video and stumbled upon this: https://www.raspberrypi.org/forums/viewtopic.php?t=43738&p=348942
May xfs will also solve this issue.

konstantinj, 2016-03-27:
Hi, I don't have this issue anymore since I've started using my own macros: https://github.com/konstantinj/RPi_Cam_Web_Interface
I think the reason is that I now use a tmpfs drive for the boxing part. I've just mounted /tmp as tmpfs, do the boxing via macro there and move the file afterwards back to the media folder. So maybe this is an issue related to the sd card. Just fyi.