silvanmelchior/RPi_Cam_Web_Interface #238: Web interface does not open properly after password installed

xray125, 2016-02-15:
RPI Cam version 6.0.24
Apache 2.4.10
Raspberry Pi running Jessie
Installed RPI Cam as per instructions and works OK.
Ran the install.sh to install a user and password.
When viewing webpage it did not ask for a password and showed a webpage with 
[image]
Resolved this issue by editing /etc/apache2/sites-available/raspicam.conf
Changed from DocumentRoot /var/www
to DocumentRoot /var/www/html
Now when loading webpage I got a server error.
Looking at the /var/log/apache2/error.log
Invalid command "AuthGroupFile' , perhaps misspelled or defined by a module not included in the server configuration
Edited /var/www/html/.htaccess
Removed the line AuthGroupFile line
Now is working with Password popup OK

I believe this issue is due to the new default location of Apache2 and not all the changes have been implemented in RPI Cam
I am fairly new to this.  Can anyone suggest a way to make sure the developers are aware of this but other than posting it here?

roberttidey, 2016-02-15:
This is the right place to raise issues like this.

The problem here is that the AuthGroupFile directive is used by the authz_groupfile_module which is now not being loaded by the default Apache config. It provides a facility to log on by group name.

I have removed it for the moment as it is not really that useful and would require creating a group file anyway.

miraaz, 2016-03-03:
With Older install file that issue is fixed now. After merge done.

torstenhaase, 2016-03-20:
Hi, I still get the same error. I uninstalled an reinstalled it again but still no change. Password doesn't work at all and even after doing the changes in the first post it doesn't work. Thanks for your help.

roberttidey, 2016-03-20:
The AuthGroupFile issue was fixed. 
Tell us the following;
Which OS Wheezy, Jessie?
Which version RPiCam (latest is 6.0.35)?
Which install method RPiCam or install.sh?
Which web server Apache or nginx?
Subfolder being used?
Behaviour after entering password - No request to login?, web site shows anyway?

torstenhaase, 2016-03-20:
Hi Robert,

I have Raspian, don't know about Jessie. How can I find out?
RPICam is version 6.0.35, I used the install.sh and it is installed in the folder "html". I only enter a user and password during install but didn't change any other parameters.
Webserver is Apache (Version: Apache/2.2.22 (Debian) Server at 192.168.0.17 Port 80)

Thanks for your help
Torsten

roberttidey, 2016-03-20:
You can tell what rasbian is installed by using cat /etc/debian_version v7 is wheezy v8 is jessie

If you installed in html then you should be browsing to http://cameraip/html 

The first post in this issue tried changing Apache config for this so that you browse to http://cameraip Although this sounds convenient, I don't recommend it. You will break any username password stuff without further fixes. Plus the whole point about installing into a subfolder is to allow installing other web sites for Apache and browsing to those. E.g. http://camerip/html for RPiCamera and http://cameraip/othersite for a second web site

If you don't want other websites and you want to browse to http://cameraip  then install with subfolder blanked out and it will install at top level (/var/www) which is fine

torstenhaase, 2016-03-21:
Hi Robert, I have version 7.9 installed. I updated again and installed RPICam without a subfolder. Unfortunately the user/password still doesn't work.

Thanks for your help
Torsten

torstenhaase, 2016-03-30:
Hi Robert, an update from my side. I used the old installation script and it worked.