silvanmelchior/RPi_Cam_Web_Interface #239: Using PI Camera during Livestream

bennywoochi, 2016-02-25:
Hi,
I want to make a video during the live stream via an SSH command over the network. I want to start a video from my Smartphone via an SSH command. I wrote an App for this purpose. 
I knwo that you can make a video via the website, but I want to do this from the Android App.
Can you tell me in which script do you have the command that you can make a video from the website, so I can adjust it for my purpose.

roberttidey, 2016-02-26:
All operations are controlled by sending commands into the FIFO pipe which exists in the web install folder. The main raspimjpeg process is monitoring this and performing the commands. You can see a reference to all the commands either in the wiki or on th escheduler web page.

So from SSH you can just do echo -n 'ca 1' >/var/www/install/FIFO  which will start a video capture and echo -n 'ca 0' >/var/www/install/FIFO which will stop it.

If you want a script that will capture a timed length video then just put these 2 commands in a bash script separated by a delay (to time the video, and then call the bash script from SSH.

bennywoochi, 2016-02-26:
Thank you very much roberttidey. That works great ;). Can you also tell me how can I do the same with taking a picture?
Thanks!

roberttidey, 2016-02-26:
echo -n 'im' >/var/www/install/FIFO will take a 1 shot image
echo -n 'tl 1' >/var/www/install/FIFO will take start a time lapse sequence
echo -n 'tl 0' >/var/www/install/FIFO will take stop a time lapse sequence
echo -n 'tv 50' >/var/www/install/FIFO sets time lapse interval in 100mSec units

Check the docs out for other commands for camera settings, resolutions etc.

bennywoochi, 2016-02-28:
Thank you its working, too ;)