silvanmelchior/RPi_Cam_Web_Interface #240: Buttons not working

runeste, 2016-02-27:
Running Raspbian Jessie, and installing in sub-directory, but cannot for the love of god get the buttons to work. Nothing happens when pushing them. status file link is ok, all files and directories are owned by www-data. I see no request or errors in log from pushing the buttons.

Fresh install gives me the exact same problem.

roberttidey, 2016-02-27:
What state are the buttons in? Are they fully coloured or dimmed? Does status file have 'ready' in it?

If that all looks OK then easiest way to discover what is going on is to turn on developer tools in the browser. E.g. in Chrome CTRL+SHIFT+I then click on network tab. You will see a regular fetch of the cam_pic.php which is fetching the preview plus every few seconds a status_mjpeg call with a response like ready.

You can get rid of the regular cam_pic.php fetch by selecting mjpeg stream instead of default stream from the system link then you will just see the status fetch.

Now try pressing the record image button. You should see cmd_pipe with a command of im.

If you don't see the command then something is wrong with the web side set up. Maybe browser settings. Try clearing all browser data.

If you do see the command but nothing happens then something is wrong with FIFO set up. Try doing a manual command from a terminal window 

echo -n 'im' >/var/www/subfolder/FIFO

This should result in an image being captured.

runeste, 2016-02-27:
Hi,

Thanks for the ideas, I can indeed see the im command pipe, but nothing is happening. Upon echo'ing the FIFO, the command is stuck in run-mode, and nothing happens. I have to CTRL+C to end it. Is there an overview of permissions, or should all files in the directory be www-data owned ?

runeste, 2016-02-27:
Ok, news:  I checked the FIFO issue threads here, and its defo a FIFO issue. I disabled the motion_pipe parameter, and now I can record as normal, and buttons are working.

Im gonna go watch the FIFO thread, and this one can be closed.

Thank you again for the input !

closed 2016-02-27 by runeste

roberttidey, 2016-02-27:
OK Good to hear it is working now.

Disabling motion_pipe shouldn't be necessary with normal installs. The raspimjpeg process writes to another pipe (FIFO1) to send things like resets and motion triggers to the scheduler. Providing the scheduler is being used and is up and running then that is OK. If scheduler is not being used for some reason then it is necessary to disable raspimjpeg from writing to FIFO1 by disabling motion_pipe