silvanmelchior/RPi_Cam_Web_Interface #250: Source of RaspiMJPEG

konstantinj, 2016-03-09:
Hi,

I'm looking for the current source of RaspiMJPEG.c. I've currently do audio recording with the bash macros but I think it would be much prettier directly in the c file, at least for me. I've found https://github.com/silvanmelchior/userland/blob/master/host_applications/linux/apps/raspicam/RaspiMJPEG.c but I guess this is not the latest version since I can't find config values like "boxing_path".

konstantinj, 2016-03-10:
Currently I have the issue that no thumbnails are generated when using an alternative video_path. It would really help, having the sources.

konstantinj, 2016-03-10:
I guess this could be done in C more elegant: https://github.com/konstantinj/RPi_Cam_Web_Interface/tree/master/www/macros

roberttidey, 2016-03-11:
In the wiki there is a link to the current sources

roberttidey, 2016-03-11:
There have been several requests to include audio support. This can come in two ways; trigger a separate recording or embed in the video. You are right they would be best handled within the c side. Embed could be done during the mp4 conversion. The big problem is it is quite difficult to get any decent sync between the audio and the video particularly if the video buffering is used which can introduce several seconds of ambiguity

konstantinj, 2016-03-11:
Hi @roberttidey found the source: https://github.com/roberttidey/userland/tree/master/host_applications/linux/apps/raspicam
Was looking at @silvanmelchior 's but that's a bit older.

My solution works actually quite good with the bash macros. I'm using it for my babycam so I don't miss my sons first words. Sync is ok.

I'll do it in C when I have time.

czarina, 2017-01-07:
@konstantinj did you ever write a solution for audio+video syncing in C?

konstantinj, 2017-01-09:
@czarina Unfortunately not. Since my son is getting older I never have time :)

VOIR-PT, 2024-02-03:
@konstantinj  7 years later: You should have tine now. :)