silvanmelchior/RPi_Cam_Web_Interface #254: Change motion related arguments with DayMode period

a-l-e-x-d-s-9, 2016-03-12:
I'd like to change the values of the arguments: motion_threshold, motion_noise, motion_startframes and motion_stopframes, in the "Period Start" for "Sun based" mode.
I don't see anything that would help me in "Command reference". Maybe it can be done with macros?
Can you suggest please, how it can be done?

roberttidey, 2016-03-18:
There are commands for all these, but got missed out the docs.
vp 0/1 vector_preview
mn N motion_noise
mt N motion_threshold
mi filename motion_image
mb N motion_stopframes
me N motion_startframes

a-l-e-x-d-s-9, 2016-03-21:
I tested the suggested commands on 6.0.35.
Only mn (motion_noise) is working when I try to set: "mn 1006; mt 10; me 4; mb 100;" in "Period Start", like this:
[image]

a-l-e-x-d-s-9, 2016-03-21:
The rest remain unchanged.
Any ideas what's wrong?

a-l-e-x-d-s-9, 2016-03-21:
This 2 also has no effect: "em night; wb flash;".
:-(

roberttidey, 2016-03-21:
Seems to be working for me OK. Are you pressing Save Settings after making a change to Scheduler set up?

When parameters are changed they will appear in the log and they also get persisted in the uconfig file in the web folder (e.g./var/www). If you nano /var/www/uconfig you should see any parameters changed from the default.

After making changes the current values should show up on the main page when it is next loaded.

You can also try manually issuing the commands like echo -n 'mb 100' >/var/www/FIFO  which should do the same thing as scheduler does.

a-l-e-x-d-s-9, 2016-03-22:
Yes, the extra spaces caused the problem, thank you very much!

closed 2016-03-22 by a-l-e-x-d-s-9