silvanmelchior/RPi_Cam_Web_Interface #261: v6.0.35 Video preview problem

a-l-e-x-d-s-9, 2016-03-23:
preview.php can not preview video, but downloading video is working.
Seems that there is an extra dot after ".mp4", so it tries to play a file with extension ".mp4." and not ".mp4".
Browser error: "Failed to load resource: the server responded with a status of 404 (Not Found)"

roberttidey, 2016-03-24:
Not seeing that problem myself. If you right click the preview video and copy video link then paste it elsewhere what do you see.

Also have you edited the video_path in /etc/raspimjpeg as that defines the filename including a .mp4 on the end

a-l-e-x-d-s-9, 2016-03-24:
I haven't changed video_path in /etc/raspimjpeg, it is:
`video_path /var/www/html/media/vi_%v_%Y%M%D_%h%m%s.mp4`

If I copy the link, it has extra dot at the end, without this dot I can play the video in the browser.

You can see yourself the src value for video in my Google Chrome (Version 49.0.2623.87 64-bit) Inspect (same thing at Firefox 45.0):

[image]

roberttidey, 2016-03-25:
I think I know what the problem may be. You have recorded enough stuff so that your index is now 5 digits long (16885). The video mp4 name is derived from the thumbnail by removing the 'thumbnail suffix' characters at the end of the name.

This worked OK in previous versions because the server code worked out how to remove the suffix even if it lengthened from 4 to 5 or higher index number.

Recently I merged in some code supplied which moved this logic to javascript to make moving between previews faster. Unfortunately this code assumes a 4 digit length when removing the suffix and so it leaves behind an extra '.'

I will fix that shortly.

roberttidey, 2016-03-25:
Should be Ok now in 6.0.36. It was also broken if stuff was stored in subfolders under media; also fixed.

a-l-e-x-d-s-9, 2016-03-27:
Yeah, it's working now, thanks!

closed 2016-03-27 by a-l-e-x-d-s-9