silvanmelchior/RPi_Cam_Web_Interface #264: Proposals to improve update/installation process

a-l-e-x-d-s-9, 2016-03-27:
I have 2 proposals to improve update/installation process:
1. Make it possible for update.sh & install.sh to receive arguments for dialog forms, to replace human interaction with automation...
2. Maybe there is some way to store the server username and password encrypted somehow in config.txt (or anywhere else), and not plain text? for security reasons...

roberttidey, 2016-03-30:
1. This should already be possible for install. The parameters are read from config.txt. If "q" is included as a command line parameter then a quiet install is performed just using the values in config.txt Currently update doesn't pass that through but I'll change that.

2. I'll look at that one. It probably won't be very secure though as it would be more like obfuscation. The script needs to decode it to do its work, so it would be possible for somebody else to do the same. The other option would be not to store it and so a user would need to enter it each time, but that would conflict with quiet installs.

a-l-e-x-d-s-9, 2016-05-01:
I see that installation still asking the user either to reboot or not, it would be great if it could be possible to determine it with another argument for install/update, to make a fully automatic update process.

roberttidey, 2016-05-02:
Actually I believe most of the time the reboot is unnecessary even after a first install. It should be sufficient to just run ./start.sh The reboot request was left in as a safety measure.

I'll change wording to reflect this.