silvanmelchior/RPi_Cam_Web_Interface #265: Remove option deleted whole RPi

Rall3n, 2016-03-27:
Tried the RPi_Cam_Web_Interface.
Could not start the stream or had no connection to stream!

Tried to uninstall the interface, spit out thousands of no permission errors! Tried again, after that RPi had no images for the programs, no programs at all and could not reboot/shutdown!

miraaz, 2016-03-27:
What metod you used when uninstalled? Sounds like uninstall started remove files in root some reason.

Rall3n, 2016-03-27:
At first I typed "./RPi_Cam_Web_Installer.sh uninstall"!
I mistook it with "remove", but it just opened the terminal-based menu.
There I selected "remove" -> "Do not remove nginx and php" and then it started ouputting the permission errors!

miraaz, 2016-03-27:
Ok. Some reason script not founded www root location i think. But why? I tested script now and its finds www root... 
Can you write exactly steps how i can reprodutce that situation?

1. You closed script after install? And reopened it when wanted remove?
2. Wat webserver you used? apache2 or ngnix?

I think maby you used ngnix when installed and not closed script after install? Same bug was before. When webserver not was installed then script not founded www root right time. I fixed that issue when apache used but i mast look its happends also when ngnix choosen...
And when you not close script after install, then maby its not find www root at all...
If script closed after install and relaunched then www root finded and that not happened.

miraaz, 2016-03-27:
Hm. stil not clear! www root finded after apache2 installed line 829. And after ngnix installed line 960. When script launched and webserver is installed it founds www root in line 56.
So i not find way how www root forgot in your sitution?
Possibility: maby you used some old not fixed version installer? You updated script before you wanted uninstall?

Rall3n, 2016-03-28:
Maybe it was because I used the installer! I now installed RCWI on a new setup of raspbian with the install.sh instead of the RPi_Cam_Web_Interface_installer.sh. Currently it is working without any problems!

miraaz, 2016-03-28:
No have idea how that happend to you. I developing RPi_Cam_Web_Interface_installer.sh and using it himself also. I cant confirm, that bug is possible with current installer.

shatteredsword, 2019-07-25:
i have had this happen multiple times when trying to migrate a pi from jessie->stretch or stretch->buster. i usually uninstall this project and reinstall after the upgrade, but sometimes the uninstall errors out and ends up breaking the entire OS permanently. this has caused me some major headaches. I will try to replicate the exact steps to reproduce when i have some down time.

roberttidey, 2019-07-26:
Note you should be using ./install.sh. Some of the problems here were caused by using the older install method.

Having said that when doing a major OS upgrade I would always make an img back up of the SD to make it easier to revert if things go wrong. When switching from OS (like stretch to buster) I would normally start from a fresh install of the OS.

shatteredsword, 2019-07-29:
./install.sh has nothing to do with it. the problem happens when _uninstalling_

roberttidey, 2019-07-29:
I was distinguishing between the separated install scripts like ./install.sh (and remove.sh) compared to the use of ./RPi_Cam_Web_Installer.sh and its uninstall option which are referenced in this issue

I only support the ./install.sh and associated scripts like ./remove.sh.