silvanmelchior/RPi_Cam_Web_Interface #268: RPi Cam Control v6.0.37 keeps crashing when recording via motion

valki2, 2016-04-01:
I am using a RPI 2 with a RPICam and latest Raspian "Linux raspberrypi 4.1.19-v7+" running on it.
RPi Cam Control Version is v6.0.37.

The little raspi runs pretty stable when just using as a webcam (1 week is no problem)... but as soon as I turn on motion to capture some videos - then the system is crashing within some hours.

More details:
- raspian itself remains reachable via SSH
- Apache keeps running
- its even possible to reboot the raspi via RPi Cam Control webinterface
- Image is not visible of course as soon as the problem occurs
- After the reboot, the last recorded video is "stuck"/Broken - seems to be the result of a interrupted writing process

I tried an older version of Raspian, different SD cards and am running out of good ideas now... 
Any advices are highly appreciated :-)

roberttidey, 2016-04-01:
Internal or External motion detection? I now use internal all the time as it is simpler and uses less resource.

The schedule log will have some details in leading up to a crash.

valki2, 2016-04-02:
Hi!
I was using external and the whole config very "out-of-the-box". I changed now (2016 04 02) to internal and it seems more stable but I need to figure out now new motion values... on the first trials it seems that the trigger values (%c) keep at 0000 while already a person walks halfway through the image :)

Nevertheless the crashes with external motion remain strange - for interested and better skilled folks I allowed myself to put my full schedule log here: http://ul.to/k6mdtz28

Most promising parts are like this:
`{2016/04/01 22:38:52} Capturing started
{2016/04/01 22:38:52} Finished boxing /var/www/media/vi_0304_20160401_223847.mp4 from Box Queue at pos 17
{2016/04/01 22:38:52} Removed item from Box Queue
[2016/04/01 22:38:59] Stop capture requested
[2016/04/01 22:39:01] Send ca 0
{2016/04/01 22:39:01} Error: Could not send buffers to port, h264 callback
[2016/04/01 22:39:03] Start capture requested from Pipe
[2016/04/01 22:39:03] Send ca 1
{2016/04/01 22:39:03} Executing macro /var/www/macros/error_hard.sh "Could not send buffers to port, h264 callback"
{2016/04/01 22:39:08} RaspiMJPEG Version 5.4.2
{2016/04/01 22:39:08} Error: Could not enable camera
{2016/04/01 22:39:08} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"
{2016/04/01 22:39:13} RaspiMJPEG Version 5.4.2
{2016/04/01 22:39:13} Error: Could not enable camera
{2016/04/01 22:39:13} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"
[2016/04/01 22:39:15] Stop capture requested
[2016/04/01 22:39:15] Send ca 0
{2016/04/01 22:39:18} RaspiMJPEG Version 5.4.2
{2016/04/01 22:39:18} Error: Could not enable camera
{2016/04/01 22:39:18} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"
{2016/04/01 22:39:24} RaspiMJPEG Version 5.4.2
{2016/04/01 22:39:24} Error: Could not enable camera
{2016/04/01 22:39:24} Executing macro /var/www/macros/error_hard.sh "Could not enable camera"
`