silvanmelchior/RPi_Cam_Web_Interface #272: Internal motion detection mode 1000+ problem with resolutions 1080p, 720p, 576p

a-l-e-x-d-s-9, 2016-04-22:
When video resolution set to 1080p, 720p or 576p, internal motion detection mode 1000+ has a hard time to recognize any motion at all, even though Motion Vector Preview show huge changes, motion detection not triggered vast majority of the time, and annotation %c stays on 0. 
When video resolution set to 972p, with the same exact other settings, motion is detected properly every time.
I think that it's an old problem, that I couldn't characterize up until now.
Tested on 2 different RPi 2, v6.0.38 and v6.0.37, one with most recent software and firmware and one with a little less update system.

roberttidey, 2016-04-22:
Thanks for the report. I do use 972 mainly myself and there is something strange with other resolutions. I couldn't see anything obviously wrong, so will need to check out further.

roberttidey, 2016-04-23:
I think I found the problem. It was not directly related to resolution but to some 2d filtering which was not being calculated correctly.

Try new version.

a-l-e-x-d-s-9, 2016-04-23:
I tested the new version v6.0.39 now, indeed it detects motion much better with 1080p, 720p, 576p resolutions.
But there are couple things that confuse me:

1. If I compare 972p vs 576p. I can make the same motion (waving my hand in front of the camera), with the same configurations (besides resolution), and it would be detection with 972p, but would not be detected with 576p. I think that same motion, should produce the same angular size on both 972p and 576p, but the ratio of motion related pixels to area, should be higher with 576p in comparison to larger image in 972p. So I suspect that motion detection threshold still relies on variable related to  972p resolution and not other resolutions. 
(Tested with: Noise level = 1003; Threshold = 7; Frames to start = 2; Frames to stop = 100 )

2. With the values: Noise level = 1003; Threshold = 7; Frames to start = 2; Frames to stop = 100; 972p; motion is detected very reliably.
But when I change it to Noise level = 1004, no motion detected at all, even with huge motion very close to the camera, annotation %c showing maximum 6 and no motion detection triggered. I suspect there is something wrong there too.

roberttidey, 2016-04-24:
OK. I'll check further.
The noise level above a 1000 should just be changeing the filtering of changes detected over frames by applying a moving average. 1000 = no filtering, 1001 averaging over 2 frames, 1002 over 3 frames and so on.
The threshold is an absolute value so I would expect to use a lower value for lower resolutions which have less vector data points to contribute.

a-l-e-x-d-s-9, 2016-04-26:
Please try: 972p, Noise=1004, Threshold=7, Start Frames=2.
I see that %c never going pass 6 change frames, like it's the maximum value.
Interestingly:
Noise=1005, Threshold=8, can reach maximum 7 change frames.
Noise=1006, Threshold=9, can reach maximum 8 change frames.
Noise=1007, Threshold=10, can reach maximum 9 change frames.
Noise=1008, Threshold=11, can reach maximum 10 change frames.
Noise=1009, Threshold=12, can reach maximum 11 change frames.
Noise=1010, Threshold=13, can reach maximum 12 change frames.

a-l-e-x-d-s-9, 2016-04-26:
Noise=1004, Threshold=8, maximum change frames: 11.
Noise=1004, Threshold=9, maximum change frames: 11.
Noise=1004, Threshold=10, maximum change frames: 16.
Noise=1004, Threshold=11, maximum change frames: 16.

roberttidey, 2016-04-27:
There is some clipping of changes detected in a frame based on the threshold value. This is intended to prevent single frame flashes contributing too much and triggering unnecessarily.

When I use threshold values > 20 then all seems OK. Typically I am using thresholds of 100 or more to give reliable detection and few false triggers.

There could be a clipping problem with using very low threshold values. I'll check that out

roberttidey, 2016-04-28:
I think I know what causes this with low threshold values. The filtering is being done with integer arithmetic which is fine for larger values of 'noise'and threshold values but the rounding down can cause limitations at low values. I am checking how to improve that.

roberttidey, 2016-04-28:
Should be better now

a-l-e-x-d-s-9, 2016-04-28:
Yeah, looks that it's working better with small thresholds now.
But I see that for Noise=1004 the minimum value for change frames is 2, even when there is no motion at all, and for Noise=1010 change frames is starting with 5 and not 0.

roberttidey, 2016-04-30:
Noise 1010 effectively means a moving average over 10 frames which itself introduces a lag for the change count to build up. Because of the clipping being done to limit the contribution of very large changes this would mean that it takes about 3 to 4  frames for the motion to exceed the threshold and start the change frame count.

I may introduce an extra clipping factor config factor to allow that to be tuned from the web interface. Currently any single frame change count is clipped to 3 times the threshold value.

roberttidey, 2016-04-30:
Added motion clip control

a-l-e-x-d-s-9, 2016-05-01:
Can you explain please what clipping factor does, and when to use its small or large values?

roberttidey, 2016-05-02:
I have updated the wiki with more info on this.

a-l-e-x-d-s-9, 2016-05-03:
Thanks.
There is still something odd with 1080p vs other resolutions. 
Version 6.1.4, motion settings:
[image]
In a relative low light conditions, without any motion, 972p, 720p and 576p, act as expected and not detecting motion. But 1080p in the same conditions, its change frames keep jumping, and every few seconds it detects non existing motion.

roberttidey, 2016-05-04:
I can't see anything specific around 1080 at the moment. There are more vectors so it will be more sensitive at any particular threshold and the value 9 is quite low. 

It may be worth turning on the vector preview which gives a view of what vectors are coming out from the camera processing. Black effectively means no movement.

closed 2016-08-04 by roberttidey