silvanmelchior/RPi_Cam_Web_Interface #273: No possibility of grabbing full image ?

runeste, 2016-04-23:
Sorry, might not be an issue at all, but I see no FAQ's section here ;)

There is no way to grab an full image without actually recording one ?
Does the RASPI process no grab a full frame image, only HD resolution one (cropped). ?

Also, as a sidenote, since I am making a shitty ticket here: AWESOME program! ;)

roberttidey, 2016-04-23:
The software can capture videos at full 1920 HD resolution and still images up to max camera resolution (2592 X 1944). Time lapse capture at still image resolution as well.

runeste, 2016-04-24:
Hi, and thanks for your promt answer. Yes I know, all the capturing is working all and well. However, I am lacking any possibility of full frame preview. Is there none ?

roberttidey, 2016-04-24:
OK. It was grabbing in the title that confused me. 

You can control the width of the preview image using the preview width control under camera settings.

For example you could put in 2560 as the width. This will up the bandwidth needed to display the preview and you will need a monitor capable of displaying it unless you want to scroll.

runeste, 2016-04-24:
Changing the preview size only enlarges the already cropped picture actually, into a huge size as you said :)

However, the image base is stilled heavily cropped. See attached files of grabbed image and preview image. What bothers me is that instead of resizing the image, the preview crops it extremely.

[image]
[image]

roberttidey, 2016-04-24:
The preview is currently based on the video format that is selected so that it represents the framing that will apply when a video capture is applied. So if you have a video format that is widescreen e.g 16:9 then that will determine the preview aspect ratio and associated cropping which is done by the camera system.

You can set the image aspect ratio and the video aspect ratio independently. So you can have a 1920 x 1080 video (16:9) and a 2592 x 1944 still (4:3) but the preview will be showing the video view. I actually use a 4:3 video view most of the time so that I get maximum coverage from the sensor.

Today you could set the video to be a 4:3 format and then it would accurately represent the 4:3 still image format and you could still control the size with the preview width control.

In principle, it would be possible to provide a user selectable feature to make the preview aspect ratio represent the video or image format. I can look into that. Up till now most people have been interested in seeing the video view.

runeste, 2016-04-24:
Thank you!  That makes sense, and also allows me to modify it to my use as I do not use the video feature currently. Thanks for explaining the workings of it!  Have a wonderful day! :)

closed 2016-04-24 by runeste