silvanmelchior/RPi_Cam_Web_Interface #275: New issue arisen with /etc/raspimjpeg.1 file

elajf, 2016-04-29:
I was using an earlier version of RPi_Cam_Web_Interface, and was able to change the path of the time-lapse function in the raspimjpeg file that creates sequential images, to a simple timelapse.jpg so that each new file overwrites the previous file. However after the update I have edited the raspimjpeg file again, but the time-lapse images are still saved sequentially. Its as if there is another file affect it somewhere.
Any ideas? 
Thanks

roberttidey, 2016-04-30:
I'll check that out. I wasn't aware of any change that would have affected that.

roberttidey, 2016-04-30:
Just done a test where I changed raspimjpeg to
lapse_path /var/www/media/tl_lapse.jpg
stop started camera to get it to load settings
ran a 10 time lapse sequence
and got a single thumbnail and a single image 
tl_lapse.jpg  tl_lapse.jpg.t0000.th.jpg

Have you stop / started camera since changing raspimjpeg? If so and you still have a problem please post that part of the raspimjpeg file.

One can put settings like this in the uconfig file as well which will override the raspimjeg settings. The paths are not stored there from the web interface so if there then you must have edited it in.

elajf, 2016-04-30:
i'll try again but I was making the change then rebooting the pi, would that not have the same effect?
And sorry i listed the path wrong, I meant the /etc/raspimjpeg/raspimjpeg.1 file within the RPi_Cam_Web_Interface folder, so: RPi_Cam_Web_Interface/etc/raspimjpeg/raspimjpeg.1 

Will check again on Monday when i'm back in Uni and let you know. Thanks

roberttidey, 2016-05-01:
Rebooting will have the same effect as restarting.

Just to be clear on the config files
There is a ~/RPi_Cam_Web_Interface/etc/raspimjpeg/raspimjpeg.1 file

This is a template config file which is used during the installation to create the working copy which is a file /etc/raspimjpeg

The camera software uses the /etc/raspimjpeg file. Editing the one in the install folder will have no effect on the working system.

If you want to make changes after the install then you should sudo nano /etc/raspimjpeg

Note also if you have changed the one in the install folder then it may affect future installs.

elajf, 2016-05-01:
Ohhhhh! Its just clicked! 
That's what I was doing wrong - the result of working way too late at night! That should fix it, thanks a lot!

closed 2016-05-04 by elajf