silvanmelchior/RPi_Cam_Web_Interface #276: LED on while recording

goermezer, 2016-05-04:
Hi,

first of all thanks for this great little interface to my RPI cams. I really love it`s flexibility.

I would like to switch the LED of the camera module on and off while recording starts and ends. Is this feature planned? If not, is it possible to implement it in one of the PHP scrips? 

Thanks
Mustafa

goermezer, 2016-05-04:
Hi,

to answer myself, I found a solution by using GPIO in a python script:

`
#!/usr/bin/env python
# To use this solution, the LED must first be switched off in /boot/config.txt with disable_camera_led=1

import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
# Set GPIO for camera LED
# Use 5 for Model A/B and 32 for Model B+
CAMLED = 32
# Set GPIO to output
GPIO.setup(CAMLED, GPIO.OUT, initial=False)

# Then I use this to switch on the LED:
GPIO.output(CAMLED,True) # On

# And this to switch it off:
GPIO.output(CAMLED,False) # Off
`
Unfortunately this doesn`t work at the moment by executing it from my start_vid.sh and end_vid.sh in /var/www/html/macros.

roberttidey, 2016-05-05:
Scripts in the macros folder should be owned by www-data and be given execute privileges. To check they are running successfully. You can use the example test.sh (rename and give privileges. This just update /var/www/testmacro.txt with current date/time.

If this is working then maybe it is something to do with www-data user being able to run GPIO. I think a user has to be a member of the gpio group to do that. By default user pi is a member but www-data is not. Something like 
sudo usermod -a -G gpio www-data

goermezer, 2016-05-05:
Hi roberttidey,

the scripts are owned by www-data, they are executable (755) and Apache runs as www-data. The usermod command was a good hint - I forgot to set that on my Raspberry. Unfortunately it still do not the trick. 

Curiously when I execute the macro like this, the led lights up as expected:

`sudo su - www-data -c /var/www/html/macros/start_vid.sh`

But it doesn`t work when executed from the sheduler. I gave up here, because I had success with a separate Python script, which detects movement with a PIR-sensor. I simply use that script to start/stop the recording, take photos and at the same time I control the LED too.

Nevertheless thanks for your help.

P.S.: I`m new to github. Shall I close the Issue? Or is it good practice to leave it open for others with the same issue?

goermezer, 2016-05-06:
Ok, seems that I needed to login once again. After a reboot of my Raspberry the LED works now :-).

closed 2016-05-06 by goermezer