silvanmelchior/RPi_Cam_Web_Interface #279: Stop on_motion_detected from overlapping

mrcrocodile556, 2016-05-12:
For my project, once motion detected there will be a default playback of a short wav file of up to 15seconds.

However when many motions are detected together, the tone are all played together at the same time overlapping one another producing a muffled sound and giving me a headache.

Do you know any python loops or any specific commands i can stop the sound from overlapping? 

In other words, when the first motion is detected, it will produce my greeting tone however when second motion is detected right after, it will not play the tone causing it to overlap, greeting tone will only be played once the first one ended.

Thank you in advance!

roberttidey, 2016-05-12:
Few things
You can set motion triggers so that they have a minimum duration of capture to minimise the problem.
If using macros to trigger playback then you could use a semaphore file to make sure they run sequentially. Macro tests to see if semaphore file is present. If not then it creates it then plays audio then deletes it. If semaphore is present (from another macro playing) then it tests in a loop say every second till file goes away before proceeding.

mrcrocodile556, 2016-05-13:
Where can i find "set motion triggers" option in motion.conf ?

roberttidey, 2016-05-13:
There are 2 methods of motion detection used by this interface, selectable under camera settings.

The older one known as external uses the external motion detection program. You access and set up its parameters via the web interface. This changes the parameters in motion.conf via the API provided by motion. event_gap specifies the seconds of no motion before a stop trigger is generated. Increasing it will reduce the likelihood of multiple small captures. E.g if set to 6 seconds, if motion stops but restarts 5 seconds later then these no stop is generated and the 2 pieces of movement are combined.

The newer method of detection called internal uses the vector movement data that can be got from the camera and does not use the external motion process at all. It has a different configuration set. THe equivalent of event_gap is  'Still frames to Stop' . The default value of 150 gives 6 seconds at 25fps.

closed 2016-08-04 by roberttidey