silvanmelchior/RPi_Cam_Web_Interface #280: Annotation only shows a question mark (?)

alexg2016, 2016-06-02:
I´m runing a net installation (https://github.com/debian-pi/raspbian-ua-netinst) of Raspbian GNU/Linux 8 (jessie).

Everything works fine except the annotation. It always shows a question mark instead of the defined text on the pictures.

The scheduler log says the annotation text has changed successfully ({2016/06/02 12:47:30} Change: annotation = Test). The uconfig looks correct, I even tried to change the annotation directly in  /etc/raspimjpeg with no success.

Any ideas?

roberttidey, 2016-06-02:
Never seen that problem before. I suspect it may be some character encoding issue causing the annotation string being fed through to the camera MMAL to get rejected.

Could you zip and attach your uconfig file so I can check it out. Important to zip it so that the exact contents including character encoding get preserved.

alexg2016, 2016-06-07:
Ok, I attached the zipped uconfig.

If I do a " file -b uconfig" it says the file is "ASCII text".

[uconfig.zip](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/303198/uconfig.zip)

I will try Jessie Lite later and will report again then

roberttidey, 2016-06-07:
File looks OK and I tried with those settings on a Jessie Lite system and all was OK. So at the moment my best guess is that it is connected with the OS.

I do notice from the uconfig that you have custom text colour and custom background colour enabled but the values are at the default settings (same as disabled). This is OK on my Jessie system.

alexg2016, 2016-06-08:
I now use the Jessie Lite Image and everything works as expected. Thanks for your help :+1:

closed 2016-06-08 by alexg2016