silvanmelchior/RPi_Cam_Web_Interface #281: Critical: Installing on nginx crashes it

realies, 2016-06-09:
I've just did a fresh install with option to have the web folder installed in "html/cam", opened my local nginx server and was welcomed by a 500 Internal Server Error.

error.log
> 2016/06/09 11:08:23 [error] 13348#0: *1 rewrite or internal redirection cycle while internally redirecting to "/index.html", client: 192.168.1.120, server: localhost, request: "GET /cam HTTP/1.1", host: "192.168.1.86"

roberttidey, 2016-06-09:
The nginx part of the install is less mature than the apache but does work. 
It doesn't currently support using a port different to 80 and I think you may have problems using two subfolder depth.
Try installing in html or in cam

realies, 2016-06-09:
Done that and after seeing a never changing "Loading..." image I gave up.