silvanmelchior/RPi_Cam_Web_Interface #282: Request: Perform Actions via GPIO

CaptInsano, 2016-06-12:
I have searched around http://elinux.org/RPi-Cam-Web-Interface and I cannot find a guide for this situation:

I would like to be able to use buttons on the GPIO to trigger camera actions, one to take a photo and one to start/stop a video recording while also using RPi_Cam_Web_interface. 

I want to use the Web UI to view and download images/videos to my PC but have the option to use a physical button to take the shots/videos.

Any other "press GPIO button for picture" guides do not work as the RPi_Cam_Web_interface has taken control of the camera.

Any advice?
Thanks

roberttidey, 2016-06-12:
Use the pipe commands in raspimjpeg to do the actions you want. To capture an image send I'm to the FIFO pipe.

roberttidey, 2016-06-12:
Just to add a bit more detail to this; the procedure would be
1. Detect your GPIO change in whatever environment you want
2. Decide what camera action you want to take
3. Translate that action into equivalent pipe commands
4. Send those to the FIFO pipe input which is located in the web install folder (e.g. /var/www/FIFO)

An example of sending a command from a bat file would be echo -n 'im' >/var/www/FIFO

From Python you would do something like
            $fifo = fopen("/var/www/FIFO", "w");
            fwrite($fifo, "im");
            fclose($fifo);

You can see a list of all possible commands in the wiki or on the scheduler web page which has a reference section.

CaptInsano, 2016-06-13:
Thank you so much, I will try this later. 

Another question, it is possible to show the camera output on the raspberry pi hdmi as well as the WebUI? 

I would like to attach a mini screen to the pi as well as GPIO buttons.

roberttidey, 2016-06-13:
You can open up a browser on the local pi and see the output there as well as on remote browsers.

CaptInsano, 2016-06-13:
Any way of just having the video output full screen? It is a small 480x800 5 inch screen that I would like not to lose real estate on from browser or onscreen buttons.  

How about using VLC or some streaming software set up to camera output?

roberttidey, 2016-06-13:
The browser can stream just the picture (see wiki). Browsers can be set to hide toolbars / menus etc. Together this would give close to full screen.

People have also used vlc to pick up the mjpeg stream available from the web site. There were some posts on the forum about this.

florisjan78, 2016-06-16:
Sending instructions directly to the FIFO pipe can cause these instructions to collide with the internal commands, causing instructions like CA 1 to be missed. Before I found out about the user annotate system I would send 'an xxxx' commands to the FIFO in the way roberttidey describes 4 times a second and this happened a lot. Either the AN or the CA instruction was often lost. Sometimes a broken pipe error occured, but not often.

roberttidey, 2016-06-17:
Yes. There is no specific protection against multiple separate processes writing to the pipe. It relies on the pipe processing items fairly fast to make this unlikely. If you are using GPIO to trigger sending commands then maybe you want to make sure you don't send them too quickly or wait for them to be actioned by checking the status file.

The pipe is currently checked 10 times a second and processes a single command when it finds one. This could be improved in a couple of ways. The pipe could be checked a bit more often, maybe up to 50 times a second would be OK. Secondly the pipe could check for multiple commands. Both these need a bit of work as they involve adding extra command separators and allowing commands to be split up.

florisjan78, 2016-06-17:
Another solution would be using a third FIFO, where commands can be sent by external programs and then forwarded to raspimjpeg. This would require a lot more work though and the only real advantage would be that commands can be scheduled/prioritized, which isn't really necessary. 

So I would go for the multiple commands option, as checking 50 times a second only reduces the chance of collision by a factor of 5. This would still cause several collisions a day in the situation I had.

Can't commands just be split up by line separators?

Floris Jan

roberttidey, 2016-06-18:
Yes. Line separators was what I was thinking of putting in to allow multiple commands It was done without them in the original program and I've never changed it. It does mean changing the input processing a bit and it also risks breaking other programs or user macros that don't put them in. It might need to have some backwards compatibility handling in there as well; like a time out to process a command anyway after a short delay.

Merged instructions is a more difficult risk to deal with. In normal use where the clash is between a schedule / motion trigger and a web user the risk is low. It would need something like a lock during writing to avoid it but writing programs would then need to be 'lock aware' as well.

florisjan78, 2016-06-18:
That brings me back to a third FIFO, with locking, or a fail safe system for merged instructions. This way the original system remains unaffected and the new system can be implemented safely into existing code. A lot of work but also a lot of problems prevented. It's a workaround, I know, but if it prevents people from having failing systems it may be worth it.

roberttidey, 2016-07-14:
Latest version can now support command input into up to 9 more FIFO pipes. These are labelled FIFO11 through FIFO19. They are recognised automatically if present when the camera starts up. The installer creates FIFO11. If more are needed then they can be created with a couple of command lines.

Commands may now be terminated with LF and the input processing will handle split or merged commands if this is used. Previous non terminated commands will still work.

florisjan78, 2016-07-15:
Very nice work! I'm on holiday but as soon as I'm back I'll upgrade and
check it out.

Regards