silvanmelchior/RPi_Cam_Web_Interface #283: Request: Thumbnail delay or multiple thumbnails

florisjan78, 2016-06-14:
When the motion is triggers, the object causing the motion is only partially visible. So a delay, or multiple thumbnails at a given interval would be appreciated.

Thanks

FlorisJan

roberttidey, 2016-06-14:
If you are talking about video motion before the trigger then this is what the buffer facility is for. Under camera settings set video buffer to say 4000. It then records into a RAM buffer all the time. When a motion trigger occurs then the RAM buffer is prepended onto the recording so you get a few seconds before the trigger occurs.

If you are talking about the recording stopping to soon then adjust the motion parameters so that recording needs a longer still period before a motion stop trigger is generated.

If you are talking about snapping an image then not a lot can be done for capturing before a motion trigger occurs. You could use a low frame rate video instead.

florisjan78, 2016-06-16:
Thanks for your reponse Robert!

I would like for the thumbnail to be taken a short period after the motion trigger, not before. So the object is fully visible, not just the part causing the trigger. I can of course grab a copy of the jpg stream myself using python, but this is very cpu consuming.

FlorisJan

roberttidey, 2016-06-17:
I want to be clear what you are trying to achieve here as I can interpret this 2 ways. First is where you want to capture an image and its corresponding thumbnail after a delay from a motion trigger. Second is where you are doing normal motion triggered video captures but you want the thumbnail associated with this to be sometime after a motion trigger is detected.

A potential solution in both cases could be achieved with a macro to delay the command sent as a result of the motion trigger.
So taking the first case if scheduler is sending an im command in response to a motion trigger then instead you send a sy command to execute a macro. The macro just sleeps for the required delay and then sends im to the PIPE.
In the second case you again substitute the ca 1 command by the macro which delays then issues the ca 1 command. Now by delaying the start of the video capture you may think that you lose some of the video but if you use the buffer then you are already including several seconds before the motion trigger occurred. You see this effect already if using the buffer because the thumbnail capture occurs several seconds into the captured video.

florisjan78, 2016-06-17:
I meant the second one, sorry about my fuzzy explanation and thanks for answering both possibilities. 

The solution sounds pretty good. Also taking into consideration my comment in thread #282, because by replacing the ca 1 command with a macro, which then sends the ca 1 command, the commands cannot collide in the FIFO.

Thanks, I'll let you know how it works.

Floris Jan

florisjan78, 2016-06-19:
I've integrated the additional thumbnail into my python code so I can control exactly when it is 'taken'. Therefore I haven't tried the macro solution but I'm sure it would also work. Thanks again for your help.

closed 2016-08-04 by roberttidey