silvanmelchior/RPi_Cam_Web_Interface #286: Little bug on sorting medianumbers > 10000

digisaster, 2016-07-04:
If you have lots of little files and the number will be > 10000 then the preview.php will start with the number 10000, then it will count to the last file, and after the last file it will start with the oldest.
(so the order is wrong)

Its not a big deal, but its probable easy to fix.

P.s. great program. Thanks a lot.

roberttidey, 2016-07-05:
Y. It is sorted alphabetically so when the digits go from 4 to 5 then the sort order doesn't follow the index. I'll take a look; it is not totally trivial as the sort would need to extract the index during the operation.

roberttidey, 2016-07-05:
I have changed this so the sort order is type + date-time based. This means the index is not used directly

digisaster, 2016-07-05:
Thanks, the order is ok now.

p.s. would it be hard to create a "maximum preview nr / page" ?
if i use a smartphone with lots of pictures, the phone/tablet will almost crash because of the 
enormous amount of images. It would be nice to have a "max pictures / page"

roberttidey, 2016-07-06:
Preview is not set up to be multi-page at the moment. You can use the Filter the images shown by date. For example, it is useful to have it set to show last 24 hours.

Currently the filter needs the update to be pressed. I could look at making that action when the setting is changed.

roberttidey, 2016-07-07:
The sort, type, and filter selectors now take immediate effect, so you can more easily use the filter to limit the number of images shown.

digisaster, 2016-07-07:
Thanks , its working great.
 "maximum preview nr / page" would still be a nice thing in the feature because a slow smartphone will still "crash" when its trying to load the preview page. (because this setting is inside the preview page)
But I'm not complaining, its working perfect on a fast tablet or a normal pc

closed 2016-08-04 by roberttidey