silvanmelchior/RPi_Cam_Web_Interface #287: Execute arbitrary shell commands during timelapse

ojones117, 2016-07-14:
Hi

I'm using your software to timelapse product over time to analyse a change in colour. I need to illuminate the product at each photo but cannot have the lights on all the time as it will effect the change in colour over time. I am able to turn the lights on and off via a shell command.

My question is, is there any way to execute a shell command before and after a timelapse photo has been taken? If there isn't that function currently, are you able to direct me on where to begin looking in order to develop it myself?

Thanks

roberttidey, 2016-07-14:
There is a facility to run macros for particular events. There is one defined for after image capture (end_img.sh if present in macros folder) after each image is captured. However, there is not one before. At the time the thinking was that image capture is fast and the main purpose would be to do something with the image data. 

I can see in your case that one before would be useful. I'll add that in in the next couple of days. Note that macros can be configured to be synchronous or asynchronous. In your case you would want the start one to be synchronous so that it completes before the capture in done.

roberttidey, 2016-07-14:
I have now added a macro call start_img in the most recent version.