silvanmelchior/RPi_Cam_Web_Interface #288: Motion stops arbitrarily even though shown as running

tman32768, 2016-07-15:
I'm noticing two problems on v6.2.5 - v6.2.8 running external motion detect on RasPi 2 and 3 running jessie 

1. I am recording image snapshots during motion events.  In low light I'm getting snapshots every 2 seconds even though there are no motion events.  In regular light it fails to capture any motion events (ie. I can put my hand infant of the camera nothing happens).

2. The main screen shows "Motion Detection Stop" and appears to be running.  No motion events get captured.  If I click on "Edit Motion Settings" the buttons appear and nothing else (no settings are displayed). "Show All" and "Show Less" do not do anything.  I have to go back to the main screen, select "Motion Detection Stop" then "Motion Detection Start".  Now I can view and edit motion settings.

roberttidey, 2016-07-15:
1. If you have set up an image capture when a motion trigger start occurs then you must be getting motion triggers under low light. Log (accessed from scheduler) will say what s going on. Likewise in higher light. Normally lack of trigger would be to do with the actual motion settings in use.

2. The motions setting page when external motion is used relies on an API interface which requires external motion to be running. That is why you only see it after pressing Motion Start.

Why don't you try the internal motion which is much easier to use?

tman32768, 2016-07-15:
Will internal motion still do snapshots instead of video?  I've never used the internal because it doesn't appear have any settings I can control.  I don't like video because they fill up my drive space too fast and the resolution is too for trying to catch license plates

roberttidey, 2016-07-15:
Yes. External or Internal just provide triggers that can do whatever you configure including snapshots.

When you select internal motion under camera settings then the old motion setting button disappears and a new motion settings panel appears under camera settings.

With internal it can be useful to include %c and %f in the annotation string to start with as these show the changes being detected in real time.

tman32768, 2016-07-15:
> With internal it can be useful to include %c and %f in the annotation string to start with as these show the changes being detected in real time.

Can you provide a clear example of how to create a snapshot during motion events using internal I don't understand what you said in this sentence.  Is this under motion.conf because I don't have anything I can add/change in the web interface

roberttidey, 2016-07-15:
motion.conf is only used by external motion.

When motion (internal or external) triggers then it just sends a 1(start) or 0(stop) to the scheduler. The scheduler set up determines what action should be taken by sending any commands that are in the motion start or motion stop settings. In All Day mode then there is only one start and one stop so the action is always the same. In 'Fixed Times' or 'Sun based' you can have different actions at different times.

So for video motion one puts a ca 1 (video start) in the Motion Start and a ca 0 (video stop) in the Motion Stop.

If one just wants to take a single picture whenever Motion is detected then one would put a im command in Motion Start and leave Motion Stop blank.

If one wants to take a sequence of pictures during Motion then set up the required time lapse interval and then put a tl 1 (time lapse start) in Motion Start and a tl 0 in Motion Stop

You can use any command as listed in the reference and you can string multiple commands together (separated by ;) This allows changing camera settings at different times of the day.

tman32768, 2016-07-15:
Thanks for the clarification I was confused by the difference in setup between the internal and external.  If I use the internal settings and want to modify via the command prompt what file do I edit? /etc/raspimjpeg?

roberttidey, 2016-07-15:
You can edit the internal motion settings in /etc/raspimjpeg or you can also edit (or create) the uconfig file which is kept in the web install folder. raspimjpeg reads /etc/raspimjpeg followed by any settings in uconfig and so the latter take precedence. Whenever a change is made from the web interface it is stored in uconfig. The advantage of making any of your own changes in uconfig is that they will survive software updates whereas the /etc/raspimjpeg would get overwritten.

tman32768, 2016-07-16:
OK thanks.

Another question.  I just reinstalled everything using the new install.sh script.  After reboot the camera doesn't start up even though /etc/raspimjpeg has 
autostart standard
motion_detection true

http://elinux.org/RPi-Cam-Web-Interface#Startup_Behaviour says I need to configure startup using ./RPi_Cam_Web_Interface_Installer.sh autostart_yes but I've read in other posts that RPi_Cam_Web_Interface_Installer.sh is no longer needed/used/recommended.

Why doesn't the camera start after reboot now and what is RPi_Cam_Web_Interface_Installer.sh used for if it is still included?

roberttidey, 2016-07-16:
The RPi_Cam_Web_Interface_Installer.sh was the original installer method developed before I got involved. When Jessie was introduced the changes were quite large so I developed my own simpler installer install.sh which is what I recommend to use. Others were tweaking the original method so I did not want to remove it.

However, I only update install.sh now.

There are 2 parts of starting up. The first is whether the raspimjpeg process is started automatically during a boot up. This is controlled by the autostart option in the install dialog. Setting this to yes (not standard) means that boot instructions are installed to start up raspimjpeg. is set to yes  Do not put standard in there. The second part is whether the raspimjpeg process itself starts up with the camera active or needs a camera start command (e.g. from web interface). This camera active is controlled by the autostart config parameter in /etc/raspimjpeg which can be standard or idle. By default it is standard which means the camera is active when raspimjpeg is started. The installer does not change the default value of this.

I agree it is a little confusing because the two separate things are called autostart and it would have been better to have called the install item autoboot.

You can run ./start.sh to start the raspimjpeg process manually and if you run ./install.sh again and set autostart to yes then it should start up automatically on each boot up.

tman32768, 2016-07-17:
> There are 2 parts of starting up. The first is whether the raspimjpeg process is started automatically during a boot up. This is controlled by the autostart option in the install dialog. Setting [...]

Is there another spot in the config files, other than /etc/raspimjpeg, that tells the camera and motion services (internal and external) to start up after boot?  

The reason why I am asking is that something goofy is going on when my pi works well for a day after a clean install, then motion stops working for no reason or the camera isn't going and there is nothing helpful in the log files and I don't want to have to keep logging in via SSH to try and restart the process.  Just now for instance I've had the pi running since this morning using internal motion and I had set up the %c and %f to try and fine tune things.  I checked just now and motion seems to have stopped working, the values are locked %c=0000 and %f=0000 and I can't get motion to start.  I just deleted everything and reinstalled and now I get %c=0005 and the values change with any video motion events.

roberttidey, 2016-07-17:
External motion uses the external motion process and when this is active then you should see a motion process when doing ps -A

Internal motion uses a vector feed from the camera and does not use the motion process at all. So you should not see motion in the process list at all.

Once the system has started and motion (external) is started or not as the case may be, then any change to motion set up should be occurring either because the user has manually started or stopped motion detection or because the scheduler has been set up to start or stop motion during one of the period changes.