silvanmelchior/RPi_Cam_Web_Interface #289: Latest install.sh seems broken

ksze, 2016-07-16:
I'm talking about these two lines:
https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/master/install.sh#L240
https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/master/install.sh#L242

For some reason I get these lines in my `/etc/motion/motion.conf`:

    on_event_start echo -n '1' >/var/www/html/html/FIFO1
    on_event_end echo -n '0' >/var/www/html/html/FIFO1

Which are obviously broken, because the FIFO1 file is at `/var/www/html/FIFO1`.

ksze, 2016-07-16:
Upon further investigation, it seems to be also caused by https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/master/install.sh#L253

ksze, 2016-07-16:
A quick blame in suggests that this commit caused the breakage: https://github.com/silvanmelchior/RPi_Cam_Web_Interface/commit/aeb5ccb52655d6b127b0011c1617a95876ad37a3

ksze, 2016-07-16:
Line 253 probably should have been removed after the fix in aeb5ccb

roberttidey, 2016-07-16:
Thanks for finding that.

The line should have been removed after that original change. 

Change now made.

ksze, 2016-07-18:
Closing this since it's fixed in https://github.com/silvanmelchior/RPi_Cam_Web_Interface/commit/84b328cafb217883a93a69b6aec3c5deb60d1c84.

closed 2016-07-18 by ksze