silvanmelchior/RPi_Cam_Web_Interface #291: Removing media and saving scheduler options does not work

vcasado, 2016-07-26:
Hi all,

When I try to delete any photo or video from **Files**, it has no effect. I got the same issue when I want to change and save the scheduler settings.

Could be a permission issue?

Thanks,

roberttidey, 2016-07-26:
Sounds like a permissions issue. The installer gives read/write permissions to the media folder for user www-data which is the Apache user user. If you have linked or mounted another drive or usb storage to that folder then you must ensure that www-data has read/write permissions on the mount as well.

Check permissions and ownership by doing a ls -l on your web install folder and also your mount point if you have done that.

vcasado, 2016-07-26:
Hi Robert,

Ownership of both folders was ok, but for safety I've just removed and installed it again from scratch and same issue. The strange thing is that it saves recordings and pictures correctly, but does not delete them.

I'm using Nginx instead of Apache, I don't know if that could be a problem.

Thanks for your help.

roberttidey, 2016-07-26:
It's possible it is to do with nginx. It used to work OK under nginx but I haven't tried that recently and it's possible that some nginx defaults have changed. Most people are using Apache.

Check /etc/nginx/nginx.conf and in particular what the 'user' setting is.

I'll see if I can do a test nginx install over the next couple of days.

vcasado, 2016-07-27:
My mistake, I forgot that I had changed the Nginx and PHP user a while ago... now it works without any problem.

Thanks for the help and sorry for the inconvenience!

closed 2016-08-04 by roberttidey