silvanmelchior/RPi_Cam_Web_Interface #292: Internal motion detection stuck on recording, video preview also stuck, v6.2.10.

a-l-e-x-d-s-9, 2016-08-03:
I noticed a problem with version: v6.2.10, on multiple RPi 2, kernel: 4.4.15-v7+.
Internal motion detection stuck on recording, video preview also stuck on a picture with annotation:
%Y.%M.%D %h:%m:%s %c %f
[image]
The end of the log:
``` log
{2016/08/03 09:58:48} send smd 1
{2016/08/03 09:58:48} Start boxing /var/www/html/media/vi_130643_20160803_095712.mp4.h264 to /var/www/html/media/vi_130643_20160803_095712.mp4 Queue pos 11
[2016/08/03 09:58:49] Start capture requested from Pipe
[2016/08/03 09:58:49] Send ca 1
{2016/08/03 09:58:49} Capturing started
{2016/08/03 09:59:06} Watchdog detected problem. Stopping{2016/08/03 09:59:07} send smd 0
[2016/08/03 09:59:07] Stop capture requested
[2016/08/03 09:59:07] Send ca 0
{2016/08/03 09:59:09} send smd 1
[2016/08/03 09:59:09] Start capture requested from Pipe
[2016/08/03 09:59:09] Send ca 1
{2016/08/03 09:59:10} SIGINT/SIGTERM received, stopping
{2016/08/03 09:59:10} Capturing stopped
{2016/08/03 09:59:10} Add /var/www/html/media/vi_130644_20160803_095849.mp4 to Box Queue at pos 12
{2016/08/03 09:59:15} RaspiMJPEG Version 5.6.4
{2016/08/03 09:59:24} recommended video buffers 1
{2016/08/03 09:59:24} h264 buffers set to recommended 1
{2016/08/03 09:59:24} Set up internal detect width=82 height=61
{2016/08/03 09:59:24} MJPEG streaming, ready to receive commands
{2016/08/03 09:59:24} send smd 9
{2016/08/03 09:59:24} Opening FIFO 0 /var/www/html/FIFO 7
{2016/08/03 09:59:24} Opening FIFO 1 /var/www/html/FIFO11 8
{2016/08/03 09:59:24} Starting command loop
[2016/08/03 09:59:24] Reload parameters command requested
[2016/08/03 09:59:24] Scheduler loop is started
{2016/08/03 09:59:24} send smd 1
[2016/08/03 09:59:24] Start capture request ignored, day period not initialised yet
[2016/08/03 09:59:25] Scheduled management tasks. Next at 1470208765
[2016/08/03 09:59:25] Purged 8 Files
{2016/08/03 10:00:44} send smd 0
[2016/08/03 10:00:44] Stop capture requested
[2016/08/03 10:00:44] Send ca 0
{2016/08/03 10:00:44} Already stopped. Ignore 
[2016/08/03 10:00:53] New period detected 0
{2016/08/03 10:00:58} send smd 1
[2016/08/03 10:00:58] Start capture requested from Pipe
[2016/08/03 10:00:58] Send ca 1
{2016/08/03 10:00:58} Capturing started
{2016/08/03 10:01:24} send smd 0
[2016/08/03 10:01:24] Stop capture requested
[2016/08/03 10:01:24] Send ca 0
{2016/08/03 10:01:24} Capturing stopped
{2016/08/03 10:01:24} Add /var/www/html/media/vi_130645_20160803_100058.mp4 to Box Queue at pos 0
{2016/08/03 10:01:24} Start boxing /var/www/html/media/vi_130645_20160803_100058.mp4.h264 to /var/www/html/media/vi_130645_20160803_100058.mp4 Queue pos 0
{2016/08/03 10:01:24} send smd 1
{2016/08/03 10:01:25} Finished boxing /var/www/html/media/vi_130645_20160803_100058.mp4 from Box Queue at pos 0
{2016/08/03 10:01:25} Removed item from Box Queue
[2016/08/03 10:01:26] Start capture requested from Pipe
[2016/08/03 10:01:26] Send ca 1
{2016/08/03 10:01:26} Capturing started
{2016/08/03 10:02:48} send smd 0
[2016/08/03 10:02:48] Stop capture requested
[2016/08/03 10:02:48] Send ca 0
{2016/08/03 10:02:49} Capturing stopped
{2016/08/03 10:02:49} Add /var/www/html/media/vi_130646_20160803_100126.mp4 to Box Queue at pos 1
{2016/08/03 10:02:49} Start boxing /var/www/html/media/vi_130646_20160803_100126.mp4.h264 to /var/www/html/media/vi_130646_20160803_100126.mp4 Queue pos 1
{2016/08/03 10:02:49} send smd 1
[2016/08/03 10:02:50] Start capture requested from Pipe
[2016/08/03 10:02:50] Send ca 1
{2016/08/03 10:02:50} Capturing started
{2016/08/03 10:03:20} Watchdog detected problem. Stopping{2016/08/03 10:03:22} SIGINT/SIGTERM received, stopping
[2016/08/03 10:19:26] Scheduled management tasks. Next at 1470209966
[2016/08/03 10:19:26] Purged 5 Files
[2016/08/03 10:39:27] Scheduled management tasks. Next at 1470211167
```

a-l-e-x-d-s-9, 2016-08-03:
Another one stuck on:
%Y.%M.%D %h:%m:%s %c %f
[image]
Log:
``` log 
{2016/08/03 11:40:45} RaspiMJPEG Version 5.6.4
{2016/08/03 11:40:45} recommended video buffers 1
{2016/08/03 11:40:45} h264 buffers set to recommended 1
{2016/08/03 11:40:45} Set up internal detect width=121 height=68
{2016/08/03 11:40:45} MJPEG streaming, ready to receive commands
{2016/08/03 11:40:45} send smd 9
{2016/08/03 11:40:45} Opening FIFO 0 /var/www/html/FIFO 7
{2016/08/03 11:40:45} Opening FIFO 1 /var/www/html/FIFO11 8
{2016/08/03 11:40:45} Starting command loop
[2016/08/03 11:40:45] Reload parameters command requested
{2016/08/03 11:40:45} send smd 1
[2016/08/03 11:40:46] Scheduler loop is started
[2016/08/03 11:40:46] Start capture request ignored, day period not initialised yet
[2016/08/03 11:40:47] Scheduled management tasks. Next at 1470214847
{2016/08/03 11:40:50} send smd 0
[2016/08/03 11:41:17] Purged 34 Files
[2016/08/03 11:41:17] Stop capture requested
[2016/08/03 11:41:17] Send ca 0
{2016/08/03 11:41:17} Already stopped. Ignore 
[2016/08/03 11:41:19] New period detected 0
{2016/08/03 11:42:42} send smd 1
[2016/08/03 11:42:42] Start capture requested from Pipe
[2016/08/03 11:42:42] Send ca 1
{2016/08/03 11:42:42} Capturing started
{2016/08/03 11:42:52} send smd 0
[2016/08/03 11:42:52] Stop capture requested
[2016/08/03 11:42:52] Send ca 0
{2016/08/03 11:42:52} Capturing stopped
{2016/08/03 11:42:52} Add /var/www/html/media/vi_50882_20160803_114242.mp4 to Box Queue at pos 0
{2016/08/03 11:42:52} Start boxing /var/www/html/media/vi_50882_20160803_114242.mp4.h264 to /var/www/html/media/vi_50882_20160803_114242.mp4 Queue pos 0
{2016/08/03 11:42:54} Finished boxing /var/www/html/media/vi_50882_20160803_114242.mp4 from Box Queue at pos 0
{2016/08/03 11:42:54} Removed item from Box Queue
{2016/08/03 11:51:30} send smd 1
[2016/08/03 11:51:30] Start capture requested from Pipe
[2016/08/03 11:51:30] Send ca 1
{2016/08/03 11:51:30} Capturing started
{2016/08/03 11:51:34} send smd 0
[2016/08/03 11:51:34] Stop capture requested
[2016/08/03 11:51:34] Send ca 0
{2016/08/03 11:51:34} Capturing stopped
{2016/08/03 11:51:34} Add /var/www/html/media/vi_50883_20160803_115130.mp4 to Box Queue at pos 1
{2016/08/03 11:51:34} Start boxing /var/www/html/media/vi_50883_20160803_115130.mp4.h264 to /var/www/html/media/vi_50883_20160803_115130.mp4 Queue pos 1
{2016/08/03 11:51:35} Finished boxing /var/www/html/media/vi_50883_20160803_115130.mp4 from Box Queue at pos 1
{2016/08/03 11:51:35} Removed item from Box Queue
{2016/08/03 11:59:21} send smd 1
[2016/08/03 11:59:21] Start capture requested from Pipe
[2016/08/03 11:59:21] Send ca 1
{2016/08/03 11:59:21} Capturing started
{2016/08/03 11:59:25} send smd 0
[2016/08/03 11:59:25] Stop capture requested
[2016/08/03 11:59:25] Send ca 0
{2016/08/03 11:59:25} Capturing stopped
{2016/08/03 11:59:25} Add /var/www/html/media/vi_50884_20160803_115921.mp4 to Box Queue at pos 2
{2016/08/03 11:59:25} Start boxing /var/www/html/media/vi_50884_20160803_115921.mp4.h264 to /var/www/html/media/vi_50884_20160803_115921.mp4 Queue pos 2
{2016/08/03 11:59:26} Finished boxing /var/www/html/media/vi_50884_20160803_115921.mp4 from Box Queue at pos 2
{2016/08/03 11:59:26} Removed item from Box Queue
[2016/08/03 12:00:48] Scheduled management tasks. Next at 1470216048
{2016/08/03 12:05:43} send smd 1
[2016/08/03 12:05:43] Start capture requested from Pipe
[2016/08/03 12:05:43] Send ca 1
{2016/08/03 12:05:44} Capturing started
{2016/08/03 12:05:49} send smd 0
[2016/08/03 12:05:49] Stop capture requested
[2016/08/03 12:05:49] Send ca 0
{2016/08/03 12:05:49} Capturing stopped
[... 120 lines cut ...]
[2016/08/03 12:45:15] Send ca 0
{2016/08/03 12:45:15} Capturing stopped
{2016/08/03 12:45:15} Add /var/www/html/media/vi_50895_20160803_124512.mp4 to Box Queue at pos 13
{2016/08/03 12:45:15} Start boxing /var/www/html/media/vi_50895_20160803_124512.mp4.h264 to /var/www/html/media/vi_50895_20160803_124512.mp4 Queue pos 13
{2016/08/03 12:45:16} Finished boxing /var/www/html/media/vi_50895_20160803_124512.mp4 from Box Queue at pos 13
{2016/08/03 12:45:16} Removed item from Box Queue
{2016/08/03 12:45:16} send smd 1
[2016/08/03 12:45:17] Start capture requested from Pipe
[2016/08/03 12:45:17] Send ca 1
{2016/08/03 12:45:17} Capturing started
{2016/08/03 12:46:33} send smd 0
[2016/08/03 12:46:33] Stop capture requested
[2016/08/03 12:46:33] Send ca 0
{2016/08/03 12:46:33} Capturing stopped
{2016/08/03 12:46:33} Add /var/www/html/media/vi_50896_20160803_124517.mp4 to Box Queue at pos 14
{2016/08/03 12:46:34} Start boxing /var/www/html/media/vi_50896_20160803_124517.mp4.h264 to /var/www/html/media/vi_50896_20160803_124517.mp4 Queue pos 14
{2016/08/03 12:46:34} send smd 1
[2016/08/03 12:46:35] Start capture requested from Pipe
[2016/08/03 12:46:35] Send ca 1
{2016/08/03 12:46:35} Capturing started
{2016/08/03 12:46:59} Watchdog detected problem. Stopping{2016/08/03 12:47:02} SIGINT/SIGTERM received, stopping
[2016/08/03 13:00:51] Scheduled management tasks. Next at 1470219651
[2016/08/03 13:20:52] Scheduled management tasks. Next at 1470220852
[2016/08/03 13:40:53] Scheduled management tasks. Next at 1470222053
```

a-l-e-x-d-s-9, 2016-08-03:
Another one stuck on:
[image]
``` log
{2016/08/03 21:09:20} send smd 1
[2016/08/03 21:09:20] Start capture requested from Pipe
[2016/08/03 21:09:20] Send ca 1
{2016/08/03 21:09:20} Capturing started
{2016/08/03 21:09:26} send smd 0
[2016/08/03 21:09:26] Stop capture requested
[2016/08/03 21:09:26] Send ca 0
{2016/08/03 21:09:26} Capturing stopped
{2016/08/03 21:09:26} Add /var/www/html/media/vi_87929_20160803_210920.mp4 to Box Queue at pos 25
{2016/08/03 21:09:26} Start boxing /var/www/html/media/vi_87929_20160803_210920.mp4.h264 to /var/www/html/media/vi_87929_20160803_210920.mp4 Queue pos 25
{2016/08/03 21:09:27} Finished boxing /var/www/html/media/vi_87929_20160803_210920.mp4 from Box Queue at pos 25
{2016/08/03 21:09:27} Removed item from Box Queue
[2016/08/03 21:10:03] Scheduled management tasks. Next at 1470249003
[2016/08/03 21:10:03] Purged 16 Files
{2016/08/03 21:10:04} send smd 1
[2016/08/03 21:10:04] Start capture requested from Pipe
[2016/08/03 21:10:04] Send ca 1
{2016/08/03 21:10:04} Capturing started
{2016/08/03 21:11:34} send smd 0
[2016/08/03 21:11:34] Stop capture requested
[2016/08/03 21:11:34] Send ca 0
{2016/08/03 21:11:34} Capturing stopped
{2016/08/03 21:11:34} Add /var/www/html/media/vi_87930_20160803_211004.mp4 to Box Queue at pos 26
{2016/08/03 21:11:35} Start boxing /var/www/html/media/vi_87930_20160803_211004.mp4.h264 to /var/www/html/media/vi_87930_20160803_211004.mp4 Queue pos 26
{2016/08/03 21:11:51} send smd 1
[2016/08/03 21:11:51] Start capture requested from Pipe
[2016/08/03 21:11:51] Send ca 1
{2016/08/03 21:11:51} Capturing started
{2016/08/03 21:11:58} send smd 0
[2016/08/03 21:11:58] Stop capture requested
[2016/08/03 21:11:58] Send ca 0
{2016/08/03 21:11:58} Capturing stopped
{2016/08/03 21:11:58} Add /var/www/html/media/vi_87931_20160803_211151.mp4 to Box Queue at pos 27
{2016/08/03 21:12:24} Finished boxing /var/www/html/media/vi_87930_20160803_211004.mp4 from Box Queue at pos 26
{2016/08/03 21:12:24} Start boxing /var/www/html/media/vi_87931_20160803_211151.mp4.h264 to /var/www/html/media/vi_87931_20160803_211151.mp4 Queue pos 27
{2016/08/03 21:12:24} Removed item from Box Queue
{2016/08/03 21:12:28} send smd 1
[2016/08/03 21:12:28] Start capture requested from Pipe
[2016/08/03 21:12:28] Send ca 1
{2016/08/03 21:12:28} Capturing started
{2016/08/03 21:12:35} Finished boxing /var/www/html/media/vi_87931_20160803_211151.mp4 from Box Queue at pos 27
{2016/08/03 21:12:35} Removed item from Box Queue
{2016/08/03 21:12:39} send smd 0
[2016/08/03 21:12:39] Stop capture requested
[2016/08/03 21:12:39] Send ca 0
{2016/08/03 21:12:39} Capturing stopped
{2016/08/03 21:12:39} Add /var/www/html/media/vi_87932_20160803_211228.mp4 to Box Queue at pos 28
{2016/08/03 21:12:40} Start boxing /var/www/html/media/vi_87932_20160803_211228.mp4.h264 to /var/www/html/media/vi_87932_20160803_211228.mp4 Queue pos 28
{2016/08/03 21:12:41} Finished boxing /var/www/html/media/vi_87932_20160803_211228.mp4 from Box Queue at pos 28
{2016/08/03 21:12:41} Removed item from Box Queue
{2016/08/03 21:12:47} send smd 1
[2016/08/03 21:12:47] Start capture requested from Pipe
[2016/08/03 21:12:47] Send ca 1
{2016/08/03 21:12:48} Capturing started
{2016/08/03 21:13:01} send smd 0
[2016/08/03 21:13:01] Stop capture requested
[2016/08/03 21:13:01] Send ca 0
{2016/08/03 21:13:01} Capturing stopped
{2016/08/03 21:13:01} Add /var/www/html/media/vi_87933_20160803_211248.mp4 to Box Queue at pos 29
{2016/08/03 21:13:01} Start boxing /var/www/html/media/vi_87933_20160803_211248.mp4.h264 to /var/www/html/media/vi_87933_20160803_211248.mp4 Queue pos 29
{2016/08/03 21:13:01} send smd 1
{2016/08/03 21:13:02} Finished boxing /var/www/html/media/vi_87933_20160803_211248.mp4 from Box Queue at pos 29
{2016/08/03 21:13:02} Removed item from Box Queue
[2016/08/03 21:13:03] Start capture requested from Pipe
[2016/08/03 21:13:03] Send ca 1
{2016/08/03 21:13:03} Capturing started
{2016/08/03 21:13:09} send smd 0
[... 192 lines cut ...]
[2016/08/03 21:21:23] Stop capture requested
[2016/08/03 21:21:23] Send ca 0
{2016/08/03 21:21:23} Capturing stopped
{2016/08/03 21:21:23} Add /var/www/html/media/vi_87950_20160803_212118.mp4 to Box Queue at pos 14
{2016/08/03 21:21:23} Start boxing /var/www/html/media/vi_87950_20160803_212118.mp4.h264 to /var/www/html/media/vi_87950_20160803_212118.mp4 Queue pos 14
{2016/08/03 21:21:24} Finished boxing /var/www/html/media/vi_87950_20160803_212118.mp4 from Box Queue at pos 14
{2016/08/03 21:21:24} Removed item from Box Queue
{2016/08/03 21:21:44} send smd 1
[2016/08/03 21:21:44] Start capture requested from Pipe
[2016/08/03 21:21:44] Send ca 1
{2016/08/03 21:21:45} Capturing started
{2016/08/03 21:23:07} send smd 0
[2016/08/03 21:23:07] Stop capture requested
[2016/08/03 21:23:07] Send ca 0
{2016/08/03 21:23:07} Capturing stopped
{2016/08/03 21:23:07} Add /var/www/html/media/vi_87951_20160803_212145.mp4 to Box Queue at pos 15
{2016/08/03 21:23:08} Start boxing /var/www/html/media/vi_87951_20160803_212145.mp4.h264 to /var/www/html/media/vi_87951_20160803_212145.mp4 Queue pos 15
{2016/08/03 21:23:13} send smd 1
[2016/08/03 21:23:20] Start capture requested from Pipe
[2016/08/03 21:23:20] Send ca 1
{2016/08/03 21:23:20} Capturing started
{2016/08/03 21:23:36} Watchdog detected problem. Stopping{2016/08/03 21:23:42} SIGINT/SIGTERM received, stopping
[2016/08/03 21:30:04] Scheduled management tasks. Next at 1470250204
[2016/08/03 21:30:04] Purged 35 Files
```

roberttidey, 2016-08-04:
The common thread here is that the watchdog has detected a problem. This occurs when the preview has not been updated for a few seconds. Normally it should try to restart the camera system at that point. When running normally you should see 2 raspimjpeg processes when doing a ps -A. One is the real one, the other is the watchdog.

Are you only getting this when using internal motion detection? Are you linking the media folder to other storage like a USB drive or NAS storage? I have seen cases where a network problem can stop the stream long enough for a watchdog timeout to occur.

a-l-e-x-d-s-9, 2016-08-05:
I'm getting the problem only with internal motion detection, but I don't use anything else.
I'm not linking media folder to external storage, just a default folder on a regular SD card class 10. 
This problem happened on 3 different PIs.
I upgraded to v6.2.11, kernel 4.4.16-v7+, all of my PIs. For now, only another PI has stack since than:
``` log 
{2016/08/05 19:48:58} Add /var/www/html/media/vi_53453_20160805_194709.mp4 to Box Queue at pos 10
{2016/08/05 19:48:58} send smd 1
{2016/08/05 19:48:59} Start boxing /var/www/html/media/vi_53453_20160805_194709.mp4.h264 to /var/www/html/media/vi_53453_20160805_194709.mp4 Queue pos 10
[2016/08/05 19:49:00] Start capture requested from Pipe
[2016/08/05 19:49:00] Send ca 1
{2016/08/05 19:49:00} Capturing started
{2016/08/05 19:49:26} Watchdog detected problem. Stopping{2016/08/05 19:49:31} send smd 0
{2016/08/05 19:49:36} SIGINT/SIGTERM received, stopping
[2016/08/05 19:49:41] Stop capture requested
[2016/08/05 19:49:41] Send ca 0
[2016/08/05 20:08:52] Scheduled management tasks. Next at 1470418132
[2016/08/05 20:08:52] Purged 7 Files
[2016/08/05 20:28:53] Scheduled management tasks. Next at 1470419333
```
On this PI, after software has stack, there are still 2 raspimjpeg processes.
I don't understand, the watchdog suppose to restart the software, but it fails to do so in my case?

a-l-e-x-d-s-9, 2016-08-06:
Another one stuck with the same problem.

a-l-e-x-d-s-9, 2016-08-20:
The problem keep showing up each day, with v6.2.15.

roberttidey, 2016-08-20:
I don't know at the moment why your systems suffer from this. 

One thing the logs shows is that the watchdog fires but there is not a complete shutdown allowing a new one to be started. There is a stop_video call being made at this point to try to clean up and it is possible it is getting stuck in that bit. There are no obvious loops that could cause it to get stuck but it is possible that calls to the camera system are not returning.

There is also a delay of a few seconds between the watchdog detecting a problem and the SIGINT/SIGTERM received which is a bit strange as it is not doing much in between. Do you have other applications or systems installed which could be taking CPU time away?

I am looking at adding some more logging and forcing the stop when the watchdog fires.

a-l-e-x-d-s-9, 2016-08-20:
Nothing else is running in the system, as far as I know.

a-l-e-x-d-s-9, 2016-08-22:
``` log 
[2016/08/22 19:11:30] Send ca 1
{2016/08/22 19:11:30} Capturing started
{2016/08/22 19:11:36} send smd 0
[2016/08/22 19:11:36] Stop capture requested
[2016/08/22 19:11:36] Send ca 0
{2016/08/22 19:11:37} Capturing stopped
{2016/08/22 19:11:37} Add /var/www/html/media/vi_7455_20160822_191130.mp4 to Box Queue at pos 18
{2016/08/22 19:11:37} Start boxing /var/www/html/media/vi_7455_20160822_191130.mp4.h264 to /var/www/html/media/vi_7455_20160822_191130.mp4 Queue pos 18
{2016/08/22 19:11:38} Finished boxing /var/www/html/media/vi_7455_20160822_191130.mp4 from Box Queue at pos 18
{2016/08/22 19:11:38} Removed item from Box Queue
{2016/08/22 19:15:33} send smd 1
[2016/08/22 19:15:33] Start capture requested from Pipe
[2016/08/22 19:15:33] Send ca 1
{2016/08/22 19:15:33} Capturing started
{2016/08/22 19:18:26} send smd 0
[2016/08/22 19:18:26] Stop capture requested
[2016/08/22 19:18:26] Send ca 0
{2016/08/22 19:18:26} Capturing stopped
{2016/08/22 19:18:26} Add /var/www/html/media/vi_7456_20160822_191533.mp4 to Box Queue at pos 19
{2016/08/22 19:18:27} Start boxing /var/www/html/media/vi_7456_20160822_191533.mp4.h264 to /var/www/html/media/vi_7456_20160822_191533.mp4 Queue pos 19
{2016/08/22 19:18:31} send smd 1
[2016/08/22 19:18:42] Start capture requested from Pipe
[2016/08/22 19:18:42] Send ca 1
{2016/08/22 19:18:48} Capturing started
[2016/08/22 19:19:02] Scheduled management tasks. Next at 1471883942
{2016/08/22 19:19:03} send smd 0
{2016/08/22 19:19:07} send smd 1
[2016/08/22 19:19:13] Purged 60 Files
[2016/08/22 19:19:20] Stop capture requested
{2016/08/22 19:19:26} Watchdog detected problem. Stopping[2016/08/22 19:19:29] Send ca 0
[2016/08/22 19:19:31] Start capture requested from Pipe
[2016/08/22 19:19:31] Send ca 1
[2016/08/22 19:19:33] Stop capture requested
[2016/08/22 19:19:33] Send ca 0
{2016/08/22 19:19:38} SIGINT/SIGTERM received, stopping
{2016/08/22 19:19:39} send smd 1
[2016/08/22 19:19:42] Start capture requested from Pipe
[2016/08/22 19:19:42] Send ca 1
[2016/08/22 19:39:03] Scheduled management tasks. Next at 1471885143
[2016/08/22 19:59:04] Scheduled management tasks. Next at 1471886344
[2016/08/22 20:19:05] Scheduled management tasks. Next at 1471887545
[2016/08/22 20:39:06] Scheduled management tasks. Next at 1471888746
```

a-l-e-x-d-s-9, 2016-08-22:
``` log
[2016/08/22 11:39:33] Send ca 1
{2016/08/22 11:39:33} Capturing started
{2016/08/22 11:39:45} Finished boxing /var/www/html/media/vi_74370_20160822_113837.mp4 from Box Queue at pos 6
{2016/08/22 11:39:45} Removed item from Box Queue
{2016/08/22 11:39:47} send smd 0
[2016/08/22 11:39:47] Stop capture requested
[2016/08/22 11:39:47] Send ca 0
{2016/08/22 11:39:48} Capturing stopped
{2016/08/22 11:39:48} Add /var/www/html/media/vi_74371_20160822_113933.mp4 to Box Queue at pos 7
{2016/08/22 11:39:49} Start boxing /var/www/html/media/vi_74371_20160822_113933.mp4.h264 to /var/www/html/media/vi_74371_20160822_113933.mp4 Queue pos 7
{2016/08/22 11:39:51} send smd 1
[2016/08/22 11:39:55] Start capture requested from Pipe
[2016/08/22 11:39:55] Send ca 1
{2016/08/22 11:39:55} Capturing started
{2016/08/22 11:39:56} Finished boxing /var/www/html/media/vi_74371_20160822_113933.mp4 from Box Queue at pos 7
{2016/08/22 11:39:56} Removed item from Box Queue
{2016/08/22 11:39:58} send smd 0
[2016/08/22 11:39:58] Stop capture requested
[2016/08/22 11:39:58] Send ca 0
{2016/08/22 11:39:59} Capturing stopped
{2016/08/22 11:39:59} Add /var/www/html/media/vi_74372_20160822_113955.mp4 to Box Queue at pos 8
{2016/08/22 11:39:59} Start boxing /var/www/html/media/vi_74372_20160822_113955.mp4.h264 to /var/www/html/media/vi_74372_20160822_113955.mp4 Queue pos 8
{2016/08/22 11:40:00} Finished boxing /var/www/html/media/vi_74372_20160822_113955.mp4 from Box Queue at pos 8
{2016/08/22 11:40:00} Removed item from Box Queue
{2016/08/22 11:40:02} send smd 1
[2016/08/22 11:40:02] Start capture requested from Pipe
[2016/08/22 11:40:02] Send ca 1
{2016/08/22 11:40:02} Capturing started
{2016/08/22 11:40:06} send smd 0
[2016/08/22 11:40:06] Stop capture requested
[2016/08/22 11:40:06] Send ca 0
{2016/08/22 11:40:06} Capturing stopped
{2016/08/22 11:40:06} Add /var/www/html/media/vi_74373_20160822_114002.mp4 to Box Queue at pos 9
{2016/08/22 11:40:07} send smd 1
{2016/08/22 11:40:07} Start boxing /var/www/html/media/vi_74373_20160822_114002.mp4.h264 to /var/www/html/media/vi_74373_20160822_114002.mp4 Queue pos 9
[2016/08/22 11:40:08] Start capture requested from Pipe
[2016/08/22 11:40:08] Send ca 1
{2016/08/22 11:40:08} Capturing started
{2016/08/22 11:40:08} Finished boxing /var/www/html/media/vi_74373_20160822_114002.mp4 from Box Queue at pos 9
{2016/08/22 11:40:08} Removed item from Box Queue
{2016/08/22 11:41:04} send smd 0
[2016/08/22 11:41:04] Stop capture requested
[2016/08/22 11:41:04] Send ca 0
{2016/08/22 11:41:04} Capturing stopped
{2016/08/22 11:41:04} Add /var/www/html/media/vi_74374_20160822_114008.mp4 to Box Queue at pos 10
{2016/08/22 11:41:04} Start boxing /var/www/html/media/vi_74374_20160822_114008.mp4.h264 to /var/www/html/media/vi_74374_20160822_114008.mp4 Queue pos 10
{2016/08/22 11:41:06} send smd 1
[2016/08/22 11:41:06] Start capture requested from Pipe
[2016/08/22 11:41:06] Send ca 1
{2016/08/22 11:41:06} Capturing started
{2016/08/22 11:41:39} Finished boxing /var/www/html/media/vi_74374_20160822_114008.mp4 from Box Queue at pos 10
{2016/08/22 11:41:39} Removed item from Box Queue
{2016/08/22 11:41:52} Watchdog detected problem. Stopping{2016/08/22 11:41:56} SIGINT/SIGTERM received, stopping
[2016/08/22 11:52:27] Scheduled management tasks. Next at 1471857147
[2016/08/22 11:52:27] Purged 30 Files
```

a-l-e-x-d-s-9, 2016-08-24:
``` log
[2016/08/24 00:21:20] Send ca 1
{2016/08/24 00:21:20} Capturing started
{2016/08/24 00:21:27} send smd 0
[2016/08/24 00:21:27] Stop capture requested
[2016/08/24 00:21:27] Send ca 0
{2016/08/24 00:21:27} Capturing stopped
{2016/08/24 00:21:27} Add /var/www/html/media/vi_8081_20160824_002120.mp4 to Box Queue at pos 29
{2016/08/24 00:21:27} Start boxing /var/www/html/media/vi_8081_20160824_002120.mp4.h264 to /var/www/html/media/vi_8081_20160824_002120.mp4 Queue pos 29
{2016/08/24 00:21:28} Finished boxing /var/www/html/media/vi_8081_20160824_002120.mp4 from Box Queue at pos 29
{2016/08/24 00:21:28} Removed item from Box Queue
{2016/08/24 00:21:39} send smd 1
[2016/08/24 00:21:39] Start capture requested from Pipe
[2016/08/24 00:21:39] Send ca 1
{2016/08/24 00:21:39} Capturing started
{2016/08/24 00:21:46} send smd 0
[2016/08/24 00:21:46] Stop capture requested
[2016/08/24 00:21:46] Send ca 0
{2016/08/24 00:21:46} Capturing stopped
{2016/08/24 00:21:46} Add /var/www/html/media/vi_8082_20160824_002139.mp4 to Box Queue at pos 30
{2016/08/24 00:21:47} send smd 1
{2016/08/24 00:21:47} Start boxing /var/www/html/media/vi_8082_20160824_002139.mp4.h264 to /var/www/html/media/vi_8082_20160824_002139.mp4 Queue pos 30
{2016/08/24 00:21:48} Finished boxing /var/www/html/media/vi_8082_20160824_002139.mp4 from Box Queue at pos 30
{2016/08/24 00:21:48} Removed item from Box Queue
[2016/08/24 00:21:48] Start capture requested from Pipe
[2016/08/24 00:21:48] Send ca 1
{2016/08/24 00:21:48} Capturing started
{2016/08/24 00:21:55} send smd 0
[2016/08/24 00:21:55] Stop capture requested
[2016/08/24 00:21:55] Send ca 0
{2016/08/24 00:21:55} Capturing stopped
{2016/08/24 00:21:55} Add /var/www/html/media/vi_8083_20160824_002148.mp4 to Box Queue at pos 31
{2016/08/24 00:21:55} Start boxing /var/www/html/media/vi_8083_20160824_002148.mp4.h264 to /var/www/html/media/vi_8083_20160824_002148.mp4 Queue pos 31
{2016/08/24 00:21:56} send smd 1
{2016/08/24 00:21:56} Finished boxing /var/www/html/media/vi_8083_20160824_002148.mp4 from Box Queue at pos 31
{2016/08/24 00:21:56} Removed item from Box Queue
[2016/08/24 00:21:57] Start capture requested from Pipe
[2016/08/24 00:21:57] Send ca 1
{2016/08/24 00:21:57} Capturing started
{2016/08/24 00:22:04} send smd 0
[2016/08/24 00:22:04] Stop capture requested
[2016/08/24 00:22:04] Send ca 0
{2016/08/24 00:22:04} Capturing stopped
{2016/08/24 00:22:04} Add /var/www/html/media/vi_8084_20160824_002157.mp4 to Box Queue at pos 0
{2016/08/24 00:22:05} Start boxing /var/www/html/media/vi_8084_20160824_002157.mp4.h264 to /var/www/html/media/vi_8084_20160824_002157.mp4 Queue pos 0
{2016/08/24 00:22:07} Finished boxing /var/www/html/media/vi_8084_20160824_002157.mp4 from Box Queue at pos 0
{2016/08/24 00:22:07} Removed item from Box Queue
{2016/08/24 00:22:07} send smd 1
[2016/08/24 00:22:07] Start capture requested from Pipe
[2016/08/24 00:22:07] Send ca 1
{2016/08/24 00:22:07} Capturing started
{2016/08/24 00:23:20} send smd 0
[2016/08/24 00:23:20] Stop capture requested
[2016/08/24 00:23:20] Send ca 0
{2016/08/24 00:23:20} Capturing stopped
{2016/08/24 00:23:20} Add /var/www/html/media/vi_8085_20160824_002207.mp4 to Box Queue at pos 1
{2016/08/24 00:23:21} Start boxing /var/www/html/media/vi_8085_20160824_002207.mp4.h264 to /var/www/html/media/vi_8085_20160824_002207.mp4 Queue pos 1
{2016/08/24 00:23:28} send smd 1
[2016/08/24 00:23:35] Start capture requested from Pipe
[2016/08/24 00:23:35] Send ca 1
{2016/08/24 00:23:35} Capturing started
{2016/08/24 00:23:51} Finished boxing /var/www/html/media/vi_8085_20160824_002207.mp4 from Box Queue at pos 1
{2016/08/24 00:23:51} Removed item from Box Queue
{2016/08/24 00:24:02} send smd 0
[2016/08/24 00:24:02] Stop capture requested
[2016/08/24 00:24:02] Send ca 0
{2016/08/24 00:24:02} Capturing stopped
{2016/08/24 00:24:02} Add /var/www/html/media/vi_8086_20160824_002335.mp4 to Box Queue at pos 2
{2016/08/24 00:24:03} Start boxing /var/www/html/media/vi_8086_20160824_002335.mp4.h264 to /var/www/html/media/vi_8086_20160824_002335.mp4 Queue pos 2
{2016/08/24 00:24:05} Finished boxing /var/www/html/media/vi_8086_20160824_002335.mp4 from Box Queue at pos 2
{2016/08/24 00:24:05} Removed item from Box Queue
{2016/08/24 00:24:06} send smd 1
[2016/08/24 00:24:06] Start capture requested from Pipe
[2016/08/24 00:24:06] Send ca 1
{2016/08/24 00:24:06} Capturing started
{2016/08/24 00:26:12} send smd 0
[2016/08/24 00:26:12] Stop capture requested
[2016/08/24 00:26:12] Send ca 0
{2016/08/24 00:26:12} Capturing stopped
{2016/08/24 00:26:12} Add /var/www/html/media/vi_8087_20160824_002406.mp4 to Box Queue at pos 3
{2016/08/24 00:26:12} Start boxing /var/www/html/media/vi_8087_20160824_002406.mp4.h264 to /var/www/html/media/vi_8087_20160824_002406.mp4 Queue pos 3
{2016/08/24 00:26:13} send smd 1
[2016/08/24 00:26:14] Start capture requested from Pipe
[2016/08/24 00:26:14] Send ca 1
{2016/08/24 00:26:14} Capturing started
{2016/08/24 00:26:48} Watchdog detected problem. Stopping{2016/08/24 00:26:50} SIGINT/SIGTERM received, stopping
[2016/08/24 00:30:38] Scheduled management tasks. Next at 1471989038
[2016/08/24 00:30:38] Purged 57 Files
[2016/08/24 00:50:39] Scheduled management tasks. Next at 1471990239
```

a-l-e-x-d-s-9, 2016-08-25:
``` log 
[2016/08/25 09:57:57] Send ca 1
{2016/08/25 09:57:57} Capturing started
{2016/08/25 09:57:58} Finished boxing /var/www/html/media/vi_81311_20160825_095752.mp4 from Box Queue at pos 14
{2016/08/25 09:57:58} Removed item from Box Queue
{2016/08/25 09:58:05} send smd 0
[2016/08/25 09:58:05] Stop capture requested
[2016/08/25 09:58:05] Send ca 0
{2016/08/25 09:58:06} Capturing stopped
{2016/08/25 09:58:06} Add /var/www/html/media/vi_81312_20160825_095757.mp4 to Box Queue at pos 15
{2016/08/25 09:58:06} Start boxing /var/www/html/media/vi_81312_20160825_095757.mp4.h264 to /var/www/html/media/vi_81312_20160825_095757.mp4 Queue pos 15
{2016/08/25 09:58:07} Finished boxing /var/www/html/media/vi_81312_20160825_095757.mp4 from Box Queue at pos 15
{2016/08/25 09:58:07} Removed item from Box Queue
{2016/08/25 09:58:08} send smd 1
[2016/08/25 09:58:08] Start capture requested from Pipe
[2016/08/25 09:58:08] Send ca 1
{2016/08/25 09:58:08} Capturing started
{2016/08/25 09:58:12} send smd 0
[2016/08/25 09:58:12] Stop capture requested
[2016/08/25 09:58:12] Send ca 0
{2016/08/25 09:58:12} Capturing stopped
{2016/08/25 09:58:12} Add /var/www/html/media/vi_81313_20160825_095808.mp4 to Box Queue at pos 16
{2016/08/25 09:58:12} Start boxing /var/www/html/media/vi_81313_20160825_095808.mp4.h264 to /var/www/html/media/vi_81313_20160825_095808.mp4 Queue pos 16
{2016/08/25 09:58:13} Finished boxing /var/www/html/media/vi_81313_20160825_095808.mp4 from Box Queue at pos 16
{2016/08/25 09:58:13} Removed item from Box Queue
{2016/08/25 09:58:14} send smd 1
[2016/08/25 09:58:14] Start capture requested from Pipe
[2016/08/25 09:58:14] Send ca 1
{2016/08/25 09:58:14} Capturing started
{2016/08/25 09:58:18} send smd 0
[2016/08/25 09:58:18] Stop capture requested
[2016/08/25 09:58:18] Send ca 0
{2016/08/25 09:58:18} Capturing stopped
{2016/08/25 09:58:18} Add /var/www/html/media/vi_81314_20160825_095814.mp4 to Box Queue at pos 17
{2016/08/25 09:58:18} Start boxing /var/www/html/media/vi_81314_20160825_095814.mp4.h264 to /var/www/html/media/vi_81314_20160825_095814.mp4 Queue pos 17
{2016/08/25 09:58:19} Finished boxing /var/www/html/media/vi_81314_20160825_095814.mp4 from Box Queue at pos 17
{2016/08/25 09:58:19} Removed item from Box Queue
{2016/08/25 09:58:22} send smd 1
[2016/08/25 09:58:22] Start capture requested from Pipe
[2016/08/25 09:58:22] Send ca 1
{2016/08/25 09:58:22} Capturing started
{2016/08/25 09:58:26} send smd 0
[2016/08/25 09:58:26] Stop capture requested
[2016/08/25 09:58:26] Send ca 0
{2016/08/25 09:58:26} Capturing stopped
{2016/08/25 09:58:26} Add /var/www/html/media/vi_81315_20160825_095822.mp4 to Box Queue at pos 18
{2016/08/25 09:58:27} Start boxing /var/www/html/media/vi_81315_20160825_095822.mp4.h264 to /var/www/html/media/vi_81315_20160825_095822.mp4 Queue pos 18
{2016/08/25 09:58:28} Finished boxing /var/www/html/media/vi_81315_20160825_095822.mp4 from Box Queue at pos 18
{2016/08/25 09:58:28} Removed item from Box Queue
{2016/08/25 09:58:31} send smd 1
[2016/08/25 09:58:31] Start capture requested from Pipe
[2016/08/25 09:58:31] Send ca 1
{2016/08/25 09:58:31} Capturing started
{2016/08/25 09:58:35} send smd 0
[2016/08/25 09:58:35] Stop capture requested
[2016/08/25 09:58:35] Send ca 0
{2016/08/25 09:58:35} Capturing stopped
{2016/08/25 09:58:35} Add /var/www/html/media/vi_81316_20160825_095831.mp4 to Box Queue at pos 19
{2016/08/25 09:58:36} Start boxing /var/www/html/media/vi_81316_20160825_095831.mp4.h264 to /var/www/html/media/vi_81316_20160825_095831.mp4 Queue pos 19
{2016/08/25 09:58:37} Finished boxing /var/www/html/media/vi_81316_20160825_095831.mp4 from Box Queue at pos 19
{2016/08/25 09:58:37} Removed item from Box Queue
{2016/08/25 09:58:39} send smd 1
[2016/08/25 09:58:39] Start capture requested from Pipe
[2016/08/25 09:58:39] Send ca 1
{2016/08/25 09:58:39} Capturing started
{2016/08/25 09:58:42} send smd 0
[2016/08/25 09:58:42] Stop capture requested
[2016/08/25 09:58:42] Send ca 0
{2016/08/25 09:58:43} Capturing stopped
{2016/08/25 09:58:43} Add /var/www/html/media/vi_81317_20160825_095839.mp4 to Box Queue at pos 20
[... 59 lines cut ...]
{2016/08/25 10:01:47} Add /var/www/html/media/vi_81322_20160825_095925.mp4 to Box Queue at pos 25
{2016/08/25 10:01:47} Start boxing /var/www/html/media/vi_81322_20160825_095925.mp4.h264 to /var/www/html/media/vi_81322_20160825_095925.mp4 Queue pos 25
{2016/08/25 10:01:49} send smd 1
[2016/08/25 10:01:49] Start capture requested from Pipe
[2016/08/25 10:01:49] Send ca 1
{2016/08/25 10:01:49} Capturing started
{2016/08/25 10:02:36} Watchdog detected problem. Stopping{2016/08/25 10:02:43} send smd 0
{2016/08/25 10:02:44} SIGINT/SIGTERM received, stopping
[2016/08/25 10:02:53] Stop capture requested
[2016/08/25 10:02:53] Send ca 0
[2016/08/25 10:10:12] Scheduled management tasks. Next at 1472110212
[2016/08/25 10:10:12] Purged 33 Files
[2016/08/25 10:30:13] Scheduled management tasks. Next at 1472111413
[2016/08/25 10:50:14] Scheduled management tasks. Next at 1472112614
[2016/08/25 11:10:15] Scheduled management tasks. Next at 1472113815
[2016/08/25 11:30:16] Scheduled management tasks. Next at 1472115016
[2016/08/25 11:50:17] Scheduled management tasks. Next at 1472116217
[2016/08/25 12:10:18] Scheduled management tasks. Next at 1472117418
[2016/08/25 12:30:19] Scheduled management tasks. Next at 1472118619
[2016/08/25 12:50:20] Scheduled management tasks. Next at 1472119820
[2016/08/25 13:10:21] Scheduled management tasks. Next at 1472121021
[2016/08/25 13:30:22] Scheduled management tasks. Next at 1472122222
[2016/08/25 13:50:23] Scheduled management tasks. Next at 1472123423
[2016/08/25 14:10:24] Scheduled management tasks. Next at 1472124624
```

a-l-e-x-d-s-9, 2016-08-25:
Stuck without Watchdog on:
[image]

``` log
[2016/08/25 20:01:19] Send ca 1
{2016/08/25 20:01:19} Capturing started
{2016/08/25 20:01:26} send smd 0
[2016/08/25 20:01:26] Stop capture requested
[2016/08/25 20:01:26] Send ca 0
{2016/08/25 20:01:26} Capturing stopped
{2016/08/25 20:01:26} Add /var/www/html/media/vi_8828_20160825_200119.mp4 to Box Queue at pos 9
{2016/08/25 20:01:26} Start boxing /var/www/html/media/vi_8828_20160825_200119.mp4.h264 to /var/www/html/media/vi_8828_20160825_200119.mp4 Queue pos 9
{2016/08/25 20:01:27} Finished boxing /var/www/html/media/vi_8828_20160825_200119.mp4 from Box Queue at pos 9
{2016/08/25 20:01:27} Removed item from Box Queue
[2016/08/25 20:02:10] Scheduled management tasks. Next at 1472145730
[2016/08/25 20:02:10] Purged 2 Files
{2016/08/25 20:05:20} send smd 1
[2016/08/25 20:05:20] Start capture requested from Pipe
[2016/08/25 20:05:20] Send ca 1
{2016/08/25 20:05:20} Capturing started
{2016/08/25 20:05:26} send smd 0
[2016/08/25 20:05:26] Stop capture requested
[2016/08/25 20:05:26] Send ca 0
{2016/08/25 20:05:27} Capturing stopped
{2016/08/25 20:05:27} Add /var/www/html/media/vi_8829_20160825_200520.mp4 to Box Queue at pos 10
{2016/08/25 20:05:27} Start boxing /var/www/html/media/vi_8829_20160825_200520.mp4.h264 to /var/www/html/media/vi_8829_20160825_200520.mp4 Queue pos 10
{2016/08/25 20:05:28} Finished boxing /var/www/html/media/vi_8829_20160825_200520.mp4 from Box Queue at pos 10
{2016/08/25 20:05:28} Removed item from Box Queue
{2016/08/25 20:14:00} send smd 1
[2016/08/25 20:14:00] Start capture requested from Pipe
[2016/08/25 20:14:00] Send ca 1
{2016/08/25 20:14:00} Capturing started
{2016/08/25 20:14:26} send smd 0
[2016/08/25 20:14:26] Stop capture requested
[2016/08/25 20:14:26] Send ca 0
{2016/08/25 20:14:26} Capturing stopped
{2016/08/25 20:14:26} Add /var/www/html/media/vi_8830_20160825_201400.mp4 to Box Queue at pos 11
{2016/08/25 20:14:27} Start boxing /var/www/html/media/vi_8830_20160825_201400.mp4.h264 to /var/www/html/media/vi_8830_20160825_201400.mp4 Queue pos 11
{2016/08/25 20:14:27} send smd 1
[2016/08/25 20:14:28] Start capture requested from Pipe
[2016/08/25 20:14:28] Send ca 1
{2016/08/25 20:14:28} Capturing started
{2016/08/25 20:14:30} Finished boxing /var/www/html/media/vi_8830_20160825_201400.mp4 from Box Queue at pos 11
{2016/08/25 20:14:30} Removed item from Box Queue
{2016/08/25 20:14:47} send smd 0
[2016/08/25 20:14:47] Stop capture requested
[2016/08/25 20:14:47] Send ca 0
{2016/08/25 20:14:47} Capturing stopped
{2016/08/25 20:14:47} Add /var/www/html/media/vi_8831_20160825_201428.mp4 to Box Queue at pos 12
{2016/08/25 20:14:48} Start boxing /var/www/html/media/vi_8831_20160825_201428.mp4.h264 to /var/www/html/media/vi_8831_20160825_201428.mp4 Queue pos 12
{2016/08/25 20:14:52} Finished boxing /var/www/html/media/vi_8831_20160825_201428.mp4 from Box Queue at pos 12
{2016/08/25 20:14:52} Removed item from Box Queue
{2016/08/25 20:15:59} send smd 1
[2016/08/25 20:15:59] Start capture requested from Pipe
[2016/08/25 20:15:59] Send ca 1
{2016/08/25 20:15:59} Capturing started
{2016/08/25 20:16:07} send smd 0
[2016/08/25 20:16:07] Stop capture requested
[2016/08/25 20:16:07] Send ca 0
{2016/08/25 20:16:07} Capturing stopped
{2016/08/25 20:16:07} Add /var/www/html/media/vi_8832_20160825_201559.mp4 to Box Queue at pos 13
{2016/08/25 20:16:07} Start boxing /var/www/html/media/vi_8832_20160825_201559.mp4.h264 to /var/www/html/media/vi_8832_20160825_201559.mp4 Queue pos 13
{2016/08/25 20:16:08} Finished boxing /var/www/html/media/vi_8832_20160825_201559.mp4 from Box Queue at pos 13
{2016/08/25 20:16:08} Removed item from Box Queue
{2016/08/25 20:16:22} send smd 1
[2016/08/25 20:16:22] Start capture requested from Pipe
[2016/08/25 20:16:22] Send ca 1
{2016/08/25 20:16:22} Capturing started
{2016/08/25 20:16:39} send smd 0
[2016/08/25 20:16:39] Stop capture requested
[... 613 lines cut ...]
{2016/08/25 21:41:25} Removed item from Box Queue
{2016/08/25 21:42:02} send smd 1
[2016/08/25 21:42:02] Start capture requested from Pipe
[2016/08/25 21:42:02] Send ca 1
{2016/08/25 21:42:02} Capturing started
[2016/08/25 21:42:15] Scheduled management tasks. Next at 1472151735
[2016/08/25 21:42:15] Purged 3 Files
{2016/08/25 21:43:56} send smd 0
[2016/08/25 21:43:56] Stop capture requested
[2016/08/25 21:43:56] Send ca 0
{2016/08/25 21:43:56} Capturing stopped
{2016/08/25 21:43:56} Add /var/www/html/media/vi_8884_20160825_214202.mp4 to Box Queue at pos 1
{2016/08/25 21:43:56} send smd 1
{2016/08/25 21:43:56} Start boxing /var/www/html/media/vi_8884_20160825_214202.mp4.h264 to /var/www/html/media/vi_8884_20160825_214202.mp4 Queue pos 1
[2016/08/25 21:43:58] Start capture requested from Pipe
[2016/08/25 21:43:58] Send ca 1
{2016/08/25 21:43:58} Capturing started
{2016/08/25 21:44:44} send smd 0
[2016/08/25 21:44:44] Stop capture requested
[2016/08/25 21:44:44] Send ca 0
[2016/08/25 22:02:16] Scheduled management tasks. Next at 1472152936
[2016/08/25 22:02:16] Purged 16 Files
[2016/08/25 22:22:17] Scheduled management tasks. Next at 1472154137

```

a-l-e-x-d-s-9, 2016-08-30:
``` log
[2016/08/30 13:01:18] Send ca 0
{2016/08/30 13:01:18} Capturing stopped
{2016/08/30 13:01:18} Add /var/www/html/media/vi_93449_20160830_130109.mp4 to Box Queue at pos 10
{2016/08/30 13:01:18} Start boxing /var/www/html/media/vi_93449_20160830_130109.mp4.h264 to /var/www/html/media/vi_93449_20160830_130109.mp4 Queue pos 10
{2016/08/30 13:01:20} Finished boxing /var/www/html/media/vi_93449_20160830_130109.mp4 from Box Queue at pos 10
{2016/08/30 13:01:20} Removed item from Box Queue
{2016/08/30 13:01:27} send smd 1
[2016/08/30 13:01:27] Start capture requested from Pipe
[2016/08/30 13:01:27] Send ca 1
{2016/08/30 13:01:27} Capturing started
{2016/08/30 13:01:30} send smd 0
[2016/08/30 13:01:30] Stop capture requested
[2016/08/30 13:01:30] Send ca 0
{2016/08/30 13:01:31} Capturing stopped
{2016/08/30 13:01:31} Add /var/www/html/media/vi_93450_20160830_130127.mp4 to Box Queue at pos 11
{2016/08/30 13:01:31} Start boxing /var/www/html/media/vi_93450_20160830_130127.mp4.h264 to /var/www/html/media/vi_93450_20160830_130127.mp4 Queue pos 11
{2016/08/30 13:01:32} Finished boxing /var/www/html/media/vi_93450_20160830_130127.mp4 from Box Queue at pos 11
{2016/08/30 13:01:32} Removed item from Box Queue
{2016/08/30 13:02:57} send smd 1
[2016/08/30 13:02:57] Start capture requested from Pipe
[2016/08/30 13:02:57] Send ca 1
{2016/08/30 13:02:57} Capturing started
{2016/08/30 13:03:00} send smd 0
[2016/08/30 13:03:00] Stop capture requested
[2016/08/30 13:03:00] Send ca 0
{2016/08/30 13:03:01} Capturing stopped
{2016/08/30 13:03:01} Add /var/www/html/media/vi_93451_20160830_130257.mp4 to Box Queue at pos 12
{2016/08/30 13:03:01} Start boxing /var/www/html/media/vi_93451_20160830_130257.mp4.h264 to /var/www/html/media/vi_93451_20160830_130257.mp4 Queue pos 12
{2016/08/30 13:03:02} Finished boxing /var/www/html/media/vi_93451_20160830_130257.mp4 from Box Queue at pos 12
{2016/08/30 13:03:02} Removed item from Box Queue
{2016/08/30 13:03:21} send smd 1
[2016/08/30 13:03:21] Start capture requested from Pipe
[2016/08/30 13:03:21] Send ca 1
{2016/08/30 13:03:21} Capturing started
{2016/08/30 13:03:24} send smd 0
[2016/08/30 13:03:24] Stop capture requested
[2016/08/30 13:03:24] Send ca 0
{2016/08/30 13:03:25} Capturing stopped
{2016/08/30 13:03:25} Add /var/www/html/media/vi_93452_20160830_130321.mp4 to Box Queue at pos 13
{2016/08/30 13:03:25} Start boxing /var/www/html/media/vi_93452_20160830_130321.mp4.h264 to /var/www/html/media/vi_93452_20160830_130321.mp4 Queue pos 13
{2016/08/30 13:03:26} send smd 1
{2016/08/30 13:03:26} Finished boxing /var/www/html/media/vi_93452_20160830_130321.mp4 from Box Queue at pos 13
{2016/08/30 13:03:26} Removed item from Box Queue
[2016/08/30 13:03:26] Start capture requested from Pipe
[2016/08/30 13:03:26] Send ca 1
{2016/08/30 13:03:27} Capturing started
{2016/08/30 13:03:30} send smd 0
[2016/08/30 13:03:30] Stop capture requested
[2016/08/30 13:03:30] Send ca 0
{2016/08/30 13:03:30} Capturing stopped
{2016/08/30 13:03:30} Add /var/www/html/media/vi_93453_20160830_130327.mp4 to Box Queue at pos 14
{2016/08/30 13:03:31} Start boxing /var/www/html/media/vi_93453_20160830_130327.mp4.h264 to /var/www/html/media/vi_93453_20160830_130327.mp4 Queue pos 14
{2016/08/30 13:03:31} send smd 1
{2016/08/30 13:03:32} Finished boxing /var/www/html/media/vi_93453_20160830_130327.mp4 from Box Queue at pos 14
{2016/08/30 13:03:32} Removed item from Box Queue
[2016/08/30 13:03:32] Start capture requested from Pipe
[2016/08/30 13:03:32] Send ca 1
{2016/08/30 13:03:32} Capturing started
{2016/08/30 13:03:35} send smd 0
[2016/08/30 13:03:35] Stop capture requested
[2016/08/30 13:03:35] Send ca 0
{2016/08/30 13:03:35} Capturing stopped
{2016/08/30 13:03:35} Add /var/www/html/media/vi_93454_20160830_130332.mp4 to Box Queue at pos 15
{2016/08/30 13:03:36} Start boxing /var/www/html/media/vi_93454_20160830_130332.mp4.h264 to /var/www/html/media/vi_93454_20160830_130332.mp4 Queue pos 15
{2016/08/30 13:03:36} send smd 1
{2016/08/30 13:03:37} Finished boxing /var/www/html/media/vi_93454_20160830_130332.mp4 from Box Queue at pos 15
{2016/08/30 13:03:37} Removed item from Box Queue
[2016/08/30 13:03:37] Start capture requested from Pipe
[2016/08/30 13:03:37] Send ca 1
[... 687 lines cut ...]
[2016/08/30 13:16:22] Send ca 0
{2016/08/30 13:16:22} Capturing stopped
{2016/08/30 13:16:22} Add /var/www/html/media/vi_93512_20160830_131618.mp4 to Box Queue at pos 9
{2016/08/30 13:16:23} Start boxing /var/www/html/media/vi_93512_20160830_131618.mp4.h264 to /var/www/html/media/vi_93512_20160830_131618.mp4 Queue pos 9
{2016/08/30 13:16:24} Finished boxing /var/www/html/media/vi_93512_20160830_131618.mp4 from Box Queue at pos 9
{2016/08/30 13:16:24} Removed item from Box Queue
{2016/08/30 13:16:32} send smd 1
[2016/08/30 13:16:32] Start capture requested from Pipe
[2016/08/30 13:16:32] Send ca 1
{2016/08/30 13:16:32} Capturing started
{2016/08/30 13:18:10} send smd 0
[2016/08/30 13:18:10] Stop capture requested
[2016/08/30 13:18:10] Send ca 0
{2016/08/30 13:18:10} Capturing stopped
{2016/08/30 13:18:10} Add /var/www/html/media/vi_93513_20160830_131632.mp4 to Box Queue at pos 10
{2016/08/30 13:18:11} Start boxing /var/www/html/media/vi_93513_20160830_131632.mp4.h264 to /var/www/html/media/vi_93513_20160830_131632.mp4 Queue pos 10
{2016/08/30 13:18:29} send smd 1
[2016/08/30 13:18:29] Start capture requested from Pipe
[2016/08/30 13:18:29] Send ca 1
{2016/08/30 13:18:29} Capturing started
{2016/08/30 13:18:57} Watchdog detected problem. Stopping{2016/08/30 13:18:58} SIGINT/SIGTERM received, stopping
[2016/08/30 13:19:12] Scheduled management tasks. Next at 1472553552
[2016/08/30 13:19:25] Purged 41 Files
[2016/08/30 13:39:13] Scheduled management tasks. Next at 1472554753
```

a-l-e-x-d-s-9, 2016-08-30:
``` log
[2016/08/30 14:09:32] Send ca 1
{2016/08/30 14:09:32} Capturing started
{2016/08/30 14:09:36} send smd 0
[2016/08/30 14:09:36] Stop capture requested
[2016/08/30 14:09:36] Send ca 0
{2016/08/30 14:09:36} Capturing stopped
{2016/08/30 14:09:36} Add /var/www/html/media/vi_93612_20160830_140932.mp4 to Box Queue at pos 1
{2016/08/30 14:09:37} Start boxing /var/www/html/media/vi_93612_20160830_140932.mp4.h264 to /var/www/html/media/vi_93612_20160830_140932.mp4 Queue pos 1
{2016/08/30 14:09:37} send smd 1
{2016/08/30 14:09:38} Finished boxing /var/www/html/media/vi_93612_20160830_140932.mp4 from Box Queue at pos 1
{2016/08/30 14:09:38} Removed item from Box Queue
[2016/08/30 14:09:38] Start capture requested from Pipe
[2016/08/30 14:09:38] Send ca 1
{2016/08/30 14:09:38} Capturing started
{2016/08/30 14:09:42} send smd 0
[2016/08/30 14:09:42] Stop capture requested
[2016/08/30 14:09:42] Send ca 0
{2016/08/30 14:09:42} Capturing stopped
{2016/08/30 14:09:42} Add /var/www/html/media/vi_93613_20160830_140938.mp4 to Box Queue at pos 2
{2016/08/30 14:09:43} Start boxing /var/www/html/media/vi_93613_20160830_140938.mp4.h264 to /var/www/html/media/vi_93613_20160830_140938.mp4 Queue pos 2
{2016/08/30 14:09:44} send smd 1
[2016/08/30 14:09:44] Start capture requested from Pipe
[2016/08/30 14:09:44] Send ca 1
{2016/08/30 14:09:44} Capturing started
{2016/08/30 14:09:44} Finished boxing /var/www/html/media/vi_93613_20160830_140938.mp4 from Box Queue at pos 2
{2016/08/30 14:09:44} Removed item from Box Queue
{2016/08/30 14:09:48} send smd 0
[2016/08/30 14:09:48] Stop capture requested
[2016/08/30 14:09:48] Send ca 0
{2016/08/30 14:09:48} Capturing stopped
{2016/08/30 14:09:48} Add /var/www/html/media/vi_93614_20160830_140944.mp4 to Box Queue at pos 3
{2016/08/30 14:09:48} Start boxing /var/www/html/media/vi_93614_20160830_140944.mp4.h264 to /var/www/html/media/vi_93614_20160830_140944.mp4 Queue pos 3
{2016/08/30 14:09:49} Finished boxing /var/www/html/media/vi_93614_20160830_140944.mp4 from Box Queue at pos 3
{2016/08/30 14:09:49} Removed item from Box Queue
{2016/08/30 14:09:50} send smd 1
[2016/08/30 14:09:50] Start capture requested from Pipe
[2016/08/30 14:09:50] Send ca 1
{2016/08/30 14:09:51} Capturing started
{2016/08/30 14:09:54} send smd 0
[2016/08/30 14:09:54] Stop capture requested
[2016/08/30 14:09:54] Send ca 0
{2016/08/30 14:09:54} Capturing stopped
{2016/08/30 14:09:54} Add /var/www/html/media/vi_93615_20160830_140951.mp4 to Box Queue at pos 4
{2016/08/30 14:09:54} Start boxing /var/www/html/media/vi_93615_20160830_140951.mp4.h264 to /var/www/html/media/vi_93615_20160830_140951.mp4 Queue pos 4
{2016/08/30 14:09:56} Finished boxing /var/www/html/media/vi_93615_20160830_140951.mp4 from Box Queue at pos 4
{2016/08/30 14:09:56} Removed item from Box Queue
{2016/08/30 14:10:03} send smd 1
[2016/08/30 14:10:03] Start capture requested from Pipe
[2016/08/30 14:10:03] Send ca 1
{2016/08/30 14:10:04} Capturing started
{2016/08/30 14:10:16} send smd 0
[2016/08/30 14:10:16] Stop capture requested
[2016/08/30 14:10:16] Send ca 0
{2016/08/30 14:10:16} Capturing stopped
{2016/08/30 14:10:16} Add /var/www/html/media/vi_93616_20160830_141004.mp4 to Box Queue at pos 5
{2016/08/30 14:10:17} Start boxing /var/www/html/media/vi_93616_20160830_141004.mp4.h264 to /var/www/html/media/vi_93616_20160830_141004.mp4 Queue pos 5
{2016/08/30 14:10:18} Finished boxing /var/www/html/media/vi_93616_20160830_141004.mp4 from Box Queue at pos 5
{2016/08/30 14:10:18} Removed item from Box Queue
{2016/08/30 14:10:59} send smd 1
[2016/08/30 14:10:59] Start capture requested from Pipe
[2016/08/30 14:10:59] Send ca 1
{2016/08/30 14:10:59} Capturing started
{2016/08/30 14:11:07} send smd 0
[2016/08/30 14:11:07] Stop capture requested
[2016/08/30 14:11:07] Send ca 0
{2016/08/30 14:11:07} Capturing stopped
{2016/08/30 14:11:07} Add /var/www/html/media/vi_93617_20160830_141059.mp4 to Box Queue at pos 6
{2016/08/30 14:11:07} Start boxing /var/www/html/media/vi_93617_20160830_141059.mp4.h264 to /var/www/html/media/vi_93617_20160830_141059.mp4 Queue pos 6
{2016/08/30 14:11:09} Finished boxing /var/www/html/media/vi_93617_20160830_141059.mp4 from Box Queue at pos 6
[... 214 lines cut ...]
[2016/08/30 14:16:15] Start capture requested from Pipe
[2016/08/30 14:16:15] Send ca 1
{2016/08/30 14:16:15} Capturing started
{2016/08/30 14:16:21} Finished boxing /var/www/html/media/vi_93634_20160830_141554.mp4 from Box Queue at pos 0
{2016/08/30 14:16:21} Removed item from Box Queue
{2016/08/30 14:16:43} send smd 0
[2016/08/30 14:16:43] Stop capture requested
[2016/08/30 14:16:43] Send ca 0
{2016/08/30 14:16:43} Capturing stopped
{2016/08/30 14:16:43} Add /var/www/html/media/vi_93635_20160830_141615.mp4 to Box Queue at pos 1
{2016/08/30 14:16:44} Start boxing /var/www/html/media/vi_93635_20160830_141615.mp4.h264 to /var/www/html/media/vi_93635_20160830_141615.mp4 Queue pos 1
{2016/08/30 14:16:44} send smd 1
[2016/08/30 14:16:45] Start capture requested from Pipe
[2016/08/30 14:16:45] Send ca 1
{2016/08/30 14:16:45} Capturing started
{2016/08/30 14:16:49} send smd 0
[2016/08/30 14:16:49] Stop capture requested
[2016/08/30 14:16:49] Send ca 0
{2016/08/30 14:16:54} send smd 1
[2016/08/30 14:16:58] Start capture requested from Pipe
[2016/08/30 14:16:58] Send ca 1
[2016/08/30 14:35:46] Scheduled management tasks. Next at 1472558146
[2016/08/30 14:55:47] Scheduled management tasks. Next at 1472559347
[2016/08/30 15:15:48] Scheduled management tasks. Next at 1472560548
```

a-l-e-x-d-s-9, 2016-08-30:
``` log
{2016/08/30 15:23:31} RaspiMJPEG Version 5.6.5
{2016/08/30 15:23:32} recommended video buffers 1
{2016/08/30 15:23:32} h264 buffers set to recommended 1
{2016/08/30 15:23:32} Set up internal detect width=121 height=68
{2016/08/30 15:23:32} MJPEG streaming, ready to receive commands
{2016/08/30 15:23:32} send smd 9
{2016/08/30 15:23:32} send smd 1
[2016/08/30 15:23:36] RaspiCam support started
[2016/08/30 15:23:36] Capture Pipe already exists /var/www/html/FIFO1
[2016/08/30 15:23:36] Scheduler loop is started
{2016/08/30 15:23:36} Opening FIFO 0 /var/www/html/FIFO 7
{2016/08/30 15:23:36} Opening FIFO 1 /var/www/html/FIFO11 8
{2016/08/30 15:23:36} Starting command loop
[2016/08/30 15:23:36] Start capture request ignored, day period not initialised yet
[2016/08/30 15:23:36] Reload parameters command requested
[2016/08/30 15:23:36] Scheduler loop is started
[2016/08/30 15:23:37] New period detected 0
[2016/08/30 15:23:37] Scheduled management tasks. Next at 1472561017
{2016/08/30 15:25:23} send smd 0
[2016/08/30 15:25:23] Stop capture request ignored, already stopped
{2016/08/30 15:25:24} send smd 1
[2016/08/30 15:25:24] Start capture requested from Pipe
[2016/08/30 15:25:24] Send ca 1
{2016/08/30 15:25:24} Capturing started
{2016/08/30 15:25:28} send smd 0
[2016/08/30 15:25:28] Stop capture requested
[2016/08/30 15:25:28] Send ca 0
{2016/08/30 15:25:28} Capturing stopped
{2016/08/30 15:25:28} Add /var/www/html/media/vi_93637_20160830_152524.mp4 to Box Queue at pos 0
{2016/08/30 15:25:29} Start boxing /var/www/html/media/vi_93637_20160830_152524.mp4.h264 to /var/www/html/media/vi_93637_20160830_152524.mp4 Queue pos 0
{2016/08/30 15:25:29} send smd 1
{2016/08/30 15:25:30} Finished boxing /var/www/html/media/vi_93637_20160830_152524.mp4 from Box Queue at pos 0
{2016/08/30 15:25:30} Removed item from Box Queue
[2016/08/30 15:25:30] Start capture requested from Pipe
[2016/08/30 15:25:30] Send ca 1
{2016/08/30 15:25:30} Capturing started
{2016/08/30 15:25:47} send smd 0
[2016/08/30 15:25:47] Stop capture requested
[2016/08/30 15:25:47] Send ca 0
{2016/08/30 15:25:47} Capturing stopped
{2016/08/30 15:25:47} Add /var/www/html/media/vi_93638_20160830_152530.mp4 to Box Queue at pos 1
{2016/08/30 15:25:47} send smd 1
{2016/08/30 15:25:48} Start boxing /var/www/html/media/vi_93638_20160830_152530.mp4.h264 to /var/www/html/media/vi_93638_20160830_152530.mp4 Queue pos 1
[2016/08/30 15:25:49] Start capture requested from Pipe
[2016/08/30 15:25:49] Send ca 1
{2016/08/30 15:25:49} Capturing started
{2016/08/30 15:25:56} Finished boxing /var/www/html/media/vi_93638_20160830_152530.mp4 from Box Queue at pos 1
{2016/08/30 15:25:56} Removed item from Box Queue
{2016/08/30 15:27:53} send smd 0
[2016/08/30 15:27:53] Stop capture requested
[2016/08/30 15:27:53] Send ca 0
{2016/08/30 15:27:53} Capturing stopped
{2016/08/30 15:27:53} Add /var/www/html/media/vi_93639_20160830_152549.mp4 to Box Queue at pos 2
{2016/08/30 15:27:54} Start boxing /var/www/html/media/vi_93639_20160830_152549.mp4.h264 to /var/www/html/media/vi_93639_20160830_152549.mp4 Queue pos 2
{2016/08/30 15:27:56} send smd 1
[2016/08/30 15:27:56] Start capture requested from Pipe
[2016/08/30 15:27:56] Send ca 1
{2016/08/30 15:27:56} Capturing started
{2016/08/30 15:28:01} send smd 0
[2016/08/30 15:28:01] Stop capture requested
[2016/08/30 15:28:01] Send ca 0
{2016/08/30 15:28:02} Capturing stopped
{2016/08/30 15:28:02} Add /var/www/html/media/vi_93640_20160830_152756.mp4 to Box Queue at pos 3
{2016/08/30 15:28:04} send smd 1
[2016/08/30 15:28:08] Start capture requested from Pipe
[2016/08/30 15:28:08] Send ca 1
{2016/08/30 15:28:08} Capturing started
{2016/08/30 15:28:22} send smd 0
[2016/08/30 15:28:22] Stop capture requested
[... 190 lines cut ...]
{2016/08/30 15:38:12} Finished boxing /var/www/html/media/vi_93653_20160830_153746.mp4 from Box Queue at pos 3
{2016/08/30 15:38:12} Start boxing /var/www/html/media/vi_93654_20160830_153752.mp4.h264 to /var/www/html/media/vi_93654_20160830_153752.mp4 Queue pos 4
{2016/08/30 15:38:12} Removed item from Box Queue
{2016/08/30 15:38:13} Finished boxing /var/www/html/media/vi_93654_20160830_153752.mp4 from Box Queue at pos 4
{2016/08/30 15:38:13} Removed item from Box Queue
{2016/08/30 15:38:49} send smd 0
[2016/08/30 15:38:49] Stop capture requested
[2016/08/30 15:38:49] Send ca 0
{2016/08/30 15:38:50} Capturing stopped
{2016/08/30 15:38:50} Add /var/www/html/media/vi_93655_20160830_153803.mp4 to Box Queue at pos 5
{2016/08/30 15:38:50} Start boxing /var/www/html/media/vi_93655_20160830_153803.mp4.h264 to /var/www/html/media/vi_93655_20160830_153803.mp4 Queue pos 5
{2016/08/30 15:38:50} send smd 1
[2016/08/30 15:38:52] Start capture requested from Pipe
[2016/08/30 15:38:52] Send ca 1
{2016/08/30 15:38:52} Capturing started
{2016/08/30 15:39:07} Finished boxing /var/www/html/media/vi_93655_20160830_153803.mp4 from Box Queue at pos 5
{2016/08/30 15:39:07} Removed item from Box Queue
{2016/08/30 15:39:17} Watchdog detected problem. Stopping{2016/08/30 15:39:17} SIGINT/SIGTERM received, stopping
[2016/08/30 15:52:26] Scheduled management tasks. Next at 1472562746
[2016/08/30 16:12:27] Scheduled management tasks. Next at 1472563947
[2016/08/30 16:32:28] Scheduled management tasks. Next at 1472565148
[2016/08/30 16:52:29] Scheduled management tasks. Next at 1472566349
[2016/08/30 17:12:30] Scheduled management tasks. Next at 1472567550
[2016/08/30 17:32:31] Scheduled management tasks. Next at 1472568751
```

a-l-e-x-d-s-9, 2016-08-31:
``` log 
[2016/08/31 12:00:34] Send ca 1
{2016/08/31 12:00:34} Capturing started
{2016/08/31 12:00:53} send smd 0
[2016/08/31 12:00:53] Stop capture requested
[2016/08/31 12:00:53] Send ca 0
{2016/08/31 12:00:54} Capturing stopped
{2016/08/31 12:00:54} Add /var/www/html/media/vi_144176_20160831_120034.mp4 to Box Queue at pos 5
{2016/08/31 12:00:54} Start boxing /var/www/html/media/vi_144176_20160831_120034.mp4.h264 to /var/www/html/media/vi_144176_20160831_120034.mp4 Queue pos 5
{2016/08/31 12:00:55} Finished boxing /var/www/html/media/vi_144176_20160831_120034.mp4 from Box Queue at pos 5
{2016/08/31 12:00:55} Removed item from Box Queue
{2016/08/31 12:01:32} send smd 1
[2016/08/31 12:01:32] Start capture requested from Pipe
[2016/08/31 12:01:32] Send ca 1
{2016/08/31 12:01:32} Capturing started
{2016/08/31 12:01:39} send smd 0
[2016/08/31 12:01:39] Stop capture requested
[2016/08/31 12:01:39] Send ca 0
{2016/08/31 12:01:39} Capturing stopped
{2016/08/31 12:01:39} Add /var/www/html/media/vi_144177_20160831_120132.mp4 to Box Queue at pos 6
{2016/08/31 12:01:40} Start boxing /var/www/html/media/vi_144177_20160831_120132.mp4.h264 to /var/www/html/media/vi_144177_20160831_120132.mp4 Queue pos 6
{2016/08/31 12:01:41} Finished boxing /var/www/html/media/vi_144177_20160831_120132.mp4 from Box Queue at pos 6
{2016/08/31 12:01:41} Removed item from Box Queue
{2016/08/31 12:02:09} send smd 1
[2016/08/31 12:02:09] Start capture requested from Pipe
[2016/08/31 12:02:09] Send ca 1
{2016/08/31 12:02:09} Capturing started
{2016/08/31 12:02:17} send smd 0
[2016/08/31 12:02:17] Stop capture requested
[2016/08/31 12:02:17] Send ca 0
{2016/08/31 12:02:17} Capturing stopped
{2016/08/31 12:02:17} Add /var/www/html/media/vi_144178_20160831_120209.mp4 to Box Queue at pos 7
{2016/08/31 12:02:18} Start boxing /var/www/html/media/vi_144178_20160831_120209.mp4.h264 to /var/www/html/media/vi_144178_20160831_120209.mp4 Queue pos 7
{2016/08/31 12:02:19} Finished boxing /var/www/html/media/vi_144178_20160831_120209.mp4 from Box Queue at pos 7
{2016/08/31 12:02:19} Removed item from Box Queue
{2016/08/31 12:03:18} send smd 1
[2016/08/31 12:03:18] Start capture requested from Pipe
[2016/08/31 12:03:18] Send ca 1
{2016/08/31 12:03:18} Capturing started
{2016/08/31 12:03:26} send smd 0
[2016/08/31 12:03:26] Stop capture requested
[2016/08/31 12:03:26] Send ca 0
{2016/08/31 12:03:26} Capturing stopped
{2016/08/31 12:03:26} Add /var/www/html/media/vi_144179_20160831_120318.mp4 to Box Queue at pos 8
{2016/08/31 12:03:27} Start boxing /var/www/html/media/vi_144179_20160831_120318.mp4.h264 to /var/www/html/media/vi_144179_20160831_120318.mp4 Queue pos 8
{2016/08/31 12:03:28} Finished boxing /var/www/html/media/vi_144179_20160831_120318.mp4 from Box Queue at pos 8
{2016/08/31 12:03:28} Removed item from Box Queue
{2016/08/31 12:03:58} send smd 1
[2016/08/31 12:04:00] Start capture requested from Pipe
[2016/08/31 12:04:00] Send ca 1
{2016/08/31 12:04:01} Capturing started
{2016/08/31 12:04:09} send smd 0
[2016/08/31 12:04:09] Stop capture requested
[2016/08/31 12:04:09] Send ca 0
{2016/08/31 12:04:09} Capturing stopped
{2016/08/31 12:04:09} Add /var/www/html/media/vi_144180_20160831_120400.mp4 to Box Queue at pos 9
{2016/08/31 12:04:09} Start boxing /var/www/html/media/vi_144180_20160831_120400.mp4.h264 to /var/www/html/media/vi_144180_20160831_120400.mp4 Queue pos 9
{2016/08/31 12:04:10} Finished boxing /var/www/html/media/vi_144180_20160831_120400.mp4 from Box Queue at pos 9
{2016/08/31 12:04:10} Removed item from Box Queue
{2016/08/31 12:04:46} send smd 1
[2016/08/31 12:04:46] Start capture requested from Pipe
[2016/08/31 12:04:46] Send ca 1
{2016/08/31 12:04:46} Capturing started
{2016/08/31 12:04:53} send smd 0
[2016/08/31 12:04:53] Stop capture requested
[2016/08/31 12:04:53] Send ca 0
{2016/08/31 12:04:54} Capturing stopped
{2016/08/31 12:04:54} Add /var/www/html/media/vi_144181_20160831_120446.mp4 to Box Queue at pos 10
{2016/08/31 12:04:54} Start boxing /var/www/html/media/vi_144181_20160831_120446.mp4.h264 to /var/www/html/media/vi_144181_20160831_120446.mp4 Queue pos 10
{2016/08/31 12:04:55} Finished boxing /var/www/html/media/vi_144181_20160831_120446.mp4 from Box Queue at pos 10
[... 238 lines cut ...]
{2016/08/31 12:43:21} Add /var/www/html/media/vi_144200_20160831_124255.mp4 to Box Queue at pos 1
{2016/08/31 12:43:15} send smd 1
[2016/08/31 12:43:21] Start capture requested from Pipe
[2016/08/31 12:43:21] Send ca 1
{2016/08/31 12:43:26} Capturing started
{2016/08/31 12:43:35} send smd 0
[2016/08/31 12:43:35] Stop capture requested
[2016/08/31 12:43:35] Send ca 0
[2016/08/31 13:00:29] Scheduled management tasks. Next at 1472638829
[2016/08/31 13:00:29] Purged 3 Files
[2016/08/31 13:20:30] Scheduled management tasks. Next at 1472640030
[2016/08/31 13:40:31] Scheduled management tasks. Next at 1472641231
[2016/08/31 14:00:32] Scheduled management tasks. Next at 1472642432
[2016/08/31 14:20:33] Scheduled management tasks. Next at 1472643633
[2016/08/31 14:40:34] Scheduled management tasks. Next at 1472644834
[2016/08/31 15:00:35] Scheduled management tasks. Next at 1472646035
[2016/08/31 15:20:36] Scheduled management tasks. Next at 1472647236
[2016/08/31 15:40:37] Scheduled management tasks. Next at 1472648437
[2016/08/31 16:00:38] Scheduled management tasks. Next at 1472649638
[2016/08/31 16:20:39] Scheduled management tasks. Next at 1472650839
[2016/08/31 16:40:40] Scheduled management tasks. Next at 1472652040
[2016/08/31 17:00:41] Scheduled management tasks. Next at 1472653241
[2016/08/31 17:20:42] Scheduled management tasks. Next at 1472654442
[2016/08/31 17:40:43] Scheduled management tasks. Next at 1472655643
```

a-l-e-x-d-s-9, 2016-09-02:
``` log
[2016/09/01 18:37:50] Send ca 1
{2016/09/01 18:37:50} Capturing started
{2016/09/01 18:37:56} send smd 0
[2016/09/01 18:37:56] Stop capture requested
[2016/09/01 18:37:56] Send ca 0
{2016/09/01 18:37:56} Capturing stopped
{2016/09/01 18:37:56} Add /var/www/html/media/vi_12216_20160901_183750.mp4 to Box Queue at pos 11
{2016/09/01 18:37:56} Start boxing /var/www/html/media/vi_12216_20160901_183750.mp4.h264 to /var/www/html/media/vi_12216_20160901_183750.mp4 Queue pos 11
{2016/09/01 18:37:57} Finished boxing /var/www/html/media/vi_12216_20160901_183750.mp4 from Box Queue at pos 11
{2016/09/01 18:37:57} Removed item from Box Queue
{2016/09/01 18:38:00} send smd 1
[2016/09/01 18:38:00] Start capture requested from Pipe
[2016/09/01 18:38:00] Send ca 1
{2016/09/01 18:38:00} Capturing started
{2016/09/01 18:38:06} send smd 0
[2016/09/01 18:38:06] Stop capture requested
[2016/09/01 18:38:06] Send ca 0
{2016/09/01 18:38:06} Capturing stopped
{2016/09/01 18:38:06} Add /var/www/html/media/vi_12217_20160901_183800.mp4 to Box Queue at pos 12
{2016/09/01 18:38:07} Start boxing /var/www/html/media/vi_12217_20160901_183800.mp4.h264 to /var/www/html/media/vi_12217_20160901_183800.mp4 Queue pos 12
{2016/09/01 18:38:08} Finished boxing /var/www/html/media/vi_12217_20160901_183800.mp4 from Box Queue at pos 12
{2016/09/01 18:38:08} Removed item from Box Queue
{2016/09/01 18:43:42} send smd 1
[2016/09/01 18:43:42] Start capture requested from Pipe
[2016/09/01 18:43:42] Send ca 1
{2016/09/01 18:43:42} Capturing started
{2016/09/01 18:43:51} send smd 0
[2016/09/01 18:43:51] Stop capture requested
[2016/09/01 18:43:51] Send ca 0
{2016/09/01 18:43:51} Capturing stopped
{2016/09/01 18:43:51} Add /var/www/html/media/vi_12218_20160901_184342.mp4 to Box Queue at pos 13
{2016/09/01 18:43:52} send smd 1
{2016/09/01 18:43:52} Start boxing /var/www/html/media/vi_12218_20160901_184342.mp4.h264 to /var/www/html/media/vi_12218_20160901_184342.mp4 Queue pos 13
{2016/09/01 18:43:53} Finished boxing /var/www/html/media/vi_12218_20160901_184342.mp4 from Box Queue at pos 13
{2016/09/01 18:43:53} Removed item from Box Queue
[2016/09/01 18:43:53] Start capture requested from Pipe
[2016/09/01 18:43:53] Send ca 1
{2016/09/01 18:43:53} Capturing started
{2016/09/01 18:43:59} send smd 0
[2016/09/01 18:43:59] Stop capture requested
[2016/09/01 18:43:59] Send ca 0
{2016/09/01 18:43:59} Capturing stopped
{2016/09/01 18:43:59} Add /var/www/html/media/vi_12219_20160901_184353.mp4 to Box Queue at pos 14
{2016/09/01 18:44:00} Start boxing /var/www/html/media/vi_12219_20160901_184353.mp4.h264 to /var/www/html/media/vi_12219_20160901_184353.mp4 Queue pos 14
{2016/09/01 18:44:01} Finished boxing /var/www/html/media/vi_12219_20160901_184353.mp4 from Box Queue at pos 14
{2016/09/01 18:44:01} Removed item from Box Queue
[2016/09/01 18:45:43] Scheduled management tasks. Next at 1472745943
[2016/09/01 18:45:43] Purged 1 Files
{2016/09/01 18:55:09} send smd 1
[2016/09/01 18:55:09] Start capture requested from Pipe
[2016/09/01 18:55:09] Send ca 1
{2016/09/01 18:55:09} Capturing started
{2016/09/01 18:56:02} send smd 0
[2016/09/01 18:56:02] Stop capture requested
[2016/09/01 18:56:02] Send ca 0
{2016/09/01 18:56:02} Capturing stopped
{2016/09/01 18:56:02} Add /var/www/html/media/vi_12220_20160901_185509.mp4 to Box Queue at pos 15
{2016/09/01 18:56:03} Start boxing /var/www/html/media/vi_12220_20160901_185509.mp4.h264 to /var/www/html/media/vi_12220_20160901_185509.mp4 Queue pos 15
{2016/09/01 18:56:18} Finished boxing /var/www/html/media/vi_12220_20160901_185509.mp4 from Box Queue at pos 15
{2016/09/01 18:56:18} Removed item from Box Queue
[2016/09/01 19:05:44] Scheduled management tasks. Next at 1472747144
[2016/09/01 19:05:44] Purged 1 Files
{2016/09/01 19:10:19} send smd 1
[2016/09/01 19:10:19] Start capture requested from Pipe
[2016/09/01 19:10:19] Send ca 1
{2016/09/01 19:10:19} Capturing started
{2016/09/01 19:12:24} send smd 0
[2016/09/01 19:12:24] Stop capture requested
[2016/09/01 19:12:24] Send ca 0
[... 115 lines cut ...]
{2016/09/01 19:22:14} Capturing started
{2016/09/01 19:22:21} send smd 0
[2016/09/01 19:22:31] Stop capture requested
[2016/09/01 19:22:31] Send ca 0
{2016/09/01 19:22:31} send smd 1
{2016/09/01 19:22:43} Capturing stopped
{2016/09/01 19:22:43} Add /var/www/html/media/vi_12230_20160901_192214.mp4 to Box Queue at pos 7
{2016/09/01 19:22:43} Reset motion state
[2016/09/01 19:22:43] Start capture requested from Pipe
[2016/09/01 19:22:43] Send ca 1
{2016/09/01 19:22:43} Capturing started
{2016/09/01 19:22:43} send smd 1
[2016/09/01 19:22:45] Start capture requested from Pipe
[2016/09/01 19:22:45] Send ca 1
{2016/09/01 19:22:51} Already capturing. Ignore 
{2016/09/01 19:23:33} Finished boxing /var/www/html/media/vi_12229_20160901_191942.mp4 from Box Queue at pos 6
{2016/09/01 19:23:33} Start boxing /var/www/html/media/vi_12230_20160901_192214.mp4.h264 to /var/www/html/media/vi_12230_20160901_192214.mp4 Queue pos 7
{2016/09/01 19:23:33} Removed item from Box Queue
{2016/09/01 19:23:46} Watchdog detected problem. Stopping{2016/09/01 19:23:51} SIGINT/SIGTERM received, stopping
[2016/09/01 19:33:48] Scheduled management tasks. Next at 1472748828
[2016/09/01 19:33:48] Purged 1 Files
[2016/09/01 19:53:49] Scheduled management tasks. Next at 1472750029
[2016/09/01 20:13:50] Scheduled management tasks. Next at 1472751230
[2016/09/01 20:33:51] Scheduled management tasks. Next at 1472752431
```

a-l-e-x-d-s-9, 2016-09-02:
``` log
[2016/09/02 15:27:27] Send ca 1
{2016/09/02 15:27:27} Capturing started
{2016/09/02 15:27:39} send smd 0
[2016/09/02 15:27:39] Stop capture requested
[2016/09/02 15:27:39] Send ca 0
{2016/09/02 15:27:39} Capturing stopped
{2016/09/02 15:27:39} Add /var/www/html/media/vi_12681_20160902_152727.mp4 to Box Queue at pos 3
{2016/09/02 15:27:40} Start boxing /var/www/html/media/vi_12681_20160902_152727.mp4.h264 to /var/www/html/media/vi_12681_20160902_152727.mp4 Queue pos 3
{2016/09/02 15:27:41} Finished boxing /var/www/html/media/vi_12681_20160902_152727.mp4 from Box Queue at pos 3
{2016/09/02 15:27:41} Removed item from Box Queue
{2016/09/02 15:27:50} send smd 1
[2016/09/02 15:27:50] Start capture requested from Pipe
[2016/09/02 15:27:50] Send ca 1
{2016/09/02 15:27:50} Capturing started
{2016/09/02 15:28:06} send smd 0
[2016/09/02 15:28:06] Stop capture requested
[2016/09/02 15:28:06] Send ca 0
{2016/09/02 15:28:06} Capturing stopped
{2016/09/02 15:28:06} Add /var/www/html/media/vi_12682_20160902_152750.mp4 to Box Queue at pos 4
{2016/09/02 15:28:06} Start boxing /var/www/html/media/vi_12682_20160902_152750.mp4.h264 to /var/www/html/media/vi_12682_20160902_152750.mp4 Queue pos 4
{2016/09/02 15:28:07} Finished boxing /var/www/html/media/vi_12682_20160902_152750.mp4 from Box Queue at pos 4
{2016/09/02 15:28:07} Removed item from Box Queue
{2016/09/02 15:28:15} send smd 1
[2016/09/02 15:28:15] Start capture requested from Pipe
[2016/09/02 15:28:15] Send ca 1
{2016/09/02 15:28:16} Capturing started
{2016/09/02 15:28:22} send smd 0
[2016/09/02 15:28:22] Stop capture requested
[2016/09/02 15:28:22] Send ca 0
{2016/09/02 15:28:22} Capturing stopped
{2016/09/02 15:28:22} Add /var/www/html/media/vi_12683_20160902_152816.mp4 to Box Queue at pos 5
{2016/09/02 15:28:22} Start boxing /var/www/html/media/vi_12683_20160902_152816.mp4.h264 to /var/www/html/media/vi_12683_20160902_152816.mp4 Queue pos 5
{2016/09/02 15:28:23} Finished boxing /var/www/html/media/vi_12683_20160902_152816.mp4 from Box Queue at pos 5
{2016/09/02 15:28:23} Removed item from Box Queue
{2016/09/02 15:28:58} send smd 1
[2016/09/02 15:28:58] Start capture requested from Pipe
[2016/09/02 15:28:58] Send ca 1
{2016/09/02 15:28:59} Capturing started
{2016/09/02 15:29:05} send smd 0
[2016/09/02 15:29:05] Stop capture requested
[2016/09/02 15:29:05] Send ca 0
{2016/09/02 15:29:05} Capturing stopped
{2016/09/02 15:29:05} Add /var/www/html/media/vi_12684_20160902_152859.mp4 to Box Queue at pos 6
{2016/09/02 15:29:06} Start boxing /var/www/html/media/vi_12684_20160902_152859.mp4.h264 to /var/www/html/media/vi_12684_20160902_152859.mp4 Queue pos 6
{2016/09/02 15:29:07} Finished boxing /var/www/html/media/vi_12684_20160902_152859.mp4 from Box Queue at pos 6
{2016/09/02 15:29:07} Removed item from Box Queue
{2016/09/02 15:29:43} send smd 1
[2016/09/02 15:29:43] Start capture requested from Pipe
[2016/09/02 15:29:43] Send ca 1
{2016/09/02 15:29:43} Capturing started
{2016/09/02 15:29:49} send smd 0
[2016/09/02 15:29:49] Stop capture requested
[2016/09/02 15:29:49] Send ca 0
{2016/09/02 15:29:49} Capturing stopped
{2016/09/02 15:29:49} Add /var/www/html/media/vi_12685_20160902_152943.mp4 to Box Queue at pos 7
{2016/09/02 15:29:50} Start boxing /var/www/html/media/vi_12685_20160902_152943.mp4.h264 to /var/www/html/media/vi_12685_20160902_152943.mp4 Queue pos 7
{2016/09/02 15:29:51} Finished boxing /var/www/html/media/vi_12685_20160902_152943.mp4 from Box Queue at pos 7
{2016/09/02 15:29:51} Removed item from Box Queue
{2016/09/02 15:29:52} send smd 1
[2016/09/02 15:29:52] Start capture requested from Pipe
[2016/09/02 15:29:52] Send ca 1
{2016/09/02 15:29:52} Capturing started
{2016/09/02 15:30:04} send smd 0
[2016/09/02 15:30:04] Stop capture requested
[2016/09/02 15:30:04] Send ca 0
{2016/09/02 15:30:04} Capturing stopped
{2016/09/02 15:30:04} Add /var/www/html/media/vi_12686_20160902_152952.mp4 to Box Queue at pos 8
{2016/09/02 15:30:04} Start boxing /var/www/html/media/vi_12686_20160902_152952.mp4.h264 to /var/www/html/media/vi_12686_20160902_152952.mp4 Queue pos 8
{2016/09/02 15:30:05} Finished boxing /var/www/html/media/vi_12686_20160902_152952.mp4 from Box Queue at pos 8
[... 174 lines cut ...]
{2016/09/02 15:59:38} send smd 0
[2016/09/02 15:59:38] Stop capture requested
[2016/09/02 15:59:38] Send ca 0
{2016/09/02 15:59:38} Capturing stopped
{2016/09/02 15:59:38} Add /var/www/html/media/vi_12701_20160902_155906.mp4 to Box Queue at pos 23
{2016/09/02 15:59:39} Start boxing /var/www/html/media/vi_12701_20160902_155906.mp4.h264 to /var/www/html/media/vi_12701_20160902_155906.mp4 Queue pos 23
{2016/09/02 15:59:41} Finished boxing /var/www/html/media/vi_12701_20160902_155906.mp4 from Box Queue at pos 23
{2016/09/02 15:59:41} Removed item from Box Queue
{2016/09/02 16:00:12} send smd 1
[2016/09/02 16:00:12] Start capture requested from Pipe
[2016/09/02 16:00:12] Send ca 1
{2016/09/02 16:00:12} Capturing started
[2016/09/02 16:01:20] Scheduled management tasks. Next at 1472822480
{2016/09/02 16:04:11} send smd 0
[2016/09/02 16:04:11] Stop capture requested
[2016/09/02 16:04:11] Send ca 0
{2016/09/02 16:04:11} Capturing stopped
{2016/09/02 16:04:11} Add /var/www/html/media/vi_12702_20160902_160012.mp4 to Box Queue at pos 24
{2016/09/02 16:04:11} Start boxing /var/www/html/media/vi_12702_20160902_160012.mp4.h264 to /var/www/html/media/vi_12702_20160902_160012.mp4 Queue pos 24
{2016/09/02 16:04:16} send smd 1
[2016/09/02 16:04:16] Start capture requested from Pipe
[2016/09/02 16:04:16] Send ca 1
{2016/09/02 16:04:16} Capturing started
{2016/09/02 16:04:52} Watchdog detected problem. Stopping{2016/09/02 16:04:59} SIGINT/SIGTERM received, stopping
```

a-l-e-x-d-s-9, 2016-09-03:
``` log 
[2016/09/03 21:57:06] Send ca 1
{2016/09/03 21:57:06} Capturing started
{2016/09/03 21:57:09} send smd 0
[2016/09/03 21:57:09] Stop capture requested
[2016/09/03 21:57:09] Send ca 0
{2016/09/03 21:57:10} Capturing stopped
{2016/09/03 21:57:10} Add /var/www/html/media/vi_102073_20160903_215706.mp4 to Box Queue at pos 5
{2016/09/03 21:57:10} Start boxing /var/www/html/media/vi_102073_20160903_215706.mp4.h264 to /var/www/html/media/vi_102073_20160903_215706.mp4 Queue pos 5
{2016/09/03 21:57:11} Finished boxing /var/www/html/media/vi_102073_20160903_215706.mp4 from Box Queue at pos 5
{2016/09/03 21:57:11} Removed item from Box Queue
{2016/09/03 21:57:14} send smd 1
[2016/09/03 21:57:14] Start capture requested from Pipe
[2016/09/03 21:57:14] Send ca 1
{2016/09/03 21:57:14} Capturing started
{2016/09/03 21:57:17} send smd 0
[2016/09/03 21:57:17] Stop capture requested
[2016/09/03 21:57:17] Send ca 0
{2016/09/03 21:57:17} Capturing stopped
{2016/09/03 21:57:17} Add /var/www/html/media/vi_102074_20160903_215714.mp4 to Box Queue at pos 6
{2016/09/03 21:57:18} Start boxing /var/www/html/media/vi_102074_20160903_215714.mp4.h264 to /var/www/html/media/vi_102074_20160903_215714.mp4 Queue pos 6
{2016/09/03 21:57:19} Finished boxing /var/www/html/media/vi_102074_20160903_215714.mp4 from Box Queue at pos 6
{2016/09/03 21:57:19} Removed item from Box Queue
{2016/09/03 21:57:20} send smd 1
[2016/09/03 21:57:21] Start capture requested from Pipe
[2016/09/03 21:57:21] Send ca 1
{2016/09/03 21:57:21} Capturing started
{2016/09/03 21:57:24} send smd 0
[2016/09/03 21:57:24] Stop capture requested
[2016/09/03 21:57:24] Send ca 0
{2016/09/03 21:57:24} Capturing stopped
{2016/09/03 21:57:24} Add /var/www/html/media/vi_102075_20160903_215721.mp4 to Box Queue at pos 7
{2016/09/03 21:57:25} Start boxing /var/www/html/media/vi_102075_20160903_215721.mp4.h264 to /var/www/html/media/vi_102075_20160903_215721.mp4 Queue pos 7
{2016/09/03 21:57:26} Finished boxing /var/www/html/media/vi_102075_20160903_215721.mp4 from Box Queue at pos 7
{2016/09/03 21:57:26} Removed item from Box Queue
{2016/09/03 21:57:28} send smd 1
[2016/09/03 21:57:28] Start capture requested from Pipe
[2016/09/03 21:57:28] Send ca 1
{2016/09/03 21:57:28} Capturing started
{2016/09/03 21:57:37} send smd 0
[2016/09/03 21:57:37] Stop capture requested
[2016/09/03 21:57:37] Send ca 0
{2016/09/03 21:57:37} Capturing stopped
{2016/09/03 21:57:37} Add /var/www/html/media/vi_102076_20160903_215728.mp4 to Box Queue at pos 8
{2016/09/03 21:57:38} Start boxing /var/www/html/media/vi_102076_20160903_215728.mp4.h264 to /var/www/html/media/vi_102076_20160903_215728.mp4 Queue pos 8
{2016/09/03 21:57:39} Finished boxing /var/www/html/media/vi_102076_20160903_215728.mp4 from Box Queue at pos 8
{2016/09/03 21:57:39} Removed item from Box Queue
{2016/09/03 21:57:40} send smd 1
[2016/09/03 21:57:40] Start capture requested from Pipe
[2016/09/03 21:57:40] Send ca 1
{2016/09/03 21:57:40} Capturing started
{2016/09/03 21:57:44} send smd 0
[2016/09/03 21:57:44] Stop capture requested
[2016/09/03 21:57:44] Send ca 0
{2016/09/03 21:57:44} Capturing stopped
{2016/09/03 21:57:44} Add /var/www/html/media/vi_102077_20160903_215740.mp4 to Box Queue at pos 9
{2016/09/03 21:57:44} Start boxing /var/www/html/media/vi_102077_20160903_215740.mp4.h264 to /var/www/html/media/vi_102077_20160903_215740.mp4 Queue pos 9
{2016/09/03 21:57:45} Finished boxing /var/www/html/media/vi_102077_20160903_215740.mp4 from Box Queue at pos 9
{2016/09/03 21:57:45} Removed item from Box Queue
{2016/09/03 21:57:45} send smd 1
[2016/09/03 21:57:46] Start capture requested from Pipe
[2016/09/03 21:57:46] Send ca 1
{2016/09/03 21:57:46} Capturing started
{2016/09/03 21:57:49} send smd 0
[2016/09/03 21:57:49] Stop capture requested
[2016/09/03 21:57:49] Send ca 0
{2016/09/03 21:57:49} Capturing stopped
{2016/09/03 21:57:49} Add /var/www/html/media/vi_102078_20160903_215746.mp4 to Box Queue at pos 10
{2016/09/03 21:57:49} Start boxing /var/www/html/media/vi_102078_20160903_215746.mp4.h264 to /var/www/html/media/vi_102078_20160903_215746.mp4 Queue pos 10
{2016/09/03 21:57:50} Finished boxing /var/www/html/media/vi_102078_20160903_215746.mp4 from Box Queue at pos 10
[... 199 lines cut ...]
[2016/09/03 22:00:37] Send ca 0
{2016/09/03 22:00:37} Capturing stopped
{2016/09/03 22:00:37} Add /var/www/html/media/vi_102095_20160903_215939.mp4 to Box Queue at pos 27
{2016/09/03 22:00:38} send smd 1
{2016/09/03 22:00:38} Start boxing /var/www/html/media/vi_102095_20160903_215939.mp4.h264 to /var/www/html/media/vi_102095_20160903_215939.mp4 Queue pos 27
[2016/09/03 22:00:39] Start capture requested from Pipe
[2016/09/03 22:00:39] Send ca 1
{2016/09/03 22:00:39} Capturing started
{2016/09/03 22:01:03} Finished boxing /var/www/html/media/vi_102095_20160903_215939.mp4 from Box Queue at pos 27
{2016/09/03 22:01:03} Removed item from Box Queue
{2016/09/03 22:03:01} send smd 0
[2016/09/03 22:03:01] Stop capture requested
[2016/09/03 22:03:01] Send ca 0
{2016/09/03 22:03:02} Capturing stopped
{2016/09/03 22:03:02} Add /var/www/html/media/vi_102096_20160903_220039.mp4 to Box Queue at pos 28
{2016/09/03 22:03:02} Start boxing /var/www/html/media/vi_102096_20160903_220039.mp4.h264 to /var/www/html/media/vi_102096_20160903_220039.mp4 Queue pos 28
{2016/09/03 22:03:03} send smd 1
[2016/09/03 22:03:03] Start capture requested from Pipe
[2016/09/03 22:03:03] Send ca 1
{2016/09/03 22:03:03} Capturing started
{2016/09/03 22:03:57} Watchdog detected problem. Stopping{2016/09/03 22:04:04} SIGINT/SIGTERM received, stopping
[2016/09/03 22:15:36] Scheduled management tasks. Next at 1472931336
[2016/09/03 22:15:36] Purged 31 Files
[2016/09/03 22:35:37] Scheduled management tasks. Next at 1472932537
```

a-l-e-x-d-s-9, 2016-09-04:
``` log
[2016/09/04 12:34:59] Send ca 1
{2016/09/04 12:34:59} Capturing started
{2016/09/04 12:35:19} send smd 0
[2016/09/04 12:35:19] Stop capture requested
[2016/09/04 12:35:19] Send ca 0
{2016/09/04 12:35:19} Capturing stopped
{2016/09/04 12:35:19} Add /var/www/html/media/vi_103379_20160904_123459.mp4 to Box Queue at pos 10
{2016/09/04 12:35:20} send smd 1
[2016/09/04 12:35:21] Start capture requested from Pipe
[2016/09/04 12:35:21] Send ca 1
{2016/09/04 12:35:29} Capturing started
{2016/09/04 12:35:32} send smd 0
[2016/09/04 12:35:32] Stop capture requested
[2016/09/04 12:35:32] Send ca 0
{2016/09/04 12:35:32} Capturing stopped
{2016/09/04 12:35:32} Add /var/www/html/media/vi_103380_20160904_123528.mp4 to Box Queue at pos 11
{2016/09/04 12:35:36} Finished boxing /var/www/html/media/vi_103378_20160904_123344.mp4 from Box Queue at pos 9
{2016/09/04 12:35:40} Start boxing /var/www/html/media/vi_103379_20160904_123459.mp4.h264 to /var/www/html/media/vi_103379_20160904_123459.mp4 Queue pos 10
{2016/09/04 12:35:40} Removed item from Box Queue
{2016/09/04 12:35:44} Finished boxing /var/www/html/media/vi_103379_20160904_123459.mp4 from Box Queue at pos 10
{2016/09/04 12:35:44} Start boxing /var/www/html/media/vi_103380_20160904_123528.mp4.h264 to /var/www/html/media/vi_103380_20160904_123528.mp4 Queue pos 11
{2016/09/04 12:35:44} Removed item from Box Queue
{2016/09/04 12:35:45} Finished boxing /var/www/html/media/vi_103380_20160904_123528.mp4 from Box Queue at pos 11
{2016/09/04 12:35:45} Removed item from Box Queue
[2016/09/04 12:39:22] Scheduled management tasks. Next at 1472983162
[2016/09/04 12:39:22] Purged 39 Files
{2016/09/04 12:54:33} send smd 1
[2016/09/04 12:54:33] Start capture requested from Pipe
[2016/09/04 12:54:33] Send ca 1
{2016/09/04 12:54:33} Capturing started
{2016/09/04 12:54:36} send smd 0
[2016/09/04 12:54:36] Stop capture requested
[2016/09/04 12:54:36] Send ca 0
{2016/09/04 12:54:37} Capturing stopped
{2016/09/04 12:54:37} Add /var/www/html/media/vi_103381_20160904_125433.mp4 to Box Queue at pos 12
{2016/09/04 12:54:37} Start boxing /var/www/html/media/vi_103381_20160904_125433.mp4.h264 to /var/www/html/media/vi_103381_20160904_125433.mp4 Queue pos 12
{2016/09/04 12:54:38} Finished boxing /var/www/html/media/vi_103381_20160904_125433.mp4 from Box Queue at pos 12
{2016/09/04 12:54:38} Removed item from Box Queue
{2016/09/04 12:55:10} send smd 1
[2016/09/04 12:55:10] Start capture requested from Pipe
[2016/09/04 12:55:10] Send ca 1
{2016/09/04 12:55:10} Capturing started
{2016/09/04 12:55:13} send smd 0
[2016/09/04 12:55:13] Stop capture requested
[2016/09/04 12:55:13] Send ca 0
{2016/09/04 12:55:13} Capturing stopped
{2016/09/04 12:55:13} Add /var/www/html/media/vi_103382_20160904_125510.mp4 to Box Queue at pos 13
{2016/09/04 12:55:13} Start boxing /var/www/html/media/vi_103382_20160904_125510.mp4.h264 to /var/www/html/media/vi_103382_20160904_125510.mp4 Queue pos 13
{2016/09/04 12:55:15} Finished boxing /var/www/html/media/vi_103382_20160904_125510.mp4 from Box Queue at pos 13
{2016/09/04 12:55:15} Removed item from Box Queue
{2016/09/04 12:55:30} send smd 1
[2016/09/04 12:55:30] Start capture requested from Pipe
[2016/09/04 12:55:30] Send ca 1
{2016/09/04 12:55:30} Capturing started
{2016/09/04 12:55:34} send smd 0
[2016/09/04 12:55:34] Stop capture requested
[2016/09/04 12:55:34] Send ca 0
{2016/09/04 12:55:34} Capturing stopped
{2016/09/04 12:55:34} Add /var/www/html/media/vi_103383_20160904_125530.mp4 to Box Queue at pos 14
{2016/09/04 12:55:35} Start boxing /var/www/html/media/vi_103383_20160904_125530.mp4.h264 to /var/www/html/media/vi_103383_20160904_125530.mp4 Queue pos 14
{2016/09/04 12:55:36} Finished boxing /var/www/html/media/vi_103383_20160904_125530.mp4 from Box Queue at pos 14
{2016/09/04 12:55:36} Removed item from Box Queue
{2016/09/04 12:57:43} send smd 1
[2016/09/04 12:57:43] Start capture requested from Pipe
[2016/09/04 12:57:43] Send ca 1
{2016/09/04 12:57:44} Capturing started
{2016/09/04 12:57:47} send smd 0
[2016/09/04 12:57:47] Stop capture requested
[2016/09/04 12:57:47] Send ca 0
[... 116 lines cut ...]
[2016/09/04 13:19:24] Purged 27 Files
{2016/09/04 13:20:20} send smd 1
[2016/09/04 13:20:20] Start capture requested from Pipe
[2016/09/04 13:20:20] Send ca 1
{2016/09/04 13:20:20} Capturing started
{2016/09/04 13:20:58} send smd 0
[2016/09/04 13:20:58] Stop capture requested
[2016/09/04 13:20:58] Send ca 0
{2016/09/04 13:20:58} Capturing stopped
{2016/09/04 13:20:58} Add /var/www/html/media/vi_103394_20160904_132020.mp4 to Box Queue at pos 25
{2016/09/04 13:20:59} Start boxing /var/www/html/media/vi_103394_20160904_132020.mp4.h264 to /var/www/html/media/vi_103394_20160904_132020.mp4 Queue pos 25
{2016/09/04 13:21:00} send smd 1
[2016/09/04 13:21:00] Start capture requested from Pipe
[2016/09/04 13:21:00] Send ca 1
{2016/09/04 13:21:00} Capturing started
{2016/09/04 13:21:09} send smd 0
[2016/09/04 13:21:13] Stop capture requested
[2016/09/04 13:21:13] Send ca 0
[2016/09/04 13:39:25] Scheduled management tasks. Next at 1472986765
[2016/09/04 13:39:25] Purged 13 Files
[2016/09/04 13:59:26] Scheduled management tasks. Next at 1472987966
[2016/09/04 14:19:27] Scheduled management tasks. Next at 1472989167
[2016/09/04 14:39:28] Scheduled management tasks. Next at 1472990368
[2016/09/04 14:59:29] Scheduled management tasks. Next at 14729915691
```

a-l-e-x-d-s-9, 2016-09-04:
``` log
[2016/09/04 20:53:39] Send ca 1
{2016/09/04 20:53:39} Capturing started
{2016/09/04 20:53:48} send smd 0
[2016/09/04 20:53:48] Stop capture requested
[2016/09/04 20:53:48] Send ca 0
{2016/09/04 20:53:48} Capturing stopped
{2016/09/04 20:53:48} Add /var/www/html/media/vi_13988_20160904_205339.mp4 to Box Queue at pos 18
{2016/09/04 20:53:49} Start boxing /var/www/html/media/vi_13988_20160904_205339.mp4.h264 to /var/www/html/media/vi_13988_20160904_205339.mp4 Queue pos 18
{2016/09/04 20:53:50} Finished boxing /var/www/html/media/vi_13988_20160904_205339.mp4 from Box Queue at pos 18
{2016/09/04 20:53:50} Removed item from Box Queue
{2016/09/04 20:54:51} send smd 1
[2016/09/04 20:54:51] Start capture requested from Pipe
[2016/09/04 20:54:51] Send ca 1
{2016/09/04 20:54:51} Capturing started
{2016/09/04 20:54:59} send smd 0
[2016/09/04 20:54:59] Stop capture requested
[2016/09/04 20:54:59] Send ca 0
{2016/09/04 20:54:59} Capturing stopped
{2016/09/04 20:54:59} Add /var/www/html/media/vi_13989_20160904_205451.mp4 to Box Queue at pos 19
{2016/09/04 20:54:59} Start boxing /var/www/html/media/vi_13989_20160904_205451.mp4.h264 to /var/www/html/media/vi_13989_20160904_205451.mp4 Queue pos 19
{2016/09/04 20:55:00} Finished boxing /var/www/html/media/vi_13989_20160904_205451.mp4 from Box Queue at pos 19
{2016/09/04 20:55:00} Removed item from Box Queue
{2016/09/04 20:55:08} send smd 1
[2016/09/04 20:55:08] Start capture requested from Pipe
[2016/09/04 20:55:08] Send ca 1
{2016/09/04 20:55:08} Capturing started
[2016/09/04 20:56:11] Scheduled management tasks. Next at 1473012971
[2016/09/04 20:56:11] Purged 30 Files
{2016/09/04 21:00:12} send smd 0
[2016/09/04 21:00:12] Stop capture requested
[2016/09/04 21:00:12] Send ca 0
{2016/09/04 21:00:12} Capturing stopped
{2016/09/04 21:00:12} Add /var/www/html/media/vi_13990_20160904_205508.mp4 to Box Queue at pos 20
{2016/09/04 21:00:13} Start boxing /var/www/html/media/vi_13990_20160904_205508.mp4.h264 to /var/www/html/media/vi_13990_20160904_205508.mp4 Queue pos 20
{2016/09/04 21:01:48} send smd 1
[2016/09/04 21:01:48] Start capture requested from Pipe
[2016/09/04 21:01:48] Send ca 1
{2016/09/04 21:01:49} Capturing started
{2016/09/04 21:02:11} Watchdog detected problem. Stopping{2016/09/04 21:02:13} SIGINT/SIGTERM received, stopping
{2016/09/04 21:02:13} Capturing stopped
{2016/09/04 21:02:13} Add /var/www/html/media/vi_13991_20160904_210149.mp4 to Box Queue at pos 21
{2016/09/04 21:02:18} RaspiMJPEG Version 5.6.5
{2016/09/04 21:02:37} recommended video buffers 1
{2016/09/04 21:02:37} h264 buffers set to recommended 1
{2016/09/04 21:02:37} Set up internal detect width=82 height=61
{2016/09/04 21:02:37} MJPEG streaming, ready to receive commands
{2016/09/04 21:02:37} send smd 9
{2016/09/04 21:02:37} Opening FIFO 0 /var/www/html/FIFO 7
{2016/09/04 21:02:37} Opening FIFO 1 /var/www/html/FIFO11 8
{2016/09/04 21:02:37} Starting command loop
[2016/09/04 21:02:37] Reload parameters command requested
[2016/09/04 21:02:37] Scheduler loop is started
{2016/09/04 21:02:37} send smd 1
[2016/09/04 21:02:37] Start capture request ignored, day period not initialised yet
[2016/09/04 21:02:38] Scheduled management tasks. Next at 1473013358
[2016/09/04 21:02:40] Purged 20 Files
{2016/09/04 21:02:49} send smd 0
[2016/09/04 21:02:55] Stop capture requested
[2016/09/04 21:02:55] Send ca 0
{2016/09/04 21:02:55} Already stopped. Ignore 
[2016/09/04 21:03:00] New period detected 0
{2016/09/04 21:03:53} send smd 1
[2016/09/04 21:03:53] Start capture requested from Pipe
[2016/09/04 21:03:53] Send ca 1
{2016/09/04 21:03:53} Capturing started
{2016/09/04 21:04:00} send smd 0
[2016/09/04 21:04:00] Stop capture requested
[2016/09/04 21:04:00] Send ca 0
{2016/09/04 21:04:01} Capturing stopped
[... 92 lines cut ...]
{2016/09/04 21:18:27} send smd 0
[2016/09/04 21:18:27] Stop capture requested
[2016/09/04 21:18:27] Send ca 0
{2016/09/04 21:18:27} Capturing stopped
{2016/09/04 21:18:27} Add /var/www/html/media/vi_14000_20160904_211718.mp4 to Box Queue at pos 8
{2016/09/04 21:18:35} Start boxing /var/www/html/media/vi_14000_20160904_211718.mp4.h264 to /var/www/html/media/vi_14000_20160904_211718.mp4 Queue pos 8
{2016/09/04 21:18:36} send smd 1
[2016/09/04 21:18:36] Start capture requested from Pipe
[2016/09/04 21:18:36] Send ca 1
{2016/09/04 21:18:36} Capturing started
{2016/09/04 21:18:42} send smd 0
{2016/09/04 21:18:55} Watchdog detected problem. Stopping[2016/09/04 21:18:58] Stop capture requested
[2016/09/04 21:18:58] Send ca 0
{2016/09/04 21:19:02} SIGINT/SIGTERM received, stopping
{2016/09/04 21:19:04} send smd 1
[2016/09/04 21:19:06] Start capture requested from Pipe
[2016/09/04 21:19:06] Send ca 1
[2016/09/04 21:22:39] Scheduled management tasks. Next at 1473014559
[2016/09/04 21:42:40] Scheduled management tasks. Next at 1473015760
[2016/09/04 22:02:41] Scheduled management tasks. Next at 1473016961
[2016/09/04 22:22:42] Scheduled management tasks. Next at 1473018162
[2016/09/04 22:42:43] Scheduled management tasks. Next at 1473019363
[2016/09/04 23:02:44] Scheduled management tasks. Next at 1473020564
[2016/09/04 23:22:45] Scheduled management tasks. Next at 1473021765
```