silvanmelchior/RPi_Cam_Web_Interface #299: uninstalling removed all my files in /var/www/html

aguaviva, 2016-08-13:
The uninstaller is supposed to remove ONLY the stuff you installed... the remove.sh actually removed all my /var/www/html files, uninstalled stuff I needed and who knows what else the remove.sh script messed up :(  Also saving jpg's in the /dev/ is a no-no. I am rather furious.

roberttidey, 2016-08-14:
The remove script attempts to move the media folder to ~/media before completely cleaning the web install folder.

I don't understand the reference to saving jpgs in /dev/. The /dev/shm/mjpeg folder is used to hold the jpeg stream to support the preview but these are temporary files stored there as it is a memory folder and therefore does not cause flash wear.

It is always a good idea to back up important items.

aguaviva, 2016-08-14:
/dev is only for devices, not for storing temporal items. This is a pretty basic one. Consider using pipes.

As for backing up important items... this is not necessary as long as the app is well behaved, this is the first time I see so many bad practices together, so please don't blame the user. if you don't want to fix it at least document it quite clearly that you are going to erase full directories.

roberttidey, 2016-08-14:
This software is open source and is free. You can choose to use it or not. We welcome polite comments or pull requests to make improvements to the software. Likewise the documentation wiki on elinux can be modified by anybody.

/dev is indeed for devices. One of those is shm which is a shared memory device intended for passing data efficiently and that it is what is being used for here. http://www.cyberciti.biz/tips/what-is-devshm-and-its-practical-usage.html