silvanmelchior/RPi_Cam_Web_Interface #300: reboot pi 3 while motion detection ON

popoviciri, 2016-08-21:
Couldn't find anyone with this issue: while external motion detection is ON, reboot hangs with "sh -c echo -n '0' >/var/www/html/FIFO1 &" ran by www-data user. Once this is killed, reboot/halt resumes.
Running 6.2.17 with Nginx and external motion on Jessie lite on 5 RPIs. Three rpi3 via on-board wifi have this issue. Other two RPI1 (one via LAN and another via USB wifi) are perfectly fine. Appreciate any feedback!
Regards,
radu
[image]

roberttidey, 2016-08-21:
If this is a standard install this command must be being issued by the motion process but I am not sure why it has a & (async op) on the end unless that is being added automatically by motion itself.

Normally motion should not be started until the rest of the system has got going but if for some reason motion is being started early then it might hang because the FIFO1 has not got going.

Have you done anything to cause motion to start up by itself? You should not install motion separately but let the install.sh do it all.

If you have put motion_detection true in /etc/raspimjpeg then it might be worth trying setting it false and putting md 1 in the period start of the scheduler as this will delay the start up of motion even more.

It is also puzzling why the behaviour is different between the Pis. The Pi3 will start up quicker so that might be a factor but more likely it is some difference in config

popoviciri, 2016-08-22:
Thank you for replying so quickly.
I have a standard install on all PIs; motion installed by install.sh. Also I do not start motion by raspimjpeg. Only on one Pi I did fiddle around with a python telegram bot to start/stop motion via text messages.  That same service is running motion at startup but only if was running before the cron reboot. Not relevant since the others have the same issue.
Must be something I did on all systems (separate installs by the way). Another common factor is that all RPi3s have the v2 pinoir camera.
Thanks for your help!

The last lines from level 9 in /var/log/motion.log):
[1] [NTC] [NET] [Aug 22 01:59:21] netcam_read_html_jpeg: leaving netcam connected.
[1] [INF] [NET] [Aug 22 01:59:21] netcam_image_conv: jpeg_error 0
[1] [DBG] [EVT] [Aug 22 01:59:21] exec_command: Executing external command 'echo -n '0' >/var/www/html/FIFO1'
[1] [NTC] [ALL] [Aug 22 01:59:21] motion_loop: End of event 5
[1] [NTC] [ALL] [Aug 22 01:59:21] motion_loop: Thread exiting
[1] [INF] [ALL] [Aug 22 01:59:21] motion_cleanup: Calling vid_close() from motion_cleanup
[1] [INF] [VID] [Aug 22 01:59:21] vid_close: calling netcam_cleanup
[1] [ALR] [NET] [Aug 22 01:59:21] netcam_handler_loop: netcam camera handler: finish set, exiting
[0] [NTC] [ALL] [Aug 22 01:59:21] main: Threads finished
[0] [NTC] [ALL] [Aug 22 01:59:22] main: Motion terminating
[0] [NTC] [ALL] [Aug 22 01:59:22] motion_remove_pid: Closing logfile (/var/log/motion.log).

roberttidey, 2016-08-22:
I have re-ordered some start up code to make usre FIFO is alive before program tries to start external motion, but I don't think that is your problem as it should only apply if motion_detection was set true in /etc/raspimjpeg

If there is something else that may start up the motion process outside of raspimjpeg then that could cause the problem. Normally raspimjpeg stops and starts motion itself as required by motion detection.

The difference between your systems could be explained just by the difference in CPU speed. If something else is starting up motion then it will be race as to what gets going first.

If you want to control motion detection externally then I would send md 1 (start) and md 0 (stop) to the FIFO rather than try to start the motion process.

popoviciri, 2016-08-23:
Appreciate the effort! Problem is still persistent.
I am convinced that this is something in my setups so there is not much you could do. I just thought this might be caused by raspimjpeg since the call to FIFO is hanging the shutdown. For now I added a "echo 'md 0' >/var/www/html/FIFO" before the reboot command in the daily cron. When I reboot manually I must just pay attention to stop motion detection if running.
In the coming weekend I'll spend a few more hours on this. Perhaps a clean install only on the sd card. If I find the issue I'll post it back here. Otherwise close this.
Thanks again!

popoviciri, 2016-08-28:
Some more tryouts which did not solve my error:
- clean install: Jessie Lite + update + install git & clone & install.sh
- motion_3.4.1raspbianjessierc01-1_armhf.deb
- different EOL in motion.config: win/unix/mac
- disable wifi, bluetooh, alsa etc..
- different wifi adapter;
- different usb adapter;
- US utc8 locale;
- many other things I forgot

What works:
- internal motion;
- external motion with on_event_end = (NULL);
- running the following script in my daily cron:

#!/bin/bash
if [ $(cat /dev/shm/mjpeg/status_mjpeg.txt) == "md_ready" ]; then
    echo 'md 0' >/var/www/html/FIFO
    sleep 5
else if [ $(cat /dev/shm/mjpeg/status_mjpeg.txt) == "md_video" ]; then
    echo -n '0' >/var/www/html/FIFO1 | echo 'md 0' >/var/www/html/FIFO
    sleep 5
    fi
fi
sudo reboot

A workaround but it works. 
Again: only on my RPI3 setups (all three of them) I have this issue. Very strange nobody else has it.

closed 2017-02-08 by popoviciri