silvanmelchior/RPi_Cam_Web_Interface #301: "no process found" error

MyLifeForAiur, 2016-08-23:
When I try to start the RPi Cam Web Interface all I get in response is 
"
raspimjpeg: no process found
php: no process found
motion: no process found
"
What am I doing wrong or forgetting to do in the setup?

roberttidey, 2016-08-23:
When starting up it first tries to close any running versions of those processes so those are not necessarily errors.

After a start up then ps -A should show 2 copies of raspimjpeg running and one copy of php.

You should then be able to browse to the camera at http://yourcameraip/ , or http://yourcameraip/yoursubfolder if you have used a subfolder like html to install into.

geotheory, 2016-12-24:
I'm also encountering this. Camera works manually.  But the programme fails:

```
pi@raspi:~/RPi_Cam_Web_Interface $ sudo ./start.sh
motion: no process found
motion: no process found
pi@raspi:~/RPi_Cam_Web_Interface $
```

Alternatively:

```
pi@raspi:~/RPi_Cam_Web_Interface $ sudo sh start.sh
+ readlink -f start.sh
+ dirname /home/pi/RPi_Cam_Web_Interface/start.sh
+ cd /home/pi/RPi_Cam_Web_Interface
+ source ./config.txt
start.sh: 40: start.sh: source: not found
+ fn_stop
+ sudo killall raspimjpeg
+ sudo killall php
motion: no process found
+ sudo killall motion
motion: no process found
+ sudo mkdir -p /dev/shm/mjpeg
+ sudo chown www-data:www-data /dev/shm/mjpeg
+ sudo chmod 777 /dev/shm/mjpeg
+ sleep 1
+ sudo su -c raspimjpeg > /dev/null & www-data
+ [ -e /etc/debian_version ]
+ sleep 1
+ sudo su -c php /var/www//schedule.php > /dev/null & www-data
```

Navigating to pi's IP`http://192.168.1.100` (also adding `:80`) just loads an Apache folder (i.e. `Index of / ... Apache/2.4.10 (Raspbian) Server at 192.168.1.100 Port 80`) and not the Cam Control interface.

The `ps -A` report includes following lines:
```
  992 ?        00:00:00 kworker/u2:2
 1189 ?        00:00:00 raspimjpeg
 1192 ?        00:00:00 raspimjpeg
 1219 ?        00:00:00 php
 1231 pts/0    00:00:00 ps
```

pishotta, 2016-12-24:
In the start and stop scripts, it might be a good idea to add " 2> /dev/null" to some of those killall statements.  At least the ones for "killall motion", since it really isn't an error.

roberttidey, 2016-12-31:
I have added error redirects to install and the raspimjpeg binary to eliminate these