silvanmelchior/RPi_Cam_Web_Interface #303: change button

Deathsmile971, 2016-09-07:
Hi dudes, i'm using RPi_Cam_Web_Interface and i want to change some things, at first i want to change buttons in a green button when it is active and red when it is disabled i also want to change the background of my website but when i try to edit the code i have nothing in return. Sorry for my bad english, awaiting your answer.

roberttidey, 2016-09-11:
To make changes to the web side you must change the files in the actual web install folder (e.g. /var/www/html). Changing the ones in the RPi_Cam_Web_Interface folder will have no effect; these are just the files that get copied to the working location when an install is done.

Deathsmile971, 2016-09-11:
Hi, I change the right file index.php, i want change the color of the
button in the css file preview.css but nothing passed.. I wanted a green
button when the camera is active and red button when the camera is
inactive. I try to add a button named "Contact" in the index.php file but
nothing passed don't know why..
I have to change the file in the console maybe ?
Sorry for my bad english and thanks for you're time !

Deathsmile971, 2016-09-21:
I can change files now thanks. I wanted to configure motion, I have to notice that this one works only when I use the port 8081 (live stream) but prevent me then from seeing my own camera on my Web interface. How to have and the display of the camera on my Web interface (port 8001) and the stream live (port 8081) functional at the same time?

Thank you for reading to me and see you soon.