silvanmelchior/RPi_Cam_Web_Interface #305: Changing media path directory

kashanirod, 2016-09-08:
Hi there! 

Thank you for this application! Its works wonderfully and comes equipped with plenty of features. 

I am trying to use an external media storage to store the camera data on, however when changing the paths, they do not write to it. 

I'm using USBMOUNT and writing to the correct directory. Im changing the directory in the RASPIMJPEG configuration file but with no luck. I have tried just writing to another location on the PI and am unable to achieve this as welI. 

Thank you

roberttidey, 2016-09-11:
It is best not to change the directory in the raspimjeg config file but instead link your external storage to the same location e.g. /var/html/media. All the media files will then be stored on the external device. This is achieved by adding a mount in fstab

rajeshkrarora1, 2019-04-07:
how to create the mount as the procedure given at [https://elinux.org/RPi-Cam-Web-Interface#CIFS_.2F_Samba_.2F_Windows_Share](url) does not work

roberttidey, 2019-04-07:
Can you give more details about what is not working?

Does each step in the process work without an error message?

Have you rebooted the Pi?

Can you list the contents of the share from the Pi?

What is the behaviour when the recording tries to use the share?

rajeshkrarora1, 2019-04-07:
Hi sir's
The share is working fine, i did not follow the instructions correctly at
first

wanamaa, 2019-10-28:
> It is best not to change the directory in the raspimjeg config file but instead link your external storage to the same location e.g. /var/html/media. All the media files will then be stored on the external device. This is achieved by adding a mount in fstab

@roberttidey I'm a complete noob with Linux and am trying to set up a camera to write to an external drive so I can record a manufacturing cell all day, ideally 5 days a week.  I have a mount point at /mnt/storage for my drive at /dev/sda1.  So, if I wanted fstab to bind with the media folder, would I enter it in like below?

`/var/www/html/media  /mnt/storage/media  none  bind`

I have to remote into the Pi unfortunately and messing with fstab has crashed my build a couple times, I want to make sure I get it correct.

Another noob question - my SD card is 32GB (the biggest my Pi 2 will allow).  My understanding with binding is that the files are essentially duplicated at both locations.  What happens when/if the SD card gets full? Would I need to create a script to delete the files?

roberttidey, 2019-10-28:
I normally use soft link to do this. This creates a link called media in the var/www/html which points to a folder on your mounted drive. The files then get stored in that folder just once but can be  referred to by their original path name.

So create a media folder in your mount
mkdir /mnt/storage/media

remove the media folder in /var/www/html (save any files in there you want to keep first)

sudo rm -rf /var/www/html/media

Now create the link
sudo ln -s /mnt/storage/media /var/www/html/media

The media folder should be owned by user www-data

So do
sudo chown www-data:www-data /var/www/html/media

If you do ls - /var/www/media you should see a media folder but pointing to your drive mount

I would use ext4 as the filing system on the drive as that is the native linux filing system, but I think NTFS will also work.

wanamaa, 2019-10-28:
I ran the commands, rebooted the program and am now seeing the space listed as ~1.8TB - a file transferred over properly and I'm seeing it on my hard drive as well as the interface.  Thank you so much @roberttidey!

wanamaa, 2019-10-29:
Update: So I tried running this today, having it split the file every 10 minutes so I don't have a HUGE file to go through later.  It seems that, once it is going to move and make the split, it locks the tool up.  Has anyone run into this before?  I'll try and look through the logs.

roberttidey, 2019-10-29:
When running on external drives like you now have it is a good idea to use the facility to temporarily record to the SD card and have them boxed / moved to the media directory as each completes.

Look in wiki in second part of CIFS / Samba / Windows Share for instructions on how to do this.