silvanmelchior/RPi_Cam_Web_Interface #308: %a to display contents of user_annotate.txt issue

akobelan, 2016-09-28:
When using %a to append text to the overlayed text on the streaming image, a "?" character is appended to the displayed text in addition to the message I'm trying to display.

For example, I want to append "@300 rpm, +10 deg" and the string "@300 rpm, +10 deg?" gets appended.

Running on Jessie.

roberttidey, 2016-09-28:
You probably have a line terminator in the file which is being represented as a ? Just make sure the text is just the characters to be included.

bloodbank, 2017-07-01:
@akobelan did you ever resolve this issue?

roberttidey, 2017-07-01:
I didn't think there was an issue to resolve. If you make sure there is no line feed in the user annotation text file then you should not get the ?

For example if using python to write the file then use write rather than print, or include a ',' at the end of a print to avoid the line ending. If using bash echo then use echo -n

I could add an extra check in the code and remove a trailing line feed if present but one still has to make sure that the user annotation doesn't contain any other unsupported characters. For example, there might be CR or characters outside the normal basic ASCII character set.

bloodbank, 2017-07-01:
Ill try writing with python. I just tested with vi/nano and checked both for hidden and cat -E, cant find any special chars.

pi@raspberrypi:~ $ cat -E /dev/shm/mjpeg/user_annotate.txt
tse$

roberttidey, 2017-07-01:
If you zip an example file and post it here then I can check it out

bloodbank, 2017-07-02:
I made a python script which writes to the user_annotate.txt file, and the question mark is gone now. Must be some editor issue that adds hidden characters or something.

drtp, 2018-02-13:
Hi,

perhaps I might add another question related to the annotation feature. I would like to display the current temperature, which works fine for me. However, if I use a ° character, only two ?? are displayed. Is there any possibility to display e.g. °C correctly or is it necessary to use degC instead?

drtp, 2018-04-13:
Try to call the file "user_annotate.txt" with a preceding "-n":

`echo -n 5 C | ssh pi@192.168.xxx.xxx 'cat > /dev/shm/mjpeg/user_annotate.txt'`