silvanmelchior/RPi_Cam_Web_Interface #309: Not asking for Username & Password

Overcon, 2016-10-02:
Hi,

i just installed the RPi_Cam interface and it is functioning except it is not asking for the username and password when I access the HTML folder. It shows the full options, which I do not want people to access unless they have an account or know the default username and password I configured.

Can anyone help me get the username and password working? 

I did a standard installation according to the instructions located here: http://elinux.org/RPi-Cam-Web-Interface#Simple_live-preview_only_page

IE basic Installation. The first time I told it no user / password. Then I re-ran the installation script and added a username and password, but when I access the site it doesn't ask for the username and password I entered.

Running Noobs, Latest Jessie version latest updates. I have an existing website hosted, running WordPress, that is why I installed it in the sub directory html. I didn't change the port.

Thanks!

roberttidey, 2016-10-02:
Currently the username / password should be restricting access to the site. It doesn't control whether they can see the options or not. E.g. if you have set up username/password and try to access http://ipaddress  (defaults ti index.php) then it should ask for the credentials.

I just tried it with min.php and it did ask for credentials as expected.

Note that once you have logged in then a browser will maintain those credentials even if you close the browser tab. You need to close the browser to get the request again.

There is a request to separate out logins so that simple or settings views can be controlled via the login. I am starting to look at that but it is a significant change as currently the security is just a pure Apache function.

If the base security is not working for you then let me know and give as much detail as you can. Is the Wordpress running under Apache?

Overcon, 2016-10-02:
It has never asked me for a password, even after I changed the setting and added the username and password.

Yes, Wordpress is running under Apache2 and MySQL./ I am just learning this stuff so if you need any information please let me know what and I can get it for you.

I followed this guide to get Wordpress running under Apache2 on the Raspberry Pi 3.

https://www.raspberrypi.org/learning/lamp-web-server-with-wordpress/worksheet/

roberttidey, 2016-10-02:
It is probably something to do with the Wordpress install confusing the installation of the RPi Cam.

Can you zip and post the /etc/apache2/sites-available/raspicam.conf file so I can check it out?

Overcon, 2016-10-03:
Sure, here it is.
[raspi_conf.zip](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/505355/raspi_conf.zip)

roberttidey, 2016-10-03:
The conf file looks OK. 

I just retested re-installing with passwords on and off and it works OK for me.

Further thoughts.

Changes to config only take effect if apache2 service is restarted. install script does this but it might be worth doing again just to make sure (sudo service apache2 restart)

There should be a .htaccess file in the /var/www/html folder that points to the user / password security info. It should just have the following content. Note that . files are normally hidden but you can cat /var/www/html/.htaccess
AuthName "RPi Cam Web Interface Restricted Area"
AuthType Basic
AuthUserFile /usr/local/.htpasswd
Require valid-user

It s also possible that another apache2 conf file is also defining the html subfolder and this may be taking precedence and preventing the raspicam.conf file taking effect. Normally the installler disables the 000-default.conf from doing this but if there are others then that may be the cause. See what other conf files are  in the /etc/apache2/sites-available folder

Overcon, 2016-10-03:
I have already restarted Apache2 to see if that was the issue, no change. I also restarted my browser and flushed the cache to make sure that wasn't the issue. 

The .htaccess file in that dir reads:

~:~ $ sudo cat /var/www/html/.htaccess

# BEGIN WordPress
<IfModule mod_rewrite.c>
RewriteEngine On
RewriteBase /
RewriteRule ^index\.php$ - [L]
RewriteCond %{REQUEST_FILENAME} !-f
RewriteCond %{REQUEST_FILENAME} !-d
RewriteRule . /index.php [L]
</IfModule>

Here are the other files located in the /etc/apache2/sites-available folder:

./default-ssl.conf
./raspicam.conf
./000-default.conf

Overcon, 2016-10-03:
Actually I just noticed. When I login to my webpage hosted on the same RBPI it no longer works with WOrdpress after installing the RPi Cam Preview. It just shows a directory and the HTML folder as a list.

Overcon, 2016-10-03:
Do you know how to revert it to get the default Website back up and running and then just point to the min.php file? I don't mind the min.php view to not have a login, I just didn't want people logging into the site and navigating to the config section. But I would like to keep the Wordpress site up and running, which it no longer is after installing this.

Overcon, 2016-10-03:
Yeah I am unfortunately not an expert or very knowledgeable on HTML, but it looks like WP had also installed itself in /var/www/html and installing the RPi Cam Preview basically used the same file structure, so overwrote some of the WordPress files.

It looks like WordPress had it setup so when you typed the domainname.com it would redirect the root to /www/html/index.php which is the same that Rpi Cam Preview uses. Not sure how I will be able to get them back as independent sites and get the permissions working. I might have to trash the WordPress one as I don't see many options for restoring it and still use the cam preview.

Might be something to consider for future installation scripts, to check if WordPress is installed, though not many people run WordPress on a Pi, though I thought it was pretty nifty, not terribly slow, especially with the pi overclocked.

roberttidey, 2016-10-03:
Thanks for the update.  Sounds like the fastest way to recover may be to reinstall Wordpress say in its html and then install the rpi into a different subfolder like camera.

The installer could do a bit more checking but it also has to allow overwriting itself so it would need to check for some specific file not installed by itself.

chappy1978, 2017-03-16:
Hi
 I am having the same problem with the login. I have never logged in and do not have Wordpress installed. Im working with a Pi0 and have the updated install of Jessie. No errors that I have found just no password.

crus1313, 2019-03-21:
I forgot my pass to the web interface (classic wright?:)). 
How can I reset it? I can access the pi zero via ssh , or physically.

roberttidey, 2019-03-22:
You could just run ./install.sh again and put in new password.

Alternatively you could just do the bit that sets up a password manually.

in ssh do
sudo htpasswd -b -B /usr/local/.htpasswd $user $webpasswd

where you substitute the user and password you want.