silvanmelchior/RPi_Cam_Web_Interface #313: Can#

Cr4xy, 2016-11-29:


roberttidey, 2016-11-29:
Just tried this and it works OK for me so need to find a bit more out as to what is going wrong.

I assume other commands like capture image start video are working OK?

Pressing the stop button asks cmd_pipe to send a ru 0 command and start requests a ru 1 command.

First thing I would try is to issue these commands manually from a terminal by issuing a echo 'ru 0' > /var/www/html/FIFO  (change path if installed into different folder). This should stop the stream and change buttons. Issuing a 'ru 1' the same way should restore the stream.

Also check the scheduler log to see what it thinks is going.

Cr4xy, 2016-11-29:
Thank you @roberttidey, I tried issuing the command mentioned above, but it didn't appear to do anything. The stream just continued as if nothing happenend.

Here is the scheduleLog.txt:

```{2016/11/28 12:50:24} RaspiMJPEG Version 5.6.7
{2016/11/28 12:50:25} recommended video buffers 1
{2016/11/28 12:50:25} h264 buffers set to recommended 1
{2016/11/28 12:50:25} Opening FIFO 0 /var/www/FIFO 7
{2016/11/28 12:50:25} Opening FIFO 1 /var/www/FIFO11 8
{2016/11/28 12:50:25} MJPEG streaming, ready to receive commands
{2016/11/28 12:50:25} send smd 9
{2016/11/29 13:02:25} Starting command loop
{2016/11/29 13:02:25} SIGINT/SIGTERM received, stopping
{2016/11/29 13:02:49} RaspiMJPEG Version 5.6.7
{2016/11/29 13:02:49} recommended video buffers 1
{2016/11/29 13:02:49} h264 buffers set to recommended 1
{2016/11/29 13:02:50} Opening FIFO 0 /var/www/FIFO 7
{2016/11/29 13:02:50} Opening FIFO 1 /var/www/FIFO11 8
{2016/11/29 13:02:50} MJPEG streaming, ready to receive commands
{2016/11/29 13:02:50} send smd 9

Cr4xy, 2016-12-01:
I resolved the issue by reinstalling,
I think the problem was that the root was in `/var/www` instead of `/var/www/html`.

closed 2016-12-01 by Cr4xy