silvanmelchior/RPi_Cam_Web_Interface #314: Remotely sending commands

man-vs-electron, 2016-11-30:
I'd like to be able to turn motion detection on and off "remotely," i.e. from other machines on the local area network.  Specifically, I'd like to be able to issue some URL to the Pi that would execute "md 1" and "md 0."  That way, my home security system could turn motion detection off when I disable monitoring in other parts of the system and turn it back on when I enable it.  

One thought would be something like this:

http://pi_cam.local/command/md_0

would turn monitoring off.

Is this already built in? If not, how hard would it be to add it?  If you could point me to the right points in the code, I might try adding it myself.

pishotta, 2016-11-30:
Hi,

That is precisely what the "Motion Detection Start/Stop" button (on the main web page) does.  You should be able to connect to that server from a web browser on a different system elsewhere in your LAN.  So I don't think anything "new" needs to be added.

-- Fred P.

pishotta, 2016-11-30:
I suppose if you want some dedicated URLs for the tasks, then a simple php file which writes an "md 1" or "md 0" to file "/var/your-cam-path/FIFO" would do the job, or something that issued the bash command, e.g., 

echo "md 1" > /var/your-cam-path/FIFO

roberttidey, 2016-12-01:
The current system uses cmd_pipe.php to issue commands to the pipe so just use a URL with this and md 0 or md 1 as a parameter

man-vs-electron, 2016-12-02:
Thanks for the responses.  I know that the webui can be used to turn motion capture on and off, but I want to do with without the UI sometimes.  I want to be able to start my home monitoring software and have it directly toggle motion capture on multiple devices.  

Both responses looked helpful.  Just to fill out roberttidy's respons, the specific url would be:

http://path.to.camera/html/cmd_pipe.php?cmd=md%200

To turn motion capture off and 

http://path.to.camera/html/cmd_pipe.php?cmd=md%201

To turn it on.

Thanks!

closed 2016-12-02 by man-vs-electron