silvanmelchior/RPi_Cam_Web_Interface #315: Repositories moved?

Michael833, 2016-12-01:
I'm trying to use your software for a project I found on [element14](https://www.element14.com/community/community/raspberry-pi/raspberrypi_projects/blog/2016/04/25/noir-v2-video-streaming-baby-monitor#comment-94921), but when I do it seems to be saying that some of the repositories moved. Here is the code I'm running on my RPi 2B:
`git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git`
`cd RPi_Cam_Web_Interface`
`chmod u+x *.sh`
`./install.sh`

And here is one of the error messages (they are all like this):
`Err http://mirrordirector.raspbian.org/raspbian/jessie/main libavdevice55 armhf 6:11.7-1~deb8u1+rpi1
  404  Not Found [IP: 5.153.225.207 80]`

Maybe I'm misunderstanding something, but that's what I think is happening. I've gone to that address and I get a 404 error myself too.

roberttidey, 2016-12-01:
Two ideas. Make sure you have done an altogether update and upgrade before running install. Also sometimes there is a problem retrieving a repository from raspberry. Running again later usually resolves this.

Michael833, 2016-12-02:
Robert, 

Thanks so much, I really appreciate your suggestions. I'm very very new to all this RPi stuff and wasn't aware of the update and upgrade functions, figuring that my fresh install was all I needed. This seems to have done the trick. Thanks again for your help.

closed 2016-12-02 by Michael833