silvanmelchior/RPi_Cam_Web_Interface #317: Only one scheduler command is processed

florisjan78, 2016-12-28:
Hello,

I've recenty started using the scheduler on an outdoors camera, using sun based mode. I had the following on period start: em auto;tl 1
and noticed that the timelapse would not start. Not a big problem as em is on auto in the config anyway so using just tl 1 solved the problem. Here's what's logged:

`[2016/12/27 12:10:07] Scheduler loop is started
[2016/12/27 12:10:08] New period detected 3
[2016/12/27 12:10:08] Send em auto
{2016/12/27 12:10:08} Change: exposure_mode = auto
[2016/12/27 12:10:10] Send tl 1
....`

However now I would like to change some more settings and am running into the same problem. The period start line is now: tl 1; br 60; em night

And only the timelapse actually starts, the rest of the commands seems to be ignored. Here's the log:

`[2016/12/28 17:56:50] New period detected 4
[2016/12/28 17:56:50] Send tl 1
{2016/12/28 17:56:50} Timelapse started
[2016/12/28 17:56:52] Send br 60
[2016/12/28 17:56:54] Send em night
[2016/12/28 17:57:06] Scheduled management tasks. Next at 1482947810
....`

I'm probably doing something wrong, could you tell me what?

thanks,

Fj

roberttidey, 2016-12-28:
Thanks for report. There was a change that allowed for line feeds on the end of commands. That might have confused the scheduler in multiple commands but when I checked that version the scheduler does add those in OK.

I'll check out further.

roberttidey, 2016-12-28:
Hmm. I have tried this out on one of my cameras and it seemed to be OK.

I tested by putting em night;tl 1 in a period start (night) and then did a save settings of the scheduler which causes it to start again and therefore process the period start commands to get in sync.

Log shows
[2016/12/28 22:54:07] Saved schedule settings
[2016/12/28 22:54:07] Send Schedule reset
[2016/12/28 22:54:07] Reload parameters command requested
[2016/12/28 22:54:07] Scheduler loop is started
[2016/12/28 22:54:08] New period detected 1
[2016/12/28 22:54:08] Send em night
{2016/12/28 22:54:08} Change: exposure_mode = night
[2016/12/28 22:54:10] Send tl 1
{2016/12/28 22:54:10} Timelapse started
{2016/12/28 22:54:11} Capturing image
{2016/12/28 22:54:12} Capturing image

Note that it not only says 'send tl 1' which comes from scheduler but also says 'Timelapse started' which comes from raspimjpeg responding to the command request.

If you can't progress could you post the scheduler.json file from your web install folder so I can see the exact settings in use.

florisjan78, 2016-12-29:
I see you do get responses on all commands, weird. 

Here's my schedule.json:

{"Fifo_In":"\/var\/www\/html\/FIFO1","Fifo_Out":"\/var\/www\/html\/FIFO","Cmd_Poll":"0.03","Mode_Poll":"10","Management_Interval":"3600","Management_Command":"","PurgeVideo_Hours":"0","PurgeImage_Hours":"0","PurgeLapse_Hours":"0","GMTOff
set":"0","PurgeSpace_ModeEx":"0","PurgeSpace_Level":"10","DawnStart_Minutes":"-180","DayStart_Minutes":"0","DayEnd_Minutes":"0","DuskEnd_Minutes":"180","Latitude":"52.1","Longtitude":"5.29","Max_Capture":"0","DayMode":"0","AutoCapture_In
terval":"0","AutoCamera_Interval":"0","Commands_On":["ca 1","ca 1","ca 1","ca 1","ca 1","","","","","",""],"Commands_Off":["ca 0","ca 0","ca 0","ca 0","ca 0","","","","","",""],"Modes":["","tl 0; br 60; em night","tl 1; br 60; em auto","
tl 1; br 50; em auto","tl 1; br 60; em night","","","","","",""],"Times":["09:00","10:00","11:00","12:00","13:00","14:00"]}

roberttidey, 2016-12-29:
Nothing obviously wrong there. Very similar config works for me.

Two things to try.

1) Check that schedule.php is correct version. In particular if you check around line 446 then you should see a line
            fwrite($fifo, $cmd . "\n");

This where it is sending the command and adding a line feed on.

2) As an experiment increase the sleep time a couple of lines later from 2 to say 4. This is the gap between commands. It shouldn't be needed but I want to see if it has an effect on your system.

florisjan78, 2016-12-29:
function sendCmds($cmdString) {
      global $schedulePars;

      $cmds = explode(';', $cmdString);
      foreach ($cmds as $cmd) {
         if ($cmd != "") {
            writeLog("Send $cmd");
            $fifo = fopen($schedulePars[SCHEDULE_FIFOOUT], "w");
            fwrite($fifo, $cmd . "\n"); //line 446
            fclose($fifo);
            sleep(2);
         }
      }
   }
   
I changed the sleep to 4 and restarted the Pi, no difference;

[2016/12/30 00:11:23] New period detected 1
[2016/12/30 00:11:23] Send tl 0
{2016/12/30 00:11:23} Timelapse stopped
[2016/12/30 00:11:27] Send br 60
[2016/12/30 00:11:42] Send em night
[2016/12/30 00:11:46] Scheduled management tasks. Next at 1483056683

roberttidey, 2016-12-30:
Strange.

Could you open a terminal window and then issues the three commands manually?

echo 'tl 0' >/var/www/html/FIFO
echo 'br 60' >/var/www/html/FIFO
echo 'em night' >/var/www/html/FIFO

Then look in scheduler log. You won't see the commands being sent there but you should see the raspimjpeg logging processing the commands. raspimjpeg log entries have curly brackets around the time.

florisjan78, 2016-12-30:
This did result in responses:

{2016/12/30 12:18:43} Timelapse stopped
{2016/12/30 12:18:53} Change: brightness = 60
{2016/12/30 12:19:06} Change: exposure_mode = night

closed 2017-06-28 by florisjan78