silvanmelchior/RPi_Cam_Web_Interface #318: System periodically hangs when running in Motion Detect mode

KarlBaumann, 2016-12-29:
I have this problem with multiple RPIs, they run for few days or even few hours and then this error shows up 
[image]
Usually reboot solves the problem, but one RPI has to be unplugged from power source to get it working again, since nor Apache, nor SSH is not working anymore.

roberttidey, 2016-12-29:
Check Logs to see if they show anything leading up to failure.

It is also worth making sure the power arrangements are rock solid. I have seen many instances where a marginal power at the raspberry can cause this particularly as the camera plus motion detection up the current a bit from normal. In particular if the power is going in through the normal micro usb then the quality of the cable is very important. Many USB cables have quite a high resistance and cause a significant voltage drop no matter how good the base power adapter is.

KarlBaumann, 2016-12-29:
Interesting point about the cable. 
My RPI is currently in hanged state, would you like to SSH in to it and see whats wrong? I am not so good with linux, will be difficult to spot the error in logs myself.

KarlBaumann, 2016-12-29:
/var/log/messages
[image]

KarlBaumann, 2016-12-29:
/var/log/user.log
[image]
This might be the time when it stopped working

roberttidey, 2016-12-29:
Yes. Definitely check out the cables. If you have a meter then you can measure the voltage at the raspberry pi end when it is running. It should be greater than 4.8V. The application logs are available under scheduler settings.

It looks like you are using external motion detection. I would recommend switching to internal motion detection as it is uses less resources and is more stable.

KarlBaumann, 2016-12-29:
Okay, I will check the voltage. 
I am using default settings, would external motion be selected by default?

Are you sure you dont want to ssh in my RPI? Just drop me an email karlis@dunb.lv , I will send you credentials.

roberttidey, 2016-12-29:
Yes. It is external by default. You can change this under camera settings. The original motion settings button disappears and a new concertina set of internal motion settings appears.